libretro/Lakka #240: Lakka not read xorg.conf

Alexandre-Garcia, 2015-11-21:
First let me say that I follow this project a few months, and I have been impressed by the progress and the project conception.
I use Lakka on a computer with a Core 2 Duo and a 9500GT, connected to a tube TV via an S-video cable.
In the latest nightly version (20151121 ...), when I put the xorg.conf file in the config folder (/storage/.config/), the system appears not read it, so that the contents do not appear on TV.

Kivutar, 2015-11-22:
Ah, that's because we removed X11 in favor of DRM/KMS in the latest builds. 

Have you tried the setting called "Monitor Index" ?

Also try to add video=SVIDEO-1:d in the cmdline.

Kivutar, 2015-11-22:
Or this video=LVDS-1:d

Alexandre-Garcia, 2015-11-22:
I tried both the parameters in the cmdline and the "Monitor Index", however unsuccessfully.
After reading about KMS, I realized I should put "video = SVIDEO-1: and" (e to enable), to force the output to on, however did not work either.
By listing the directory /sys/class/drm/ I see only the DVI-I and VGA-1-1 outputs.
I would like to put the latest version on my hardware, because the menu is much better now with the core autodetect.
For now I will use the 20151025 version, the last using the X11.
This change is permanent?

Kivutar, 2015-11-23:
This change is not permanent, I will try to make some changes to allow building images with x11 and drm/kms easily.

I guess that the **d** is meant to disable? Then can you try to disable all except the one you want?

Alexandre-Garcia, 2015-11-23:
Ops. I typed wrong. The right is "S-VIDEO-1:**e**", and **d** is to disable.

I tried disable all except the one I want, as you requested, adding "S-VIDEO-1:e DVI-I-1:d VGA-1:d", and resulted in black screen in all outputs.

duduke, 2015-11-24:
Try this to get the list of outputs and their status:
for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done

And try this:
video=TV-1:e
to enable TV-Out (SVIDEO/Composite) although if it is not listed under sys/class/drm it's possible that the opensource nouveau driver does not enumerate it correctly.

Alexandre-Garcia, 2015-11-26:
The command was wrong because of asterisks. I deduced the correct italics. the return was:
DVI-I-1:disconnected
VGA-1:connected

I tested the command "video = TV-1: e", but it does not work.

I think the trouble is with the nouveau driver ... probably does not enable TVDAC decoder, or something like that.

Kivutar, 2015-12-07:
I plan to update the kernel soon, we will see if it fixes this issue.

Kivutar, 2016-01-09:
@Alexandre-Garcia I upgraded the kernels and video blobs and kernel configs. Can you try again please?

Alexandre-Garcia, 2016-01-28:
@Kivutar 
I tried releases r21295 and r21276, using the commands mentioned in previous comments, but not worked.
The symptoms are the same, DVI and VGA outputs appear, but TV-OUT or S-Video doesnt.

gouchi, 2016-02-17:
According to the nouveau matrix feature tv-out is not yet implemented for nv50 family
https://nouveau.freedesktop.org/wiki/FeatureMatrix

Alexandre-Garcia, 2016-02-17:
Now it is clear why didn't work.
I have a 6200 (NV40) at home. I'll try with this.

Alexandre-Garcia, 2016-02-18:
Good news. With the 6200 s-video works all except the N64 emulator.
To run a game screen is just flickering. I tested with two cards and no work (I didn't even know had two 6200). 
I do not think the board is too weak because one of them was on a PC with Windows that runs Mupen64 perfectly.

gouchi, 2016-02-18:
Do you use the Lakka 32 bits version?  Because it seems there is an issue with Mupen64Plus on i386 #138

Alexandre-Garcia, 2016-02-18:
64 Bits. Same release with 9500gt works fine, and with onboard intel 950 works too, but very slow.

Kivutar, 2016-03-14:
@Alexandre-Garcia I upgraded the kernel+mesa+libdrm recently. Could you check if it fixes the bug please?

Alexandre-Garcia, 2016-03-16:
@Kivutar Now i get "retro_read_file: Failed to open /storage/savefiles/Mario Kart 64 (Europe).srm: No such file or directory
Segmentation fault" with 	Lakka-Generic.x86_64-devel-20160312232001-r21367-ge423427
Genesis, Snes and PS1 cores are working fine.

Alexandre-Garcia, 2016-03-17:
I find this thread on libretro forum about the same issue apparently. 
http://libretro.com/forums/showthread.php?t=5265

Alexandre-Garcia, 2016-03-17:
Now I tested with release "Lakka-Generic.x86_64-devel-20141105153004-r19688-g50ca5b9", which uses the proprietary driver with xorg,  to see if it could not be the nouveau driver, and the same error occurred.

Kivutar, 2016-03-17:
Looks like a mupen64 specific issue about save files?

Alexandre-Garcia, 2016-03-17:
I don't hink so, because the output "retro_read_file: Failed to open..." occurs every first time I play a rom.
I put a ROM with a save file and get only a segmentation fault.
Then I just change the nvidia 6200 graphics card to a 9500 gt and works fine with the same ROM.
Just for information, the same hardware (with the nvidia 6200), run the sames ROMs with Windows and Retroarch.

Kivutar, 2016-03-17:
The problem you describe is an IO issue that is certainly unrelated to graphics.

Alexandre-Garcia, 2016-03-17:
I think have 2 issues in my log.
First is an IO isso, reported on line "Failed to open /storage/savefiles/Mario Kart 64 (Europe).srm: No such file or directory".
The second is a graphic related SEGFAULT, reported on line "Segmentation fault".
I think this because with the 9500gt, the IO error also occurs, but after that the .srm file is created and the core works fine.
I don´t know how extract more informations from system, but if helps I rotate a TCP port to my Lakka PC, for SSH access. Then you can connect and take more info.

Alexandre-Garcia, 2016-03-17:
Testing with old release r19688, starting retroarch through SSH, i receive these:
RetroArch [ERROR] :: init_video_input :: Cannot open video driver ... Exiting ...
RetroArch [ERROR] :: rarch_main_init :: Fatal error received in: "init_video_input()"
Segmentation fault
Again, replacing the 6200 VGA  to 9500 GT, works.

Alexandre-Garcia, 2016-03-18:
It is possible that this core need "geometrics instructions"?
For all the tests I made, with all the old video cards nvidia I have, those who worked are the NV50 family or above, which has the feature "Geometry programs" implemented as the matrix indicated by gouchi.

Alexandre-Garcia, 2016-03-20:
Just for information.
I made a new test on a fresh installation of Ubuntu 14.04, with nouveau driver, using the stable retroarch of oficial PPA and mupen64plus directly from the repository ubuntu, just to satisfy my curiosity checking if it could be something specific of mupen64plus.
The mupen64plus worked properly.
The Retroarch crash when trying to run the Nintendo64 ROM.
Other cores of Retroarch run well.

gouchi, 2016-03-20:
Not sure if it related to this [issue](https://github.com/libretro/mupen64plus-libretro/issues/303) ?

Alexandre-Garcia, 2016-03-25:
I'm not getting compile mesa with debug mode, so I can not tell if it's related.
The result gdb is as follows:
$ gdb retroarch
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Lendo símbolos de retroarch...Lendo símbolos de /usr/lib/debug/.build-id/e6/11b425ae38b9a0f2d96d1926a34b7297932bce.debug...concluído.
concluído.
(gdb) run
Starting program: /usr/bin/retroarch 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffec73a700 (LWP 10275)]
[New Thread 0x7fffebf39700 (LWP 10276)]
[New Thread 0x7fffe2c77700 (LWP 10309)]
[Thread 0x7fffe2c77700 (LWP 10309) exited]
[New Thread 0x7fffe2c77700 (LWP 10317)]
filestream_read_file: Failed to open /home/alexandre/Downloads/Super Mario 64.srm: No such file or directory

Program received signal SIGSEGV, Segmentation fault.
0x00000000004e06cc in gfx_ctx_ctl (state=state@entry=GFX_CTL_CHECK_WINDOW, 
    data=data@entry=0x7fffffffcd50) at gfx/video_context_driver.c:275
275	gfx/video_context_driver.c: Arquivo ou diretório não encontrado (File or directory not found).

Kivutar, 2016-03-26:
Ok when you are at that point, can you type backtrace?

Alexandre-Garcia, 2016-03-26:
(gdb) backtrace
#0  0x00000000004e06cc in gfx_ctx_ctl (state=state@entry=GFX_CTL_CHECK_WINDOW, 
    data=data@entry=0x7fffffffcd50) at gfx/video_context_driver.c:275
#1  0x00000000004e4756 in gl_alive (data=0x0) at gfx/drivers/gl.c:2955
#2  0x0000000000429829 in video_driver_ctl (
    state=state@entry=RARCH_DISPLAY_CTL_IS_ALIVE, data=data@entry=0x0)
    at gfx/video_driver.c:1616
#3  0x000000000041c264 in runloop_iterate_time_to_exit (quit_key_pressed=false)
    at runloop.c:1244
#4  runloop_iterate (sleep_ms=sleep_ms@entry=0x7fffffffce80) at runloop.c:1365
#5  0x0000000000411fcf in rarch_main (argc=<optimized out>, 
    argv=<optimized out>, data=0x0) at frontend/frontend.c:163
#6  0x00007ffff25b5ec5 in __libc_start_main (main=0x40f540 <main>, argc=1, 
    argv=0x7fffffffdfe8, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fffffffdfd8) at libc-start.c:287
#7  0x000000000040f570 in _start ()

Kivutar, 2016-04-02:
@twinaphex or @maister any idea?

twinaphex, 2016-04-02:
Check if this gets you past the crash:

https://github.com/libretro/RetroArch/commit/15fa7f9cd19170d1e02fab79caf93536f8167b20

Alexandre-Garcia, 2016-04-25:
Now I have a GL_INVALID_OPERATION before SIGSEGV, and the backtrace presents other result.

(gdb) run
Starting program: /usr/bin/retroarch 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffec73a700 (LWP 13203)]
[New Thread 0x7fffebf39700 (LWP 13204)]
[New Thread 0x7fffe2c4e700 (LWP 13241)]
Mesa: User error: GL_INVALID_OPERATION in glUniform1(uniform "MVPMatrix"@0 is matrix)
[Thread 0x7fffe2c4e700 (LWP 13241) exited]
[New Thread 0x7fffe2c4e700 (LWP 13264)]
filestream_read_file: Failed to open /home/alexandre/Downloads/Super Mario 64.srm: No such file or directory

Program received signal SIGSEGV, Segmentation fault.
font_driver_bind_block (font_data=<optimized out>, block=0x0)
    at gfx/font_driver.c:263
263	gfx/font_driver.c: Arquivo ou diretório não encontrado.
(gdb) bt
#0  font_driver_bind_block (font_data=<optimized out>, block=0x0)
    at gfx/font_driver.c:263
#1  0x0000000000482e63 in menu_driver_ctl (
    state=state@entry=RARCH_MENU_CTL_DEINIT, data=data@entry=0x0)
    at menu/menu_driver.c:558
#2  0x000000000042d4e4 in uninit_drivers (flags=127) at driver.c:398
#3  0x000000000042df83 in driver_ctl (state=<optimized out>, 
    data=0x7fffffffce10) at driver.c:443
#4  0x000000000042dfa4 in driver_ctl (
    state=state@entry=RARCH_DRIVER_CTL_UNINIT_ALL, data=data@entry=0x0)
    at driver.c:449
#5  0x00000000004159c0 in rarch_ctl (state=state@entry=RARCH_CTL_DESTROY, 
    data=data@entry=0x0) at retroarch.c:1386
#6  0x0000000000412278 in main_exit (args=0x0) at frontend/frontend.c:80
#7  rarch_main (argc=<optimized out>, argv=<optimized out>, data=0x0)
    at frontend/frontend.c:170
#8  0x00007ffff25b5ec5 in __libc_start_main (main=0x40f750 <main>, argc=1, 
    argv=0x7fffffffdfe8, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fffffffdfd8) at libc-start.c:287
#9  0x000000000040f780 in _start ()

gouchi, 2016-05-11:
If we enable logging will get more output. I reproduce the error on my PC with my old mesa 10.3.2 (and with the latest open source graphic stack).

gdb ./retroarch
GNU gdb (Debian 7.7.1+dfsg-5) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./retroarch...done.
(gdb) r
Starting program: /home/gouchi/Telechargements/RetroArch/retroarch 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffebac6700 (LWP 14846)]
RetroArch [INFO] :: Config: loading config from: /home/gouchi/.config/retroarch/retroarch.cfg.
[New Thread 0x7fffeb2c5700 (LWP 14847)]
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29975.00 Hz.
RetroArch [INFO] :: Found shader "/home/gouchi/.config/retroarch/shaders/retroarch.glslp"
RetroArch [INFO] :: Video @ 320x240
RetroArch [ERROR] :: Failed to connect to Wayland server.
RetroArch [INFO] :: Found GL context: x
RetroArch [INFO] :: Detecting screen resolution 1680x1050.
RetroArch [INFO] :: [GLX]: X = 0, Y = 0, W = 320, H = 240.
RetroArch [INFO] :: [GLX]: Found swap function: glXSwapIntervalMESA.
RetroArch [INFO] :: [GLX]: glXSwapInterval(1)
RetroArch [INFO] :: [GL]: Vendor: nouveau, Renderer: Gallium 0.4 on NV46.
RetroArch [INFO] :: [GL]: Version: 2.1 Mesa 10.3.2.
RetroArch [INFO] :: Querying GL extension: ARB_ES2_compatibility => exists
RetroArch [INFO] :: Querying GL extension: ARB_texture_non_power_of_two => exists
RetroArch [INFO] :: Querying GL extension: ARB_fragment_program => exists
RetroArch [INFO] :: Querying GL extension: ARB_sync => exists
RetroArch [INFO] :: [GL]: Using ARB_sync to reduce latency.
RetroArch [INFO] :: Querying GL extension: ARB_texture_float => doesn't exist
RetroArch [INFO] :: Querying GL extension: EXT_texture_sRGB => exists
RetroArch [INFO] :: Querying GL extension: ARB_framebuffer_sRGB => doesn't exist
RetroArch [INFO] :: GL: Using resolution 836x1011
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 4 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: [GL]: Using GL_RGB565 for texture uploads.
[... 190 lines cut ...]
#3  driver_ctl (state=<optimized out>, data=<optimized out>) at driver.c:444
#4  0x000000000042a897 in driver_ctl (state=state@entry=RARCH_DRIVER_CTL_UNINIT_ALL, data=data@entry=0x0) at driver.c:450
#5  0x0000000000411810 in rarch_ctl (state=state@entry=RARCH_CTL_DESTROY, data=data@entry=0x0) at retroarch.c:1384
#6  0x000000000040f70d in main_exit (args=0x0) at frontend/frontend.c:80
#7  rarch_main (argc=<optimized out>, argv=<optimized out>, data=0x0) at frontend/frontend.c:170
#8  0x00007ffff2affb45 in __libc_start_main (main=0x40cda0 <main>, argc=1, argv=0x7fffffffe2f8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, 
    stack_end=0x7fffffffe2e8) at libc-start.c:287
#9  0x000000000040cdd0 in _start ()
(gdb) 

ARB_framebuffer_object is not implemented for [NV40](https://nouveau.freedesktop.org/wiki/CodeNames/#NV40) graphic card family.

The error comes from because video_driver_data is false
https://github.com/libretro/RetroArch/blob/master/gfx/video_driver.c#L715

Because in gl_init function, gl_init_hw_render return false
https://github.com/libretro/RetroArch/blob/master/gfx/drivers/gl.c#L2840

Because in gl_init_hw_render function, gl_check_fbo_proc return false
https://github.com/libretro/RetroArch/blob/master/gfx/drivers/gl.c#L784

If I recompile RetroArch with --disable-fbo I got other errors.

At first, is it possible to use [GL_EXT_framebuffer_object instead](https://people.freedesktop.org/~imirkin/glxinfo/glxinfo.html#p=compat&v=Mesa%2010.6.0) of GL_ARB_framebuffer_object ? 
Because it supported by "old" cards or you must use GL_ARB_framebuffer_object.

gouchi, 2016-05-11:
This [commit](https://github.com/libretro/RetroArch/commit/c9d3936ee53cc123bafd72b05e8ec70e12efbb10)  fix the issue.

Alexandre-Garcia, 2016-05-12:
Ok ... we have progress ...
Now I can play nintendo 64 games in retroarch, but the graphics are wrong.
[image]
[image]
Besides this there is a loss of performance (which is not even a problem for me), because in mupen64plus can play in 640x480, however the mupen64plus-libretro for the same performance, I have to put in 320x240.

twinaphex, 2016-05-12:
Several things you are conflating here -

1 - Performance - will probably be equivalent or even better when new_dynarec is updated to the latest version (the one in libretro is lagging behind a bit)

2 - The issues you see here - which plugin are you using? Glide64? It is usually totally GPU-specific whenever issues like this happen. Things render fine on Intel HD GPUs and Tegra which is what I predominantly tested it on.

Anyway, big changes are in the pipeline for mupen64 libretro that will put it well ahead of anything else out there.

Alexandre-Garcia, 2016-05-12:
I tested it with all plugins (glide64, rice, angry lion ..) and all had similar results.
Still using the Nvidia 6200 LE, with the nouveau driver.
GLX version: 1.4
OpenGL vendor string: nouveau
OpenGL renderer string: Gallium 0.4 on NV44
OpenGL version string: 2.1 Mesa 11.0.2
OpenGL shading language version string: 1.20

gouchi, 2016-05-14:
@Alexandre-Garcia can you try with this [image](https://nouveau.pmoreau.org) containing latest graphic stack (libdrm, mesa, nouveau) to see if you reproduce your issue ?

Alexandre-Garcia, 2016-05-23:
@gouchi 
Since I had no experience with ArchLinux, it took me a while to be able to install the OS and dependencies and compile Retroarch. In the end the result was the same as that found in Ubuntu.
I believe it no more sense to continue treating this issue, because this 6200 LE is really a limited GPU. As I said earlier, I was initially using a 9500 GT,  which works perfectly, with the exception of S-VIDEO output.
But if you want to continue to test I am disposal.

@Kivutar 
Still have the intention to make changes that will to allow building images with x11(with nvidia proprietary driver) and drm/kms easily?

Kivutar, 2016-05-26:
I made those changes, but now our X11 packages are outdated. We need to fix them. You can open another issue about that.

closed 2016-05-26 by Kivutar

Alexandre-Garcia, 2016-05-27:
In fact in the meantime the already got to compile Lakka, updating the X11 and Nvidia drivers (the 340.76 version was giving error while compiling, probably because of kernel updates, so I put the 340.96 and it worked).
In this way I thank all who were involved.

Kivutar, 2016-05-27:
Great, feel free to send a PR with the updated X11 and nvidia stuff. If you think it can be useful.