libretro/Lakka #241: Problems using two wireless Xbox360 controllers at the same time

derechtedude, 2015-11-22:
Hey everybody, wanted to connect two wireless xbox360 Controllers via one wireless Receiver.
I configured both but after calibrating the second one the first one stops two work in the menu and i can only continue with the second Controller.
Is it possible to use two Controllers at the same time?
cheers mat

oblitum, 2015-11-28:
@derechtedude I've used two wireless xbox 360 with lakka without issues (linux/cubox-i).

boonediggity, 2015-12-09:
I am having similar issues. I have used them both at the same time with the same wireless receiver so I know it works. What made the biggest difference was updating to the latest build (at least 20151124 version) there were a lot of controller fixes for me. 

My issue now is that I can only get one controller to function and control the menu at a time. When I change controller settings via the gui menu I can usually get controller 2 to respond but then controller 1 stops responding. I can then go into a game and in some cases, I am controlling both cursors (controllers) at the same time via one physical controller. For example, playing N64 Mario Kart and selecting which player avatar I want, the P1 and P2 selectors mover in unison around then screen. I also notice that when selecting menu options in game it's like the direction keys are double tapped and always skip over an item. It's been frustrating. 

I have also tried having two XBox 360 controllers and two PS3 controllers at the same time. That works for a little while but as soon as Retroarch restarts things get really messed up and I haven't figured out how to fix it or what even broke. For now I'll install the latest update and try some more troubleshooting.

hizzlekizzle, 2016-03-18:
This is typically an issue with the device indexes in settings > input for both players getting assigned to the same physical device. You should be able to go into the p1 and p2 input settings and just make sure they're using the different devices. Only P1 should be able to control the menu.

OGWillikers, 2016-05-31:
I can't reproduce the issue on the latest build.  Tried lots of cores, pulling batteries, powering controllers back on at as close to the same time as possible.  First controller to connect always works as player 1 and controls the menu.

Kivutar, 2016-05-31:
So is it fixed?