libretro/Lakka #248: ROM Scanning Issues

kryptalivian, 2015-12-11:
On Raspberry Pi 1- Lakka will scan my directory and scan all 432 roms but it will only display a portion. It will not show Genesis and most of my Super Nintendo Games. I tested and can confirm the roms work perfect on other distros. :-(

fr500, 2015-12-11:
Other distros or other emulators?

Kivutar, 2015-12-11:
Use the No-Intro ROM set, and all will scan properly.

kryptalivian, 2015-12-11:
They loaded properly with Retropie on my RPI2. Kivutar - would this be what your talking about? https://archive.org/download/No-Intro-Collection_2015-03-03

Kivutar, 2015-12-11:
RetroPie took a different approach to scanning : they allow all dumps, but a percentage of them will give false positive. We support No-Intro DATs that contains also tested ROMs, and 100% should scan.

Your link is OK (except for NES roms that will lack header, choose the 20140825 set for NES).

However, I'm not sure about the legality of linking a ROM repository here. Users are supposed to use only ROMs of cartridges they really own. So please edit your message and hide the link.

Alternatively, we provide a RetroPie like way to produce playlists, documented here https://github.com/libretro/Lakka/wiki/Playlists

If for some reason you want to keep your current set of ROMs, then use this alternative way.

Good luck :)

molsky, 2015-12-11:
I have problem when doing scanning again it will add existing items again to ROM list and therefore soon there will be a lot of duplicate items in list if you do this 'Add ROMs -> scan folders' cycle many times. 
Raspberry Pi 2

Kivutar, 2015-12-11:
You're right. This is a know bug. For now the scanning is an additive process. We could implement a deduplication mechanism, but this would make scanning slow...
While we are looking for a solution, please delete your playlists and scan again.

lollo78, 2015-12-13:
Hi Kivutar, could you please include an "external scapers" function (TheGamesDB, GameFAQ and so on) to catch the original game covers? 
And please, could you include a function to autoadapt the boxart aspect-ratio according the images loaded? I try to do this manually (impossible to do for a full romset: for this reason my scraper request) and the SNES cover appear stretched.
Also, the no-Intro thumbs available on the Web are screenshot of the game, not covers.
Thanks a lot for all your hard work. :*

Kivutar, 2015-12-13:
Instead of letting every user scrape, we should just scrape one time ourselves and provide packs. Sadly I'm busy with more technical stuff at the moment. If one of you want to start this collection effort, it would be a very appreciated contribution for our community and other communities.

Please open another issue about the aspect ratio, with version and screenshot. I don't see this bug with the release.

Kivutar, 2016-01-04:
I close this thread, since it was about using no-intro rom sets.

closed 2016-01-04 by Kivutar

MasterBoss83, 2016-12-07:
how the heck do i even use this script? man i am so lost...all i want is a freaking gaming console that reads all of my imported roms files :(

Kivutar, 2016-12-07:
If you can't get the script to work, then just load the games using Load Content Detect Core in the main menu.

cudencuden, 2016-12-07:
asking for free help is not about throwing a fit and making demands.

MasterBoss83, 2016-12-07:
how about throwing a fit on deez nuts cudencuden? punk ass. i wasn't throwing a fit  whatsoever lol. so, kindly run along and blow me cudencuden.

Thanks for the info Kivutar, i got it worked out a different way. 

Whoever updates lakka, might want to update the CRC check file or whatever list roms checks against..cause every rom i own works properly and works in lakka as well if i go through loading them all through a different method.

gzuspr, 2017-02-18:
Ok I'm finally throwing my hat in and asking for some advice. I've looked through countless hours of videos and read plenty of information. I'm having a problem downloading roms  into my Lakka for the rpi3. I've have gotten as far as downloading rom zip files from online sources, created a file and transfer to the lakka through network share. After rebooting lakka and going to scan directory or scan file I see my files. So at least I'm making contact but I don't know if the files I've copied are wrong or have improperly transferred them because every time I reboot again. I get bubkiss.... pls help or direct me to the right place, starting to feel like an idiot. Btw yes I've use the no-intro file 16g zip file to... nothing