libretro/Lakka #251: PS3 controller autoconfig not working if it is closed before to launch start

lollo78, 2015-12-13:
Hi guys, 
with the Retroarch bin inside the latest stable build of lakka I have an issue with PS3 controller.
Autoconfig works only if I power up the controller before to launch retroarch.
If the controller is closed and retroarch is launcher it doesn't work.
At the same time, if I close the controller and retroarch is launched, it loose the connection. if I power up controller it doesn't work.

I don't see anymore the overlay notification when controller is connected or disconnected.

lollo78, 2015-12-15:
Any news about this? 
I restored an old Retroarch bin file and all works fine (but, of course there is the previous display bug that causing icons to stack at the bottom of the screen).
With the new retroarch bin (with the last stable release and the newest nightly build) I have the issue above.

I use a Raspberry Pi 2.

Kivutar, 2015-12-16:
I was able to reproduce this problem. I will try to fix it as soon as possible.

Kivutar, 2015-12-16:
I need to know the exact version of Lakka you're using please.

lollo78, 2015-12-16:
This happen for me with 
http://sources.lakka.tv/nightly/RPi2.arm/Lakka-RPi2.arm-devel-20151207161710-r21226-gaa8871a.img.gz

and all the nighlty build released after the stable release of 7 december 2015.

Doesn't happen with the previous releases.

Kivutar, 2015-12-16:
Thanks, it's a retroarch bug, I've found the guilty commit. We may fix it soon.

lollo78, 2015-12-16:
Thank you very much

Kivutar, 2016-01-09:
This is fixed now.

closed 2016-01-09 by Kivutar