libretro/Lakka #252: Duplicated content history

lollo78, 2015-12-13:
Hi again guys.
The "History tab" inside Lakka is very nice, but on my configuration into the Lakka GUI I see also a tab, with a black square (because no icon is available for it), called "content_history".
It's the same of History tab, and they use the same file: content_history.lpl
There's a way to fix it for use only History tab?
Thx

Kivutar, 2015-12-13:
Hello, this doesn't happen on my setup. Can you remove this content_history.lpl and use only the content_history.rpl located in .config/retroarch?

Kivutar, 2016-01-09:
I think this was due to having two content_history files in your paths. On a fresh install, I can't reproduce this problem. It may affect old systems after an upgrade. But there is nothing we can do to prevent this kind of bugs. I close it for now.

closed 2016-01-09 by Kivutar