libretro/Lakka #253: Issues logging into retroachievements

kryptalivian, 2015-12-14:
I go to user-> get to the portion where you click username and it pops up on my screen saying Username. When I type, it doesn't show any of the characters you've typed. Is there a way to fix this or maybe a file I can edit and input the login credentials myself? Sorry about all the questions lately!

One last thing, It's a Raspberry Pi 2 and the image I used was today's image on the site (12/13/2015)

Kivutar, 2015-12-15:
I've never tried this feature yet. You can try to enter the values manually by editing retroarch.cfg. Refer to out documentation on how to do this :)

kryptalivian, 2015-12-15:
I'm sorry, but I've done some searching and not sure what I need to enter in the file. I'm at the file and have it pulled up but im not sure what values need entered before I put in my credentials.

kryptalivian, 2015-12-17:
Finally got it going! :-)

closed 2015-12-17 by kryptalivian

Kivutar, 2015-12-17:
Nice, if you want to write a page about this in our wiki, it is open to you ;)

kryptalivian, 2015-12-17:
I'd be more than happy to! :-)

Kivutar, 2015-12-17:
Great, I created a page here https://github.com/libretro/Lakka/wiki/Achievements

You can start by explaining what is this feature about, give a link to the retro achievments website, and explain how to use it in lakka.

I really appreciate when users can contribute to the documentation :D

kryptalivian, 2015-12-17:
It's been submitted. Feel free to make changes.

Kivutar, 2015-12-18:
Thanks. Your instructions are clear.