libretro/Lakka #254: PPSSPP Segfault

lollo78, 2015-12-15:
Hi, ppsspp_libretro.so down't work for me.
If I try to launch from shell I see the error Segmentation fault.

lollo78, 2015-12-15:
I forgot to specify that I use a Raspberry Pi 2.

Kivutar, 2015-12-16:
Are you using the latest release? I tested ppsspp on rpi2 myself so it should work.

lollo78, 2015-12-16:
Yes, the latest

Kivutar, 2015-12-24:
Hi @lollo78 
I just tested the latest release with PPSSPP on the RPi2, with the Flow game. And it worked.

Kivutar, 2016-01-04:
You tried with many ROMs? I'm not able to reproduce this bug.

lollo78, 2016-01-15:
I tried with other roms (also in cso format) and I can confirm that PPSSPP works. 
But on Rpi2 it's very very slow (what a pity)

Kivutar, 2016-01-16:
Try to enable all the speed hacks. And choose a light game like FF3. Anyway, we can close this now since it doesn't segfault anymore.

If you want to improve performance, you can declare a bug in PPSSPP libretro repo. Ask them to fix the non buffered rendering. This option is broken, and can improve performance.

closed 2016-01-16 by Kivutar

lollo78, 2016-01-19:
Ok, Thx. But this feature is broken only for rpi2?

Kivutar, 2016-01-19:
No, it's broken in all platforms for ppsspp-libretro.

lollo78, 2016-01-22:
OK, I opened an issue on the PPSSP libretro core github.
I really hope for a fix.
It could be so nice...