libretro/Lakka #255: Controller Config Map Question

zombie4000, 2015-12-16:
All,

I'm working on configuring the controller below for N64.

ID 0079:0002 DragonRise Inc.

I'm curious if there is a map of a actual n64 controller and it shows the input_#_btn for that button.  This way I can map my controller numbers to the correct button.  I have most completed, having issues with "Z" button.  Not sure how to define the input_

Thanks.

hizzlekizzle, 2015-12-16:
L2

closed 2015-12-18 by Kivutar

rensa, 2016-12-18:
Sorry, could you provide some additional clarification on this? I understand that this map has the Lakka control slot on the left and the N64 core's input on the right. So the button I map to L2 in Lakka will actually invoke the Z-trigger in an N64 game. Great. But what about A, B and the C-buttons? Why are A and B in Lakka described as "A Button (C-Right)" and  "B Button (C-Down)" in the core? Ther are six N64 controller buttons here: A, B and four C-buttons. If I map A and B in Lakka to C-right and C-down on my N64 USB pad, it completely mucks up the controls in actual N64 games (even if this scheme works great for other systems with more traditional layouts).

Which inputs does Lakka give to the N64 cores for these six buttons (A, B, C-up/down/left/right)?

Kivutar, 2016-12-19:
I think the mapping is this https://wiki.libretro.com/index.php?title=Mupen64Plus