libretro/Lakka #256: Install Lakka on Mac OS X system

Fransebas, 2015-12-18:
I've got a eMac wich has 640 MB of RAM, and I think that would be a good idea to install it Lakka, but when I turn it on with the usb, It starts normally (on Mac OS), I believe that I'd need to open the BIOS but mac instead have Open Firmware, I'm sure it must be a way to allow the usb from there but I don't know how to use Open Firmware, my other option is to install windows but prefer not to do that.

twinaphex, 2015-12-18:
Well I don't think OpenElec (the distro on which Lakka is based on) ever cared about Macs. 

That being said, I think it is a good idea, not sure about the practicalities of it though. I have a bunch of old PPC Macs that I think it would be great to repurpose as Lakka machines, and I imagine the same holds true for Intel Macs too.

Guess I'd have to discuss it with @kivutar.

Fransebas, 2015-12-18:
I found a lot of bootloaders that support ppc in unix and usb-boot, for example I found the LILO bootloader, but I don't know anything of how to use it, can somebody tell how to install any of the bootloader, like LILO, and how to use it to install laka please.
This is the page of the list of bootloaders for unix https://en.wikipedia.org/wiki/Comparison_of_boot_loaders

Kivutar, 2015-12-19:
Hello, I use Lakka on my Macbook Air on a weekly basis. I just hit Alt during the bootloader to select my USB key.

Fransebas, 2015-12-19:
Yes @Kivutar  I'd install it easily on a intel base computer, I actualy install Linux on one using rEFInd but in this case is an ppc computer, and don't support usb boot, and neither can install windows to use the BIOS, I use the Open Firmware but does not recognize it. (https://lifedigital2010.wordpress.com/2011/04/03/how-to-install-mac-osx-from-usb-on-powerbook-g4/)

When I press alt only apears the OS MAC HD

Kivutar, 2015-12-20:
Powerbook G4 are PowerPCs, we don't support this CPU architecture yet.

Fransebas, 2015-12-21:
Even if I install GRUB on the USB? If so what other emulator you recomend ?, I was traying to install MAME but is way more dificul to install (or that I had guess). anyway thanks @Kivutar

hizzlekizzle, 2015-12-21:
If you can install any other Linux on there, you can just compile and install RetroArch onto it from there. Lakka is a more cohesive, convenient package, of course, but the end result is similar.

Kivutar, 2015-12-24:
RetroArch works well on Power PC, I suggest you to use it directly. We do plan to make a port of Lakka on PPC but not this month.

closed 2015-12-24 by Kivutar