libretro/Lakka #257: None of the MAME versions work on RPI2

duduke, 2015-12-18:
Hi,

Using: http://buildbot.libretro.com/nightly/linux/armhf/latest/ as a URL for the core updater, and trying to use MAME, MAME2014 causes segmentation fault.

Any idea?

Kivutar, 2016-01-09:
A lot of ARM cores from the libretro buildbot are not working in Lakka. I also noticed some problems with mednafen-psx on XU3.

duduke, 2016-01-10:
Most of the cores in /nightly/linux/armhf/latest/ haven't been built since Oct-2015.
Maybe they're just incompatible with the latest RetroArch?
Also, who's in charge of building those cores?

Kivutar, 2016-01-12:
@fr500 (radius on irc) is in charge of the buildbot iirc.

duduke, 2016-01-14:
I've created a new issue at retroarch-buildbot: 
https://github.com/libretro/libretro-buildbot/issues/5

Kivutar, 2016-05-01:
I close this issue, since it is a buildbot + mame problem. And I don't plan to officially support mame on arm board given the size of the binary.

closed 2016-05-01 by Kivutar