libretro/Lakka #259: Enhancement Request: Restart option

duduke, 2015-12-19:
Hi,

I've been multi-booting Lakka and OpenELEC on my RPI2 (via NOOBS).
Is it possible to have a "Restart" option so that I can get back to OpenELEC without power-cycling the Pi (unplug, replug power cable)?

Thanks.

Kivutar, 2016-01-09:
I agree. It is needed for the updater anyway.

duduke, 2016-01-11:
I've created an enhancement request for this in RetroArch: https://github.com/libretro/RetroArch/issues/2635.

Also, for the meantime (as I don't know how to add new RetroArch sub-menus). The following code can be changed to enable LAKKA to reboot instead of shutdown: 
command_event.c (lines 1308-1311):
1308        runloop_msg_queue_push("Shutting down...", 1, 180, true);
1309	         rarch_ctl(RARCH_CTL_FORCE_QUIT, NULL);
1310	         system("shutdown -P now");
1311	#endif

Change to:
1308        runloop_msg_queue_push("Shutting down...", 1, 180, true);
1309	         rarch_ctl(RARCH_CTL_FORCE_QUIT, NULL);
1310	         system("shutdown -r now");
1311	#endif

duduke, 2016-01-19:
I've forked RetroArch and added the following changes to add a reboot menu:
https://github.com/duduke/RetroArch/commit/7de7c98b27ae5865dcdb3f78391bf4596b5f7af1
https://github.com/duduke/RetroArch/commit/ffffc50401b9f80baec33405d8a37ff3bc72b278
https://github.com/duduke/RetroArch/commit/b284749016d1e442676b718acc15923293708ee5
https://github.com/duduke/RetroArch/commit/eecd14086b897fd435a1dd81770545c437ce5db6
But instead of a "Reboot" menu item, I'm getting 2 Shutdown Items which both shutdown instead of restart.
mind taking a look and telling me where I got it wrong?

Thanks.

Kivutar, 2016-01-19:
This looks perfect to me. I don't see where the patch is wrong...
Can you tell me the value of your two hashes?

duduke, 2016-01-19:
#define MENU_LABEL_REBOOT                                                      0x1b353023U
#define MENU_LABEL_VALUE_REBOOT                                                0xcab9e59fU
Aside from menu/menu_hash.h, is there anywhere else those hashes (which I generated via djb2) need to be?

Kivutar, 2016-01-19:
One of them should be "Reboot", 0xce815750U

duduke, 2016-01-19:
So, if MENU_LABEL_VALUE_REBOOT should be a hash of "Reboot"
What should MENU_LABEL_REBOOT should be?

duduke, 2016-01-19:
What is MENU_LABEL_VALUE_SHUTDOWN points to?

Kivutar, 2016-01-19:
I points to "Shutdown"

Kivutar, 2016-01-19:
No sorry, MENU_LABEL_VALUE_SHUTDOWN points to "Shutdown"

duduke, 2016-01-19:
I've changed MENU_LABEL_VALUE_REBOOT to 0xce815750U (which is a hash of Reboot).
It still shows 2 Shutdown menuitems.
I still don't know what should be the hash value of MENU_LABEL_REBOOT...
I've followed your original commit to add the new menuitem, i'm just missing what should be the correct hash value...

Kivutar, 2016-01-19:
That was my commit to add shutdown: https://github.com/libretro/RetroArch/commit/2b376ce3d551fb2b649743d38ba818be5109b3ba

duduke, 2016-01-19:
That commit is what I've used as the basis for my changes.

#define MENU_LABEL_SHUTDOWN                                                    0xfc460361U
#define MENU_LABEL_VALUE_SHUTDOWN                                              0x740b6741U
So 0x740b6741U translates to "Shutdown", what does 0xfc460361U translates to?

Kivutar, 2016-01-19:
It translates to "shutdown".

I'm cloning your fork, and compiling it.

duduke, 2016-01-19:
I've now updated my fork for the new hash values:
MENU_LABEL_REBOOT point to the hash value of reboot
MENU_LABEL_VALUE_REBOOT points to the has value of Reboot

Kivutar, 2016-01-19:
Not working... I was able to reproduce the bug. I'm trying to find the source of the problem... For now, I see no difference between what you and I did.

Kivutar, 2016-01-19:
Found it, even my commit was wrong!

    diff --git a/menu/intl/menu_hash_us.c b/menu/intl/menu_hash_us.c
    index 85d64a8..25484b1 100644
    --- a/menu/intl/menu_hash_us.c
    +++ b/menu/intl/menu_hash_us.c
    @@ -572,6 +572,10 @@ static const char *menu_hash_to_str_us_label(uint32_t hash)
              return "settings";
           case MENU_LABEL_QUIT_RETROARCH:
              return "quit_retroarch";
    +      case MENU_LABEL_SHUTDOWN:
    +         return "shutdown";
    +      case MENU_LABEL_REBOOT:
    +         return "reboot";
           case MENU_LABEL_HELP:
              return "help";
           case MENU_LABEL_SAVE_NEW_CONFIG:

duduke, 2016-01-19:
I will modify accordingly and test.

Kivutar, 2016-01-19:
Anyway, thanks for this work. Tell me when you PR gets merged by @twinaphex and I will upgrade RetroArch in Lakka.

duduke, 2016-01-19:
It Works!
Thanks a bunch.
I've created a PR: https://github.com/libretro/RetroArch/pull/2660

duduke, 2016-01-22:
Merged to RetroArch, thanks a lot.

Kivutar, 2016-02-02:
Your change landed in Lakka. Thanks again.

closed 2016-02-02 by Kivutar