libretro/Lakka #260: FC30 PRO will not connect

bdcoding, 2015-12-19:
Hi there,

i'm new.
i can not connect to my FC30 Pro, can you help me?

Thans a lot!

[bluetooth]# pair 01:03:87:A6:60:0E
Attempting to pair with 01:03:87:A6:60:0E
[CHG] Device 01:03:87:A6:60:0E Connected: yes
[CHG] Device 01:03:87:A6:60:0E Modalias: usb:v3810p0009d0100
[CHG] Device 01:03:87:A6:60:0E UUIDs:
        00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 01:03:87:A6:60:0E Paired: yes
Pairing successful
[CHG] Device 01:03:87:A6:60:0E Connected: no
[bluetooth]# trust 01:03:87:A6:60:0E
[bluetooth]# connect 01:03:87:A6:60:0E
Attempting to connect to 01:03:87:A6:60:0E
[CHG] Device 01:03:87:A6:60:0E Trusted: yes
Changing 01:03:87:A6:60:0E trust succeeded
[CHG] Device 01:03:87:A6:60:0E Connected: yes
Failed to connect: org.bluez.Error.NotAvailable
[CHG] Device 01:03:87:A6:60:0E Connected: no
[bluetooth]# info 01:03:87:A6:60:0E
Device 01:03:87:A6:60:0E
        Name: 8Bitdo FC30 Pro
        Alias: 8Bitdo FC30 Pro
        Class: 0x002540
        Icon: input-keyboard
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: no
        LegacyPairing: yes
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v3810p0009d0100

Gamepad FW is 1.66 (actual release)

bdcoding, 2015-12-19:
In Mode 1 i can connect but i can not use it. Lakka wont make a move!

Kivutar, 2016-01-28:
I'm uploading new nightly builds with updated config files for the 8bitdo joypads. Make sure to upgrade your pads to the latest firmware too. Can you give it a try please?

bdcoding, 2016-02-17:
Thanks for the update! It works as well!!!!

closed 2016-02-18 by Kivutar