libretro/Lakka #261: XU4 - boot.ini display settings

synportack24, 2015-12-19:
I'm having problems setting the resolution in the boot.ini
For the official ODROID images I always set the following in my boot.ini:
setenv videoconfig "drm_kms_helper.edid_firmware=edid/1920x1080.bin"

For the Lakka I set the equivalent:
setenv video_output "drm_kms_helper.edid_firmware=edid/1920x1080.bin" : 1080p (Use system EDID)

however, still nothing gets displayed, I'm I missing something?

Kivutar, 2015-12-20:
What about removing this part?

     : 1080p (Use system EDID)

Kivutar, 2016-01-01:
If this is still not fixed and affects all XU4 distros, can you, please, declare this bug on the Odroid forums or whatever issue tracker they use?

Kivutar, 2016-01-29:
No reply for one month. I close it for now. Feel free to reopen it with additional information.

closed 2016-01-29 by Kivutar