libretro/Lakka #262: Dynamic wallpaper broken with Black backgrounds

lollo78, 2015-12-20:
If I scroll quickly left and right the emulators, sometimes the dynamic wallpaper function loses the backgrounds and shows only a black background in all the emulators. I have to quit and restart retroarch to see again the bichromatic wallpaper.

Also, DOOM icon and Quake1 monochromatic icons are missing, like some bichromatic wallpapers. More, some wallpapers use the same colors and icons for different emulators: this create some confusion.
I'm not a graphic, but waiting an expert fix I created this package:
https://dl.dropboxusercontent.com/u/30093753/icons%20and%20wallpapers.zip
Hope this help

Kivutar, 2015-12-28:
@maister fixed a leak in the wallpapers. Can you check again with the latest dev build please?

@lollo78 Do you know how to make pull requests? Could you send a pull request to the retroarch-assets repository with your missing wallpapers please?

lollo78, 2015-12-30:
The black background issue is present also into the latest Nightly build -30 december-.
I'll try to create the pull request forma the missing icons and wallpapers

lollo78, 2016-01-15:
@Kivutar I Just done the pull request. I hope I did all fine. I never did it before ;)

Kivutar, 2016-01-16:
I merged it. Thanks. Does this black wallpaper bug still happen?

lollo78, 2016-01-16:
Thanks a lot for the merge.
I have the "black wallpaper" issue also with the last Lakka nightly build:
Lakka-RPi2.arm-devel-20160109121338-r21276-g4dadcab

You can see a video of the issue here:
https://dl.dropboxusercontent.com/u/30093753/VID_20160116_231440.3gp
(sorry for the bad quality).

lollo78, 2016-01-20:
I added Colecovision icons and wallpaper and sent a pull request.
Bye

Kivutar, 2016-01-20:
Thanks for your icons and wallpaper, they are well done.

I saw your video, and tried to reproduce the bug on RPi2, but wasn't able to reproduce it yet.

lollo78, 2016-01-21:
Thanks a lot and thankt to you for all your hard work.
Now I understand better how to make icons in SVG and wallpapers.
When I have some time, I'll do again my previous commits with a better quality. 

The bug is in retroarch, maybe it appears only when there are a lot of platforms and boxarts.

lollo78, 2016-03-14:
Any news about this issue?
I also noticed another strange behavior... and I'm starting to think that maybe this Retroarch issue is related on managing large playlist file. You know that I scanned a lot of platforms collection (from No-Intro 2016 fixed) and now I have a lot of .lpl files, some of which are about some Megabyte. I noticed that if I quickly scroll left or right, I see a lag when retroarch arrive at platform that use a big playlist file. And sometimes (if you insist to scroll when lag appear) this lag freeze Retroarch (reboot required).

A way to reproduce the bug (I hope... or means that I am mad): scan a lot of no-intro platforms (complete, with all the platform games - not only a selection-), especially those that have thousands of files, and use also all the boxart images per-patform (from our repo project ;)).
After this, don't be gently. Don't press one by one the direction key on the controller to switch among platform (left and right) or games (up and down). Don't give retroarch the time to load data.
Try this kind of "stress test". Hold down your finger (for a long time) on the right key or down key (inside SNES for example). After a while, you'll see the issue. 
If not, I'll call the psychiatry department :D

Of course, I am speaking about Raspberry Pi 2/3. Never tested on other systems.

Kivutar, 2016-03-14:
You're right. This bug also affects PC.

lollo78, 2016-03-14:
I'm happy to discover that I'm not mad :smile:

lollo78, 2016-03-14:
Maybe, preloading all the .lpl in RAM on boot could help?

Kivutar, 2016-03-14:
That is a good idea. But then, it will become difficult to have dynamic playlists. Let start a thread about this on the retroarch issue tracker, please. I need some advice from good C coders for this problem.

lollo78, 2016-03-14:
I repost the bug here:
https://github.com/libretro/RetroArch/issues/2791

Kivutar, 2016-11-15:
Try to increase the `gpu_mem` in `config.txt`.

`gpu_mem_1024=512` on a RPi3.

Kivutar, 2016-11-15:
It fixes nothing...

jcreznor, 2017-01-26:
In the end someone fix this bug, because I have that same problem