libretro/Lakka #263: [request] counter roms for each platform

lollo78, 2015-12-21:
Hi, it's possibile to add a counter that show (on the bottom right, or up left near the platform name) the number of roms inside each platform menu?
Thx

lollo78, 2016-02-27:
Hi kivutar, could you add this feature?
It could be very usefull to check quickly if all the No-Intro Roms are scanned from the .dat file inside our boxart project.

Kivutar, 2016-02-27:
I will add it yes, when we have a way to display more than two font sizes on a screen.

lollo78, 2016-02-27:
Could you add it like the "fps_show" features (that show frame rate in the bottom left corner). And, in the same way, give to the users the possibility to activate it or disable it inside the menu. In effect, this feature could be usefull for checks/tests and so on, but not for all the people.

Kivutar, 2016-02-28:
Ah, sure, like this it's easier.

lollo78, 2016-02-28:
:kissing: 
I'm waiting...
:smiley: