libretro/Lakka #264: Wireless Dualshock 4 connected but not working

GabrieleV, 2015-12-26:
Hello,
I've successfully paired and connected a Sony Wireless Dualshock 4 controller (MODEL CUH-ZCT1E printed on the back).

The device is paired,trusted and connected, as shown by bluetoothctl:
```
[bluetooth]# info 00:01:6C:BF:07:09
Device 00:01:6C:BF:07:09
	Name: Wireless Controller
	Alias: Wireless Controller
	Class: 0x002508
	Icon: input-gaming
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v054Cp05C4d0100
```
Unfortunately, the controller doesent' work.
It works correctly connected with a simple micro usb cable

I'm running latest version, after
   lakka-update

```
Lakka (unofficial) Version: devel-20151225181118-r21254-g2c76737
Lakka git: 2c7673701f99c67369cd04112dd2e0ebb34034c3
```
What's wrong ?

Thank you !

Kivutar, 2015-12-26:
Hello, can you try to plug a keyboard and manually bind the keys of your DS4 in the Input Settings?

If it works, you will then be able to export the config file, and give it to us, so we can do a diff and know what's going wrong.

Thanks.

GabrieleV, 2015-12-26:
Don't know ho to make a manual binding in the gui :-(
I've tried a little investigation.

Here is the log lines of:
```journalctl --unit retroarch --follow```

```
Dec 26 13:24:48 Lakka retroarch[570]: RetroArch [INFO] :: [udev]: Hotplug add: /dev/input/js0.
Dec 26 13:24:48 Lakka retroarch[570]: RetroArch [INFO] :: [udev]: Hotplug add: /dev/input/event7.
Dec 26 13:24:48 Lakka retroarch[570]: RetroArch [INFO] :: [udev]: Plugged pad: Wireless Controller (0a12:0001) on port #0.
Dec 26 13:24:48 Lakka retroarch[570]: RetroArch [INFO] :: Autodetect: 98 profiles found
Dec 26 13:24:48 Lakka retroarch[570]: RetroArch [INFO] :: Autodetect: no profiles found for Wireless Controller (2578/1)RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event7) supports force feedback.
Dec 26 13:24:48 Lakka retroarch[570]: RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event7) supports 16 force feedback effects.
```

So i tried to create a copy of the wired version
```
   mkdir /storage/joypads/udev
   cp /etc/retroarch-joypad-autoconfig/udev/Sony-PlayStation4-DualShock4-Controller.cfg /storage/joypads/udev/Sony-PlayStation4-DualShock4-Controller-Bluetooth.cfg
```

And edited using the Wireless IDs:
```vi /storage/joypads/udev/Sony-PlayStation4-DualShock4-Controller-Bluetooth.cfg```

```
input_device = "Sony Dualshock 4 BT"
input_driver = "udev"
input_vendor_id  = 2578
input_product_id = 0001
input_b_btn = "1"
input_y_btn = "0"
input_select_btn = "8"
input_start_btn = "9"
input_up_btn = "h0up"
input_down_btn = "h0down"
input_left_btn = "h0left"
input_right_btn = "h0right"
input_a_btn = "2"
input_x_btn = "3"
input_l_btn = "4"
input_r_btn = "5"
input_l2_btn = "6"
input_r2_btn = "7"
input_l3_btn = "10"
input_r3_btn = "11"
input_l_x_plus_axis = "+0"
input_l_x_minus_axis = "-0"
input_l_y_plus_axis = "+1"
input_l_y_minus_axis = "-1"
input_r_x_plus_axis = "+2"
input_r_x_minus_axis = "-2"
input_r_y_plus_axis = "+5"
input_r_y_minus_axis = "-5"
input_menu_toggle_btn = "12"

input_b_btn_label = "Cross"
input_y_btn_label = "Square"
input_select_btn_label = "Share"
input_start_btn_label = "Options"
input_up_btn_label = "D-Pad Up"
input_down_btn_label = "D-Pad Down"
input_left_btn_label = "D-Pad Left"
input_right_btn_label = "D-Pad Right"
input_a_btn_label = "Circle"
input_x_btn_label = "Triangle"
input_l_btn_label = "L1"
input_r_btn_label = "R1"
input_l2_btn_label = "L2"
input_r2_btn_label = "R2"
input_l3_btn_label = "L3"
input_r3_btn_label = "R3"
input_l_x_plus_axis_label = "Left Analog Right"
input_l_x_minus_axis_label = "Left Analog Left"
input_l_y_plus_axis_label = "Left Analog Down"
input_l_y_minus_axis_label = "Left Analog Up"
input_r_x_plus_axis_label = "Right Analog Right"
input_r_x_minus_axis_label = "Right Analog Left"
input_r_y_plus_axis_label = "Right Analog Down"
input_r_y_minus_axis_label = "Right Analog Up"
input_menu_toggle_btn_label = "PS"
```

Restarting retroarch, and pressing the "P" button, the configuration is now found:
```systemctl restart retroarch```

```
Dec 26 13:38:48 Lakka retroarch[539]: RetroArch [INFO] :: [udev]: Plugged pad: Wireless Controller (0a12:0001) on port #0.
Dec 26 13:38:48 Lakka retroarch[539]: RetroArch [INFO] :: Autodetect: 99 profiles found
Dec 26 13:38:49 Lakka retroarch[539]: RetroArch [INFO] :: Autodetect: selected configuration: /tmp/joypads/udev/Sony-PlayStation4-DualShock4-Controller-Bluetooth.cfg
Dec 26 13:38:49 Lakka retroarch[539]: RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event7) supports force feedback.
Dec 26 13:38:49 Lakka retroarch[539]: RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event7) supports 16 force feedback effects.
```

Unfortunately the Dualshock doesn’t work ... :-(

What can I do now ?

Thank you for your help !

Kivutar, 2015-12-26:
Binding manually is not difficult. Just go in the Input Settings. Then go in user#1 binds, and use the "Bind All" action.

GabrieleV, 2015-12-26:
Ok, I removed the added file.
I Tried now via teh GUI: nothing happens when I press the keys on the controller.
I made:
* Settings
   * Input
      * Input User 1 Binds

Here I have
```
User 1 Device Type: Retropad
...
User 1 Device Index: Wireless Controller (#0)
```
I tried both a 
```User 1 Bind All```

and single buttons configuration, but nothing happens when I press any of the buttons on the controller :-(

GabrieleV, 2015-12-26:
I've made other tests: I tried to use retroarch on my ubuntu laptop.
After pairing the DS4 with the laptop, and running retroarch, the menu items started moving, and after some seconds, it seemed that the keyboard was not responding inside retroarch.
After keeping pressed "arrow up" on the laptop keyboard, I noticed that the menu items tried to scroll up: may be the DS4 continuously sends an "arrow up" event emulating a keyboard ?
The keyboard works prefectly outside the retroarch window.

GabrieleV, 2015-12-28:
Update: I think it's a kernel module problem (hid_sony).
I tested the DS4 on my Ubuntu machine using jstest-gtk: I was surprised when I noticed that connecting the DS4 via Bluetooth, and moving the fingers over the plate in front of the pad, it changed values in axis 9 and 10. I didn't know that the pad can emulate a mouse. No other data are exchanged when using other buttons.
I contacted the supposed mantainer of the hid_sony kernel module, If he replies, I continue investigating.

Kivutar, 2015-12-28:
Ok, what image of Lakka are you using? We ship different kernel versions depending on the hardware. I can check that the required module is enabled and up-to-date.

GabrieleV, 2015-12-28:
I'm using an image downloaded yesterday:
```
Lakka (unofficial) Version: devel-20151223095921-r21250-gd666dcf
Lakka git: d666dcf3cabbe53affea3428c56fbcab1609ebee
```

The laptop test were done under Ubuntu 15.04:
```
Linux kinsi 3.19.0-42-generic #48-Ubuntu SMP Thu Dec 17 22:54:45 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

Kivutar, 2015-12-28:
Thanks. I guess you're using the Generic image then, let me see if we have the sony driver..

GabrieleV, 2016-01-12:
I've received an answer from the kernel maintainer of the hid_sony module:

```
Sony made a change to the controller firmware which changed the size of the HID descriptor and caused this problem with newer controllers.  A fix is in kernel 4.5.
```

I see in Lakka we are using
```
Lakka:~ # uname -a
Linux Lakka 4.3.0 #1 SMP Sat Jan 9 12:41:39 CET 2016 x86_64 GNU/Linux
```
Is there someone that is able to backport the hid_sony kernel module from 4.5 ?

Kivutar, 2016-01-12:
Ah, well spotted! If those changes have been done in 1 or 2 commits, we can include them as patches that will be applied during the build.

GabrieleV, 2016-01-13:
Hy,
it seems easy to merge the patch. Here is a quote of the reply from the kernel module mantainer:

> Backporting should be easy since it’s just a two line fix that removes an unneeded size check.
> It’s here if you are interested:
> https://git.kernel.org/cgit/linux/kernel/git/jikos/hid.git/commit/?h=for-4.5/sony&id=b71b5578a84d297954e4812ba0ca2d466e61cf42

If you can recompile the patched module, attach it here, so I can test it on amd64 lakka version.

Kivutar, 2016-01-13:
Perfect, I'm rebuilding with this patch.

Kivutar, 2016-01-13:
Here, the build of january 13 http://static.kivutar.me/

GabrieleV, 2016-01-13:
It works perfectly !

With this patch and the joypad config posted here, the latest Sony DS 4 Works.
If you have previously paired the controller with another device, you have power it on and reset it using a pin inserted the reset button hole on the bottom of the joypad.
Some word about pairing: be sure to fully charge the DS4 using a standard Micro USB cable.
To pair the DS4, disconnect the usb cable, and power it on pressing the (PS) button: then the blue led on the front of the joypad will flashe slowly.
After you have to press (PS) and SHARE simultaneously until the led flashes quickly: the DS4 is now discoverable: proceed with the pairing. 
There is no need to use the USB cable.
No "Authorize service ''service_uuid'' (yes/no):" will be printed.
When the device is connected the led turns on fixed blue.

See https://wiki.gentoo.org/wiki/Sony_DualShock#DualShock_4

Kivutar, 2016-01-14:
Great. I'm adding that patch on a per project basis. Some of our kernels are too old to support it.

Can you help me with documentation about DS4 on the wiki please?

GabrieleV, 2016-01-14:
Sure ! I dropped you an email.

closed 2016-01-14 by GabrieleV