libretro/Lakka #267: Failure extracting Cores or Core Info fails

duduke, 2015-12-30:
When downloading an update to a core or core info files, and the core already exists as part of the unionfs (from /usr/lib/libretro) the extraction fails.
Inspecting /storage/cores I can see that the zip was downloaded, and manually extracting it inside /storage/cores successfully updates the core. also, looking at /tmp/cores shows the new core takes precedence over the old core from /usr/lib/libretro.

I believe that the UI extraction error is because it tries to extract the zip within /tmp/cores - and since the core already exists in a read-only mode (from /usr/lib/libretro) it fails extraction.

I haven't yet thought of an adequate solution...

Kivutar, 2015-12-30:
You're right. I think that it's a bug in RetroArch. It was working well before. Sombody identified a commit that may be linked to this.

duduke, 2016-01-02:
I think that commit: https://github.com/libretro/RetroArch/commit/6ff544ba1af04c5e185b4aac55444c4c29fc210e
won't fix this as the issue for LAKKA (I believe due to the use of UNIONFS) is that when we are trying to extract to /tmp/cores (the merged directory) we are trying to overwrite RO files (coming from /usr/lib/libretro), we should instead extract to /storage/cores - which is RW. This way /tmp/cores will show both RO files from /usr/lib/libretro and new extracted content from /storage/cores.

Kivutar, 2016-01-02:
Unionfs is supposed to support overwritting very well.

And this failure also happens when you download a core that override no file.

So it doesn't look like an overriding problem to me.

I did a test yesterday, with unionfs-fuse, on arch linux, and it was working. I don't get why it's not working on Lakka.

Two month ago, this feature was working well. It's a regression.

duduke, 2016-01-02:
I will re-test to make sure. but, I definitely remember getting extraction errors when trying to manually (via SSH) unzip a core and info.zip directly inside /tmp/cores.
Again, will re-test later today...

Kivutar, 2016-01-02:
I tested this morning, donwloaded the info files from RetroArch, extraction failed, I was able to run unzip manually and it worked.

duduke, 2016-01-04:
Manually extracting info.zip inside /tmp/cores:
xbmc-livingroom:/tmp/cores # unzip info.zip 
Archive:  info.zip
replace 2048_libretro.info? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
error:  cannot delete old 2048_libretro.info
        Read-only file system
replace 3dengine_libretro.info? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
error:  cannot delete old 3dengine_libretro.info
        Read-only file system
replace 4do_libretro.info? [y]es, [n]o, [A]ll, [N]one, [r]ename: 

Maybe something is wrong with the unionfs mount parameters?

Kivutar, 2016-01-04:
Strange, I don't get the same result at all!
Let me try on my side.

Kivutar, 2016-01-09:
You're right. I also get this error.

However, when I test on Arch Linux, I don't get this problem...

cudencuden, 2016-01-09:
I get this issue on odroid c1 too

mherberg, 2016-01-19:
Any update on this? I have an RPI2 and I'd like this issue fixed. Where are the mount parameters for unionfs located in the code? I'd love to be able to test this. It's obviously a uid/permission issue.

mherberg, 2016-01-19:
Nevermind, found it under Lakka/packages/libretro/retroarch/system.d.

If for some unexplainable reason it is not working on a platform to platform basis, can we switch it to OverlayFS? OverlayFS is part of the linux kernel from 3.18 onwards (so OpenELEC 6.0 should have it) and should be better tested..

Kivutar, 2016-01-19:
We can't, our Allwinner builds are stuck at kernel 3.4, and our imx6 builds at 3.14. We have also some 3.10 for Odroid.

mherberg, 2016-01-19:
Oh that's too bad. It's a consideration when all versions are 3.18 or later I guess. Do you have any idea what may be going wrong? I'm happy to test different mounting options on the RPI2.

Kivutar, 2016-01-19:
When this unionfs were introduced in Lakka, it was working well. I think that it started failing when I upgraded the kernel+systemd one or two month ago.

Now, if you try to manually download a zip and extract it using the command line in one of our unionfs, you will see error messages.

It is very important that we fix this bug, because it prevent users to save autoconfig files for joypads.

So any help will be really appreciated. I'm suck on this :/

duduke, 2016-01-19:
How can I see/edit the unionfs mount command?

mherberg, 2016-01-21:
This is a pretty big issue so I'm hoping we can figure it out.

The mount file is located at /usr/lib/systemd/system/tmp-cores.mount and it is run by Systemd on bootup. Unfortunately I'm not that familiar with Linux (I only use Unix and have programed for OS X/iOS/Windows) and I can't seem to edit this file because when I SSH into the my Lakka box it someone how doesn't mount the system as Read-write so I get the error "Read-only file system" when I try to modify the file. I tried remounting / with remount,rw but it doesn't work. I know this is simple... anyone have an idea of what I need to do?

PS:
By default the noobs installation I had did not allow SSH in. I had to add 'ssh' to the end of my boot line in cmdline.txt just before 'quiet'. Hope this helps anyone else trying to fix this. I'd recommend this be added to the FAQ in the section about SSH in. I'm not sure if it's enabled or not with other platforms but I only figured out how to enable it by searching the OpenELEC forums.

Kivutar, 2016-01-21:
You can disable that tmp-cores.mount using systemctl.
Then, you can redefine it in your storage partition, or directly use unionfs-fuse to do the union mount.

cudencuden, 2016-01-21:
you fixed it?

Kivutar, 2016-01-21:
Not yet, but now we know what is the problem at least. Systemd doesn't read the **Options** from the mount unit.

The strange thing is that on my Arch Linux, with the exact same fuse stack, systemd can read the Option and apply **cow** during the mount...

mherberg, 2016-01-22:
Simple fix that works for me. Anyone else want to test? Add on the Options line:

[Mount]
What=unionfs#/storage/cores=RW:/usr/lib/libretro=RO
Where=/tmp/cores
Options=cow
Type=fuse

I used this website as reference: https://grangerx.wordpress.com/2010/12/31/using-fuse-unionfs-with-centos-5-5-i686/.

mherberg, 2016-01-22:
Hmm for some strange reason this does not work tmp-joypads. If I wrong unzip from another directory then it does. Strange. I will have to take better look at this over the weekend when done with work.

mherberg, 2016-01-22:
Kivutar... have you thought about just placing the mounts in fstab? Systemd will convert it to native at boot and the options should work more consistently all platforms. I think fstab is actually the preferred method. I'm will to test this out.

Kivutar, 2016-01-22:
Adding Options=cow in the unit doesn't work for me.

Try to ps aux | grep unionfs, you will see that cow is not passed.

I still pushed this yesterday https://github.com/libretro/Lakka/commit/bab6f2f8a522c3f77b62a39f24c88529ee6bd7fe

That's why it fails for the joypads.

mherberg, 2016-01-23:
Sorry to be such a noob, but I'm not used to the OpenELEC linux base. All the directories appear read-only. I don't think I was loading the edited mount file properly. How do I ensure my unit file is being loaded over the default. I thought I could copy over the tmp-cores.mount unit file to /etc/systemd and it should take precedence over the the normal location, but /etc is read only. In a similar way /etc/fstab is also read-only, which would be another way we could fix the problem.

Kivutar, 2016-01-23:
Sure, openelec uses squashfs which is read only. Thanks to that we can have firmware upgrades like on a Wii or a PSP or any console.

And we use unionfs to allow overriding some folders in that squashfs.

So, I spent some time yesterday with @soul9 and it look like the mount command in busybox is not passing the cow option as it should. We tried to work around it, without success yet.

I think that I will just patch fuse-unionfs to force that cow flag.

mherberg, 2016-02-18:
Has there been any progress on this front? I have the same issue in the February build. I'd be happy to troubleshoot the issue this weekend if I could be any help.

Have you considered placing the mounts in /etc/fstab? Many linux flavors recommend sticking to fstab anyways (although I do agree you may lose some flexibility). systemd-fstab-generator will convert it into native units a boot. Of course it may fail but if it works it avoids the need to patch unionfs.

Kivutar, 2016-02-18:
The problem with still happend with fstab. There is an incompatibility between the mount command of busybox and fuse-unionfs. I need to declare this bug in the issue tracker of one of these projects, but I've been busy. If you have time, please declare this bug and reference this issue.

mherberg, 2016-02-18:
Strange bug. I can investigate and declare the bug this weekend. If it is specific to unionfs, would you consider a transition to overlayfs if I can get proper functionality? For lower kernel numbers we could provide the library.

Kivutar, 2016-02-20:
It's specific of unionfs-fuse

mherberg, 2016-02-21:
So I compiled Lakka after adding on the kernel module (by using menuconfig) for overlayfs and everything works (took some work and FYI I had to update the version of yasm to 1.3.0 to avoid compile errors on a fresh Ubuntu virtual machine). Looking at the latest OpenELEC linux._.conf files for projects that have kernels > 3.18, the flag is already on so I assume Lakka will adopt this when you update it from the OpenELEC master tree.

OverlayFS gives the expected behavior with mount (for example, 'mount -t overlay -o lowerdir=/etc/retroarch-joypad-autoconfig,upperdir=/storage/joypads overlayfs /storage/joypads'), fstab, and in System.d mount units.

The fuse based OverlayFS library compiled and it could then be used for the projects with kernel < 3.18. I haven't tested it but since it compiled and uses fuse I assume it should work just as the kernel module does. I'll try to test it tomorrow and I'll take a look at unionfs-fuse to declare the bug.

What are your thoughts about a transition? It shouldn't be too challenging to implement.

Kivutar, 2016-02-21:
If we can have it working on pre-3.18, I'm all for switching to OverlayFS. Thanks a lot for your help here.

cudencuden, 2016-03-01:
@mherberg did you have time to test with kernel < 3.18 ?

gyates100895, 2016-03-04:
I too would be happy to offer my machine and time for testing this, as I currently have this problem. Currently what I do to mitigate this problem is stop all the tmp folder (cores, joypads, database) units then run the command that @Kivutar mentioned (on Jan 21) "`/bin/unionfs -d -o cow /storage/cores=rw:/usr/lib/libretro=ro /tmp/cores/`" for all 3 and everything appears to work. Would it be possible to change the mounting of these file systems into a Systemd service unit with the task option set? or some other way use the the `unionfs` command instead of relying on Systemd to pass the arguments?

Kivutar, 2016-03-04:
Using systemd units is an alternative yes. It would work. But I don't know how to handle unmounting by this way.

I still think that the Overlay is a cleaner solution. I'm doing some searches on how to backport that to 3.4, 3.10 and 3.14 kernels.

If you can provide any informations on how to make systemd units for mounting properly, I can use it as a temporary solution until we find our way with Overlay.

gyates100895, 2016-03-04:
if there is a way to modify the systemd directories without getting read only errors, i would be happy to write the units and test them. a little coaching may be important if i need to compile my own lakka to test though. i can follow the compiling guide but im having some trouble understanding the file structure here on github.

Kivutar, 2016-03-04:
Yeah, but the good thing is that you don't need to compile Lakka to test your systemd units. You can add your custom units in the storage directory. The OpenELEC documentation explains that, iirc. Thanks for your help.

BTW, I found that https://github.com/adilinden/overlayfs-patches and I'm trying it now.

gyates100895, 2016-03-04:
I used the openELEC guide to achieve my current dual boot, so its surprising that I didn’t try there. regardless, that was the point of direction I needed so I thank you. Ive found some useful pages, mainly forum threads. however, if you have a specific page in mind, that would be greatly appreciated.

Kivutar, 2016-03-04:
Finally I've not been able to find any documentation about user session units on openelec.

I'm applying all overlayfs patches to every respective projects in lakka. I had a hard time finding a patch for the 3.14 version used for imx6, but finally found one.

Now a new difficulty arrises: there is an option, called workdir, which is mandatory on recent versions of overlayfs, and unsupported on old versions.

julianxhokaxhiu, 2016-03-05:
Wow! Which means that now is gonna work also con RPi2 (in my case) ? :D

mherberg, 2016-03-05:
FYI workdir has to be a temporary empty directory on the same drive. I mistyped it above.

Kivutar, 2016-03-05:
I got it working yesterday night on the PC image.

Tomorrow, I will try to get it working on pre 3.18 kernels. I this case, I will have to modify the patches to name the type **overlay** instead of the legacy name **overlayfs**. Then I will need to find a way to let those legacy overlayfs not crash when we provide a workdir.

closed 2016-03-06 by Kivutar

cudencuden, 2016-03-06:
nice.

commit d22bff9e (2016-03-06): Use OverlayFS instead of UnionFS-FUSE for union directories. Should fix #267