libretro/Lakka #269: New PC install boots to flower logo and stops

mcthiel, 2015-12-31:
I just installed a new stable installation (also tried the 12/30 nightly build) of Lakka and on the first reboot after the installation process, my PC loads to the Lakka flower logo and stops. Nothing else happens. I'm not sure where to go from here.

Here's my PC specs:

Aopen Digital Engine DE45-HG Mini Computer 
2.26Ghz Intel Core 2 Duo E8400
2GB RAM
160GB HDD

Kivutar, 2015-12-31:
Most of the times, this happen when we don't support your graphic card. What is your graphic card exactly?

mcthiel, 2015-12-31:
Ok, that's what I was afraid of. It is a Intel GMA X4500MHD. Is there a list of supported cards or do you have a rec if i was to upgrade? Thanks!

Kivutar, 2016-01-01:
Difficult to find informations about this exact model on google. I found some infos about Intel 4500 MHD, or Intel X4500 HD. And everything should work just fine with the i915 driver that we ship. I don't get why it fails.

mcthiel, 2016-02-17:
I still haven't been able to get this to work. Would you recommend me changing out the graphics card? If so, which one would you recommend?

gouchi, 2016-02-18:
What is the output of lspci -nn | grep VGA ?

mcthiel, 2016-02-18:
I'm currently on Windows so that command didn't work, but I used the PCI-Z program and got the following:

CPU name: Intel(R) Core(TM)2 Duo CPU     P8400  @ 2.26GHz, 2 logical CPUs
Operating system: Microsoft Windows 7
Available memory (MB): 1916

USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #1 
VGA compatible controller: Intel Corporation Mobile 4 Series Chipset Integrated Graphics Controller 
ISA bridge: Intel Corporation ICH9M LPC Interface Controller 
Ethernet controller: Intel Corporation 82567LF Gigabit Network Connection 
USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #4 
USB controller: Intel Corporation 82801I (ICH9 Family) USB2 EHCI Controller #2 
SMBus: Intel Corporation 82801I (ICH9 Family) SMBus Controller 
Audio device: Intel Corporation 82801I (ICH9 Family) HD Audio Controller 
USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #2 
Display controller: Intel Corporation Mobile 4 Series Chipset Integrated Graphics Controller 
USB controller: Intel Corporation 82801I (ICH9 Family) USB2 EHCI Controller #1 
USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #5 
USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #3 
IDE interface: Intel Corporation 82801IBM/IEM (ICH9M/ICH9M-E) 2 port SATA Controller [IDE mode] 
IDE interface: Intel Corporation 82801IBM/IEM (ICH9M/ICH9M-E) 2 port SATA Controller [IDE mode] 
Host bridge: Intel Corporation Mobile 4 Series Chipset Memory Controller Hub 
PCI bridge: Intel Corporation 82801 Mobile PCI Bridge 
USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #6 

Created by PCI-Z 1.3, http://www.pci-z.com/ | Database: The PCI ID Repository | Version: 2014.04.19

gouchi, 2016-02-24:
Can you copy also PCI device ? Right click  then Copy > Copy PCI device for the graphic card.

mcthiel, 2016-02-25:
This is the past of the copy PCI Device: 

VEN_8086&DEV_2448&SUBSYS_064DA0A0

gouchi, 2016-02-25:
You sure you paste the line with VGA compatible controller ? Because it looks like you paste the one with  PCI bridge: Intel Corporation 82801 Mobile PCI Bridge.

To be sure can you try with https://eternallybored.org/misc/pciutils

and type lspci.exe -nn and paste the line containing VGA.

mcthiel, 2016-02-26:
Yes that was PCI bridge, here's the VGA Controller:

VEN_8086&DEV_2A42&SUBSYS_064DA0A0

gouchi, 2016-03-13:
If you can't ssh, can you try some boot arguments (Press Tab when the menu shows up and add them at the end of the line) ? add debugging or textmode so that you can get some information on what is going on.

Source: http://wiki.openelec.tv/index.php/Boot_Arguments

Kivutar, 2016-03-20:
The graphic stack and kernel got upgraded since then. Can you try the latest nightly builds please?

mcthiel, 2016-03-20:
I just tried the latest nightly build for 3/20/16. I used the live run and the following message appeared in the top left corner of the screen, then reboot, upon trying to live run again, it paused at the flower logo again. Here's the message that appeared before reboot:

[image]

Kivutar, 2016-03-21:
Those are normal messages. You have to let the image resize its second partition. Please try to boot it again?

mcthiel, 2016-03-21:
After it reboots, the flower appears left-center and sits idle at the logo screen. I left it for 5 min plus and no movement. I tried the live run on another pc that I use for work and it booted up within 30 seconds.

Kivutar, 2016-03-21:
Ok still the same video card driver issue. @gouchi any idea of what can be missing in the kernel ?

mcthiel, 2016-03-21:
Ok, thanks for your help in solving this problem. Sorry this graphics card seems to be such a pest.

gouchi, 2016-03-21:
@Kivutar  I think kernel modules should be fine.

@mcthiel : can you try to add to boot parameters **nosplash debugging** to get more information when it stops.

mcthiel, 2016-03-22:
This is what appears on the screen in the top left corner when adding those boot parameters. It then stalls.

[image]

mcthiel, 2016-03-22:
Here's a few more using nosplash debugging Progress..
[image]
[image]

Kivutar, 2016-03-22:
@gouchi I still think that this is a video card driver issue.

@mcthiel Could you please connect with SSH or a keyboard, and launch retroarch manually to see the error message on launch? This procedure is explained in "How to debug retroach" in the FAQ I think.

gouchi, 2016-03-22:
There is a little more output but no fatal error.

Can you connect by ssh and get the output of dmesg ,  journalctl | grep drm and also try restart retroarch with :

systemctl stop retroarch
retroarch --menu --verbose >> log.txt 2>&1

And get the output of log.txt

mcthiel, 2016-03-23:
Here's the output of the first two commands. However, I'm having trouble getting the output of log.txt is there a command I use to output? Or should that txt file export somewhere on my computer?

[Terminal Saved Output.txt](https://github.com/libretro/Lakka/files/185383/Terminal.Saved.Output.txt)

mcthiel, 2016-03-23:
Ok I think I got the log.txt file. Let me know if this is not it.
[log.txt](https://github.com/libretro/Lakka/files/185412/log.txt)

Kivutar, 2016-03-23:
Ok, I need to talk about this to @twinaphex

mcthiel, 2016-03-26:
Ok, thanks

mcthiel, 2016-04-21:
Checking in to see if my problem will be able to be corrected, or if I need to look into a different graphics card.

DragarX, 2016-05-12:
I'm using the ASRock G41C-GS which has Intel X4500 and I don't have a single issue at all. Are you using 32bit or 64bit installer? I found using 32bit Lakka with older 64bit intel cpus seems to work much better. Although your CPU is about as old as mine, Socket-p has a few less features and compatibility.

Kivutar, 2016-09-24:
@mcthiel any update on this?

gouchi, 2016-09-24:
Also, you can try to change [video_monitor_index](https://github.com/libretro/RetroArch/blob/master/retroarch.cfg#L157) in [the configuration file](http://www.lakka.tv/doc/Configuring-Lakka/#using-the-command-line-interface) with
`video_monitor_index = 2`

Don't forget to stop retroarch before making any edits.

Nano shortcuts are CTRL+O (Save), CTRL+W (Search), CTRL+X (Quit)

Kivutar, 2016-12-08:
No replies, so I close it for now

closed 2016-12-08 by Kivutar