libretro/Lakka #273: [request] Change core option

lollo78, 2016-01-05:
Please, include an option into th retroarch Quick Menu to "Change core".
I select the wrong core for some games and the only way to reselect the right core from the GUI is to manually load it from Main Menu/Load Content/Select File And Detective Core.
Thx

Kivutar, 2016-03-08:
Sorry, but I would like to limit the number of entries in the quick menu, which is already too bloated for my taste..

lollo78, 2016-03-10:
Please, don't close this request.
Consider to add the voice "Change core" in this retroarch  menu:
[image]
At the moment, if we select the wrong core (the first time on game launch) we have to manually delete the bad core association inside the playlist file. 
But consider that this feature is also important if anyone want to try another core with a specific rom (some rom perform better with a specific core, instead of the most popular core for the same platform).

pokemontrainerAsh, 2016-10-11:
As a normal user, I selected a broken core and I find it ridiculous that there's no way to change the core. I find a post telling me to edit the playlist file. Well, I broke the whole thing. There needs to be an intuitive way to change the core when you select the game. The way it's setup now, you only get "one chance" to select the right core. Also, shouldn't the core be tied to a system and not a particular game?

Kivutar, 2016-10-11:
You're right. I reopen this. And @fr500 found a place to do that cleanly.

hizzlekizzle, 2016-10-11:
You can change the core assigned to a playlist by going to settings > playlist and changing it there.

Kivutar, 2016-10-11:
Yeah but they are talking of changing the core assigned to a given playlist entry.

daviderickson, 2016-12-31:
Yes please add this!!  It is super frustrating to not be able to change the core assigned to a game from the GUI...

manekinekodesu, 2017-06-03:
Still, not implemented in Lakka 2.1 RC1! Very frustrating issue. An option in "Run" menu would fix this:
\> Run
\> Run with...
\> Remove
\> Information

SiTWulf, 2017-06-06:
Not news for now...
the only option to fix this is edit the playlist and change the core to DETECT to reselect a new core.

D3miS3raph, 2017-06-20:
Still hoping for an option to fix this or a button to set a rom entry back to DETECT without doing it manually.  You could code an option after you click on a rom, that in the sub menu "reset to Detect mode", you click it, and it edits the entry.  Or any number of other ways.  Very hard to play with multiple cores right now :(