libretro/Lakka #274: Image icon stuck on Gui

lollo78, 2016-01-05:
In some condition (I didn't understand how to replicate it), some icon like the white folder or the squared rocket stuck on the GUI. I notice that this happens if I load games from the Load Content menu for a non "traditional" core (GW, 4do...).
I'm using the nightly of 30th december.

lollo78, 2016-01-05:
[image]

Kivutar, 2016-01-06:
Yes, this is a very old bug. Sadly, fixing it require a major refactoring in the menu system of RetroArch and nobody want to take the risk to cause hundreads of new bugs to fix this :/

Maybe in the future, when RetroArch becomes very stable, I may start fixing this the right way.

Kivutar, 2016-08-11:
Won't fix.

closed 2016-08-11 by Kivutar

lollo78, 2016-08-12:
o_O
:O 
:-| 
:'(

Kivutar, 2016-08-12:
Yeah I know... but fixing it means breaking a ton of other stuff in RA. I would rather rewrite the entire menu system than trying to fix that.