libretro/Lakka #275: Support for custom/homemade parallel port controllers

eddyebur, 2016-01-06:
I have a homemade arcade that relies on parallel port for input. I installled lakka on it but couldn't make it detect inputs configuring via GUI. Maybe i'm missing something but if it doesn't support yet, here's an enhancement goal. Thanks devs.

twinaphex, 2016-01-06:
RetroArch has a joypad driver meant specifically for gamepads connected through the parallel port.

https://github.com/libretro/RetroArch/blob/master/input/drivers_joypad/parport_joypad.c

You'll have to go to Settings -> Driver Settings, change the Joypad driver to 'parport', and then restart.

eddyebur, 2016-01-06:
I did it but it still doesn't work and can't remap. Clues?

twinaphex, 2016-01-06:
I have never used it personally but the guy who programmed it vouched that it works for him.

Maybe it is just not finding your autoconfig file for parport. 

Anyway, can't really help you further there, I don't own any PCs with a parallel port anymore. You'd have to find a dev who cares about and a dev who can code and has the hardware to take a look at it.

eddyebur, 2016-01-06:
I found this on Lakka forums:
http://libretro.com/forums/showthread.php?t=4825
I'm using the latest release build, so I guess i'm missing those modules. Can we have it by default (if it really does solve this problem)?

Kivutar, 2016-01-09:
Let see if the guy on the forum succeeds.

eddyebur, 2016-01-14:
I was messing with the source code and I found a line in /projects/Generic/linux/linux.x86_64.conf and I changed "# CONFIG_PARPORT is not set" to "CONFIG_PARPORT=y". By doing this when I compile lakka it asks directly on the terminal if I want to add (I guess) some modules including PARPORT_PC and some others related to parallel port in general and even game consoles controllers (that may be interesting for this project). I never compiled an OS before so I think you know better than I what I'm saying here. But in the end I wasn't able to install this compiled Lakka on PC because I get black screen after Live/Installation boot menu. Maybe I missed something on compilling the code, but I followed lakka docs page. I hope this helps and I can help testing if you compile it.

balthazar, 2016-03-06:
Hello again

So in reply to eddyebur mainly, you need to make sure that when the interactive configuration kicks in, you reply "m" instead of "y". This tells the compile process to also build the requested modules and add them to the build.

You could also add "make menuconfig" somewhere in the linux package.mk file, this will prompt a GUI that will allow you to configure your kernel options before compilation.

In my case, once the modules were built, i also added the jstest tool to the lakka build so that I could confirm that the joystick inputs were properly registering through ssh, add the proper modprobe line in the autostart and configure the lakka inputs.

I hope some of this made sense, I don't know if it's useful enough for inclusion in the master branch, since in the case of parallel port joysticks it really depends on the wiring nitty-gritty and there are tons of possible modules for gamepads...

Kivutar, 2016-03-06:
To access the menuconfig of the kernel, you can go to the linux directory in the build directory. Then you type

    ARCH=x86_64 make menuconfig

(Adapt ARCH to your needs) And copy the resulting .config in your project.

eddyebur, 2016-03-06:
Thanks for feedback. I'll try it next week when I arrive home and report my results.

commit 3de828ce (2016-03-14): Enable kernel module for parallel port controllers. Should fix #275
closed 2016-03-14 by Kivutar