libretro/Lakka #278: [enhancement] sixad/sixpair for Dualshock 3 Autoconnect

fluffymadness, 2016-01-10:
Something like that:

/etc/udev/rules.d
99-sixpair.rules
----
DRIVER=="usb", SUBSYSTEM=="usb", ATTR{idVendor}=="054c", ATTR{idProduct}=="0268", RUN+="/scripts/udev-sixpair.sh"

/scripts/udev-sixpair.sh
#!/bin/bash
log=/scripts/controllers.log
echo "ps3 usb controller detected" >> $log
sixpair >> $log

Systemd Service:

[Unit]
Description=sixad - Sixaxis Daemon
# It does not conflict, if the input plugin is disabled.
#Conflicts=bluetooth.service
After=bluetooth.service
Requires=bluetooth.service

[Service]
ExecStart=/usr/bin/sixad -s
ExecStop=/usr/bin/sixad --stop
#ExecStop=/usr/bin/sixad -r
Restart=on-failure

[Install]
WantedBy=bluetooth.target
#WantedBy=multi-user.target

------
Basically sixad daemon starts on boot.
When you connect the dualshock 3 via usb udev recognizes it, starts sixpair.
Then you just need to unplug the cable, press the ps button and the controller is connected via bluetooth.

I guess this probably won't be implemented in the official build, since bluez official supports the dualshock 3. I tried to copy the compiled binaries to the image, but since lakka has a readonly filesystem, and an update overwrites everything, I think that I would have to build it myself if I want that feature.
I just don't find it very handy to manually pair my controller, cause if I don't have a pc near me, I can't connect over ssh to do it.

duduke, 2016-01-10:
I would change the title to AutoPair instead of Autoconnect.

Autoconnection (once already paired) can be done using:
create /storage/.cache/services/bluez.conf
HIDD_OPTIONS="--connect xx:xx:xx:xx:xx:xx --connect xx:xx:xx:xx:xx:xx"

Where each connect param will try to connect to a different controller.

fluffymadness, 2016-01-10:
Thanks for your answer, but still, the current process isn't very userfriendly in my opinion.Especially if you use your controller somewhere else. 

Right now there are two solutions for me:

Everytime when I used the dualshock 3 on my pc and I want to connect it to lakka, I'll have to do this procedure or at least part of it again.

Or a workaround would be an external bluetooth stick, that I carry around with me, where the controllers are paired to, which I plug into the notebook or the odroid...

I know using sixpair doesn't allow bluetooth to be used for anything else.
Recalbox solved it this way, that there is an external config file on a samba share, where you can change behavior, by commenting things out. Then it uses different drivers, cores etc.
Maybe the devteam can make an option to use/deactivate sixad.

Btw, is it somehow possible to mount the filesystem rw, so that i can copy the needed binaries to it?

duduke, 2016-01-11:
What is your main issue?:
A. Having to re-pair controllers? (because you are using them for both lakka and something else).
B. Pairing completely new controllers without touching SSH?
I'm pretty sure I can solve A without using SIXAD.

fluffymadness, 2016-01-11:
Both would be good to have.
Right now I'm modifiyng sixad to work from a single directory
namely /storage/.config/
The progress right now is, Pairing works...Dualshock gets connected. (original, shawan and gasia) and works in lakka

What does not work:
Udev rule for sixpair doesn't fire....I don't know why
My modified systemd service runs fine when I start it manually but doesnt work on boot and I don't know why

Lakka:~ # systemctl status sixad
● sixad.service - sixad - Sixaxis Daemon
   Loaded: loaded (/storage/.config/system.d/sixad.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

When this is working, the files just need to be installed via sftp drag and drop. And the permissions must be set correctly. Then it would be good to go.

How would you solve A without sixad?

fluffymadness, 2016-01-11:
done :) everything working as intended.
No command line fu for pairing and connecting.
Just plugging the controller with the cable, a few secs waiting time
Unplugging the controller
Pressing the playstation button
Connected :)

If there is interest to integrate this into the project, I would open a git and share it

duduke, 2016-01-11:
Once paired and trusted, simply update bluez.conf with the list of controllers:
/storage/.cache/services/bluez.conf
HIDD_OPTIONS="--connect xx:xx:xx:xx:xx:xx --connect xx:xx:xx:xx:xx:xx"
This causes bluez to always try to connect to those controllers. I use this for my NES30 so that they will automatically connect when powered on (even after a reboot).

fluffymadness, 2016-01-11:
but will this work with the dualshock 3, because if I pair them somewhere else the pairing mac of the ds3 controller gets set to the address of the new bluetooth adapter

Kivutar, 2016-01-11:
@duduke I didn't know about that bluez.conf, I added it to the wiki.

@fluffymadness I think that the DS3 stores the mac address of the last paired bluetooth receiver. So yes, you will have to repair each time you pair it to another receiver.

Now, I'm very happy that you successed in configuring an automatic pairing. I tried to do it myself, 3 month ago, but I was stupborn and didn't wanted to use sixpair. Now I agree to use sixpair. Please give me any instruction that are not yet in this thread, and I will package everything cleanly in Lakka.

duduke, 2016-01-11:
I don't think SixAd is required for a reconnect as if i'm not mistaken, in new versions of BlueZ when you connect DS3 via USB, it will acquire the BT MAC address of the local receiver. Since it is already authorized, it will not require running anything via SSH.
Pairing might require SixAd - but there might be a way to do it with only bluez.

fluffymadness, 2016-01-11:
hmmm...I don't know...I was thinking about that. I think the bluetooth frontends in linux must communicate with bluez somehow, but I haven't really looked into that.
Maybe there is a way over dbus or something?

fluffymadness, 2016-01-11:
so basically we could use sixpair for pairing and by using a script to generate this one 
/storage/.cache/services/bluez.conf
HIDD_OPTIONS="--connect xx:xx:xx:xx:xx:xx --connect xx:xx:xx:xx:xx:xx"
we could automate it somehow?

fluffymadness, 2016-01-11:
@kivutar here is my system.d file

>[Unit]
>Description=sixad - Sixaxis Daemon
>#It does not conflict, if the input plugin is disabled.
>#   Conflicts=bluetooth.service
>After=bluetooth.service
>Requires=bluetooth.service

>[Service]
>Environment=LD_LIBRARY_PATH=/storage/.config
>ExecStartPre=/usr/bin/hciconfig hci0 up
>ExecStart=/storage/.config/sixad -s
>ExecStop=/storage/.config/sixad --stop
>#ExecStop=/usr/bin/sixad -r
>#RemainAfterExit=yes
>Type=simple
>Restart=on-failure

>[Install]
>WantedBy=bluetooth.target
>WantedBy=multi-user.target

fluffymadness, 2016-01-11:
I used this fork
https://github.com/ryonlabaw/qtsixa/tree/master/sixad,
because with that, my SonyDS3, the ShanwanDS3, and the GASIADS3 all function.
Sadly this thing has some little bugs: 

The playerLed shows always Player1, 
ForceFeedback is not working

The edits I made to get it working in /storage/config are 
the sixad file:

```
#!/bin/bash

# sixad-bin wrapper
# written by falkTX

DEBUG=1
LEGACY=0

. /storage/.config/sixad-config/default

if [ -f /usr/bin/sudo ]; then
  SUDO="/usr/bin/sudo"
else
  SUDO=""
fi

bt_device_check () {
if (which hciconfig > /dev/null); then
  if (hciconfig dev > /dev/null); then
    VER=`hciconfig default version | grep "HCI Ver" | awk '{print$3}'`
    if [ "$VER" == "1.1" ]; then
      echo "***** NOTICE *****"
      echo "You're using a very old bluetooth dongle,"
      echo "the Sixaxis will not work properly!"
    elif [ "$VER" == "1.0" ]; then
      echo "***** WARNING *****"
      echo "You're using a _really_ old bluetooth dongle,"
      echo "the Sixaxis will just not work!"
    fi
  else
    echo "No bluetooth adapters found on the system!"
    echo "sixad will now quit"
    exit
  fi
fi
}

sixad_running_check () {
#ps -e | grep sixad-bin > /dev/null
0 > dev/null
}

bluetoothd_running_check () {
ps -e | grep bluetoothd > /dev/null
}

modprobe_check () {
/sbin/modprobe uinput
}

bt_start () {
echo "im here"
rm -f /tmp/.sixad-dbus-blocker.pid
env sleep 1

if [ -f /lib/udev/rules.d/97-bluetooth.rules ]; then
  /usr/sbin/bluetoothd --udev
elif [ -f /etc/rc.d/bluetooth ]; then
  /etc/rc.d/bluetooth start
else
  echo systemd
  systemctl start bluetooth
fi
}

bt_stop() {
if (bluetoothd_running_check); then
  killall -9 bluetoothd
fi

/storage/.config/sixad-dbus-blocker &
}

case $1 in

  --start|-start|start|-s)
REMOTE=0
bt_device_check
if (sixad_running_check); then
  echo "sixad is already running."
  echo "run '$0 --stop' to stop it"
else
 if (modprobe_check); then  #Check for root access before running, If NO access, quit
  bt_stop
  /storage/.config/sixad-bin $DEBUG $LEGACY $REMOTE
 else
  echo "You need admin/root access to run this application"
 fi
fi
  ;;

  --stop|-stop|stop)
killall -9 sixad-sixaxis
killall -9 sixad-remote
killall -9 sixad-bin
bt_start
  ;;

  --remote|-remote|remote)
REMOTE=1
bt_device_check
if (modprobe_check); then  #Check for root access before running, If NO access, quit
  bt_stop
  /storage/.config/sixad-bin $DEBUG $LEGACY $REMOTE
else
  echo "You need admin/root access to run this application"
fi
  ;;

  --restore|-restore|restore|-r)
bt_start
  ;;

  --boot-yes)
# ArchLinux
if [ -f /etc/arch-release ]; then
  $SUDO sed '/DAEMONS=/ s/)/ sixad)/g' -i /etc/rc.conf
# Gentoo
elif [ -f /etc/gentoo-release ]; then
  $SUDO rc-update add sixad
# Debian (default)
else
  if [ -f /etc/rc2.d/S90sixad ]; then true; else $SUDO ln -s /etc/init.d/sixad /etc/rc2.d/S90sixad; fi
  if [ -f /etc/rc3.d/S90sixad ]; then true; else $SUDO ln -s /etc/init.d/sixad /etc/rc3.d/S90sixad; fi
  if [ -f /etc/rc4.d/S90sixad ]; then true; else $SUDO ln -s /etc/init.d/sixad /etc/rc4.d/S90sixad; fi
  if [ -f /etc/rc5.d/S90sixad ]; then true; else $SUDO ln -s /etc/init.d/sixad /etc/rc5.d/S90sixad; fi
fi
  ;;

  --boot-no)
# ArchLinux
if [ -f /etc/arch-release ]; then
  $SUDO sed "s/ sixad//" -i /etc/rc.conf
# Gentoo
elif [ -f /etc/gentoo-release ]; then
  $SUDO rc-update delete sixad
# Debian (default)
else
  if [ -f /etc/rc2.d/S90sixad ]; then $SUDO rm /etc/rc2.d/S90sixad; fi
  if [ -f /etc/rc3.d/S90sixad ]; then $SUDO rm /etc/rc3.d/S90sixad; fi
  if [ -f /etc/rc4.d/S90sixad ]; then $SUDO rm /etc/rc4.d/S90sixad; fi
  if [ -f /etc/rc5.d/S90sixad ]; then $SUDO rm /etc/rc5.d/S90sixad; fi
fi
  ;;

  --help|-help|help|-h)
echo "[Qt]SixA Daemon"
$0
  ;;

  --version|-version|version|-v)
echo "[Qt]SixA Daemon - version 1.5.1"
  ;;

  *)
echo "usage: $0 <command>

command can be:
    -h, --help          Show help (this message)
    -v, --version       Show sixad version

    -s, --start         Start sixad
        --stop          Stop sixad
        --remote        BD Remote mode

    -r, --restore       Restore regular bluetooth

        --boot-yes      Auto-starts sixad at boot time
        --boot-no       Does not auto-start sixad at boot time

You can also check: sixad-raw, sixad-notify"
  ;;

esac
```

and in bluetooth.cpp you have to edit
the line
const char* uinput_sixaxis_cmd = "/usr/sbin/sixad-sixaxis";
with the new path

fluffymadness, 2016-01-11:
Ok. i think that was it.
If you include it, maybe you can make a switch or so, because I think it reserves bluetooth exclusively, but I'm not sure.

fluffymadness, 2016-01-11:
@duduke
sixpair just sets the bt mac for pairing
sixad-bin is the daemon that listens for a ds3 that tries to connect via bluetooth
sixad-sixaxis gets called after connection with sixad-bin

so would it be possible to use sixpair for pairing and then using a script to generate the entries in the config file?
Something like, when psx controller connected via udev rule, check if mac is already in file, if not, start sixpair and put mac adress in bluez file
/storage/.cache/services/bluez.conf
HIDD_OPTIONS="--connect xx:xx:xx:xx:xx:xx --connect xx:xx:xx:xx:xx:xx"
for autoconnecting via bluez?

Kivutar, 2016-01-12:
I think that every systemd compatible controler like bluetoothctl have a dbus api. At least, I know that KODI uses dbus to talk to connmanctl.

When we pair and connect using bluetoothctl, are there some files created in .cache?

duduke, 2016-01-12:
yes, files are created according to the BT MAC of each controller.
example: .cache/bluetooth/00:19:86:00:2B:48/61:04:25:67:71:01 
We have an info file under the controller folder which contains info about the controller and pairing key.

Documentation around BlueZ is horrible, I've been trying to find a way to auto-pair DS3 without SixAd but to no avail.

I've found some info in the ArchLinux wiki (https://wiki.archlinux.org/index.php/Gamepad#Connecting_via_Bluetooth):
"Connecting via Bluetooth
Install the bluez-plugins and bluez-utils packages, which includes the sixaxis plugin. Then start bluetooth.service, plug the controller in via USB, and the plugin should program your PC's bluetooth address into the controller automatically.
You can now disconnect your controller. The next time you hit the PlayStation button it will connect without asking anything else.
Just remind to disconnect the device once you are done, once it will stay connected, on and consuming battery."

Which suggest that there is some automation around it, but since I don't have ArchLinux installed, I don't know what they use.

Kivutar, 2016-01-13:
I do have Arch Linux installed with bluez-plugins and bluez-utils, and never noticed this automatic pairing.

lollo78, 2016-01-15:
Hi guys, I have a little noob question.
I'm not sure if is a Retroarch bug, a faulty configuration of mine or a new feature...o_O
If I connect two PS3 controller, only the #1 can interact with Retroarch menu. The #2 can't do this, but it works fine in game.
Maybe I drunk a lot, but I could have sworn that (on old builds, a lot of time ago) both connected controllers did work in Retroarch menu.

Kivutar, 2016-01-15:
@lollo78 I don't remember that two controllers could control the menu at the same time. If it was the case, then we should add back that behavior! Anyway, no more off-topic, please discuss these on the forums :)

lollo78, 2016-01-15:
Ok, I have to stay away from the beers!
Sorry for the OT
:*

Kivutar, 2016-01-15:
:*

Alcaro, 2016-01-15:
:*

danfozzy, 2016-01-19:
any update on this?

Kivutar, 2016-01-19:
My guess is that we could use commands like dbus-send in the bash script linked to the udev rule to achieve this in a bluetoothctl compatible way.

fluffymadness, 2016-01-20:
lol i can't even pair the dualshock manually...I did the pairing process as described when I connect it says connected and the leds of my dualshock keep flashing....oh well
Anyways...
@Kivutar 
The script method doesn't work for me
it always tells me no agent registered when i turn agent on via script....
I tried to find a different method...

The intel galileo site told me the following:
where we have to accept the pairing by typing in "yes". Hence, the agent within bluetoothctl still requires some interaction - if you really want no interaction you would need to go for your own agent. You may use the simple-agent test script as starting point to write your own..

Ok that led me to
http://www.bluez.org/release-of-bluez-5-0/
there is a source tar file of bluez

There is a directory called test.
Inside this dir there is a simple-agent python script that describes how to do things via d-bus....yay O.o

'''
#!/usr/bin/python

from __future__ import absolute_import, print_function, unicode_literals

from gi.repository import GObject

import sys
import dbus
import dbus.service
import dbus.mainloop.glib
from optparse import OptionParser
import bluezutils

BUS_NAME = 'org.bluez'
AGENT_INTERFACE = 'org.bluez.Agent1'
AGENT_PATH = "/test/agent"

bus = None
device_obj = None
dev_path = None

def ask(prompt):
	try:
		return raw_input(prompt)
	except:
		return input(prompt)

def set_trusted(path):
	props = dbus.Interface(bus.get_object("org.bluez", path),
					"org.freedesktop.DBus.Properties")
	props.Set("org.bluez.Device1", "Trusted", True)

def dev_connect(path):
	dev = dbus.Interface(bus.get_object("org.bluez", path),
							"org.bluez.Device1")
	dev.Connect()

class Rejected(dbus.DBusException):
	_dbus_error_name = "org.bluez.Error.Rejected"

class Agent(dbus.service.Object):
	exit_on_release = True

	def set_exit_on_release(self, exit_on_release):
		self.exit_on_release = exit_on_release

	@dbus.service.method(AGENT_INTERFACE,
					in_signature="", out_signature="")
	def Release(self):
		print("Release")
		if self.exit_on_release:
			mainloop.quit()

	@dbus.service.method(AGENT_INTERFACE,
					in_signature="os", out_signature="")
	def AuthorizeService(self, device, uuid):
		print("AuthorizeService (%s, %s)" % (device, uuid))
		authorize = ask("Authorize connection (yes/no): ")
		if (authorize == "yes"):
			return
		raise Rejected("Connection rejected by user")

	@dbus.service.method(AGENT_INTERFACE,
					in_signature="o", out_signature="s")
	def RequestPinCode(self, device):
		print("RequestPinCode (%s)" % (device))
		set_trusted(device)
		return ask("Enter PIN Code: ")

	@dbus.service.method(AGENT_INTERFACE,
					in_signature="o", out_signature="u")
	def RequestPasskey(self, device):
		print("RequestPasskey (%s)" % (device))
		set_trusted(device)
		passkey = ask("Enter passkey: ")
		return dbus.UInt32(passkey)

	@dbus.service.method(AGENT_INTERFACE,
					in_signature="ouq", out_signature="")
	def DisplayPasskey(self, device, passkey, entered):
		print("DisplayPasskey (%s, %06u entered %u)" %
						(device, passkey, entered))

	@dbus.service.method(AGENT_INTERFACE,
					in_signature="os", out_signature="")
	def DisplayPinCode(self, device, pincode):
		print("DisplayPinCode (%s, %s)" % (device, pincode))

	@dbus.service.method(AGENT_INTERFACE,
					in_signature="ou", out_signature="")
	def RequestConfirmation(self, device, passkey):
		print("RequestConfirmation (%s, %06d)" % (device, passkey))
		confirm = ask("Confirm passkey (yes/no): ")
		if (confirm == "yes"):
			set_trusted(device)
			return
		raise Rejected("Passkey doesn't match")

	@dbus.service.method(AGENT_INTERFACE,
					in_signature="o", out_signature="")
	def RequestAuthorization(self, device):
		print("RequestAuthorization (%s)" % (device))
		auth = ask("Authorize? (yes/no): ")
		if (auth == "yes"):
			return
		raise Rejected("Pairing rejected")

	@dbus.service.method(AGENT_INTERFACE,
					in_signature="", out_signature="")
	def Cancel(self):
		print("Cancel")

def pair_reply():
	print("Device paired")
	set_trusted(dev_path)
	dev_connect(dev_path)
	mainloop.quit()

def pair_error(error):
	err_name = error.get_dbus_name()
	if err_name == "org.freedesktop.DBus.Error.NoReply" and device_obj:
		print("Timed out. Cancelling pairing")
		device_obj.CancelPairing()
	else:
		print("Creating device failed: %s" % (error))

	mainloop.quit()

if __name__ == '__main__':
	dbus.mainloop.glib.DBusGMainLoop(set_as_default=True)

	bus = dbus.SystemBus()

	capability = "KeyboardDisplay"

	parser = OptionParser()
	parser.add_option("-i", "--adapter", action="store",
					type="string",
					dest="adapter_pattern",
					default=None)
	parser.add_option("-c", "--capability", action="store",
					type="string", dest="capability")
	parser.add_option("-t", "--timeout", action="store",
					type="int", dest="timeout",
					default=60000)
	(options, args) = parser.parse_args()
	if options.capability:
		capability  = options.capability

	path = "/test/agent"
	agent = Agent(bus, path)

	mainloop = GObject.MainLoop()

	obj = bus.get_object(BUS_NAME, "/org/bluez");
	manager = dbus.Interface(obj, "org.bluez.AgentManager1")
	manager.RegisterAgent(path, capability)

	print("Agent registered")

	# Fix-up old style invocation (BlueZ 4)
	if len(args) > 0 and args[0].startswith("hci"):
		options.adapter_pattern = args[0]
		del args[:1]

	if len(args) > 0:
		device = bluezutils.find_device(args[0],
						options.adapter_pattern)
		dev_path = device.object_path
		agent.set_exit_on_release(False)
		device.Pair(reply_handler=pair_reply, error_handler=pair_error)
		device_obj = device
	else:
		manager.RequestDefaultAgent(path)

	mainloop.run()

	#adapter.UnregisterAgent(path)
	#print("Agent unregistered")
'''

Kivutar, 2016-01-20:
I suspected that this interection was going to be the problem.

Yeah, looks like dbus is the way to go.

Thanks for this precious script. I prefer to not include python in builds, but we can certainly translate this to bash using the command line dbus commands.

danfozzy, 2016-02-12:
is this included in the latest nightly build?

Kivutar, 2016-02-24:
@danfozzy no, not finished yet.

gouchi told on IRC:

> <gouchi> [22:42:35] kivutar: you should try to package https://github.com/khvzak/bluez-tools for openelec
> <gouchi> [22:43:02] kivutar: I could pair my controller with 2 commands
> <gouchi> [22:51:10] kivutar: bt-adapter -d to get the MAC then bt-device --set MAC Trusted 1 to trust the device
> <gouchi> [23:04:35] kivutar: there is an example here https://github.com/viljoviitanen/Openelec-addons/blob/master/packages/addons/bluez/bluez-tools/meta

Kivutar, 2016-02-27:
Here is what I get:

    Lakka:~ # bt-agent 
    Agent registered
    Default agent requested

And I never get back to the prompt.

fluffymadness, 2016-04-21:
@kivutar
After i dont know how long i was searching for bluez related stuff..i think i got lucky :)
https://github.com/ValveSoftware/SteamOS/issues/479

Kivutar, 2016-04-21:
Thanks, I'm testing it right now. I really hope it will be so simple :+1:

Kivutar, 2016-04-21:
Looks like it worked. I still had to power on the bluetooth manually in my case. But I didn't had to enter any commands to pair.

However, my controller and my dongle were already paired. So I'm not sure of the right procedure to reset all to do a clean test.

Can you help me to test it? What hardware are you running lakka on?

fluffymadness, 2016-04-21:
I'm using an odroid xu4.
I think you can reset the controller with the reset button on the back, there is a small hole on the right side of the middle screw.
On the pc side,with bluetoothctl, there should be a list of paired/trusted devices.
You should be able to delete the already paired controllers there.

I'm on the go right now, but I could help you test in the evening if you'd like :)

Kivutar, 2016-04-21:
I pushed that patch and uploaded a build.

fluffymadness, 2016-04-22:
It doesn't work for me with my controller....but I think that bluez-pairing didn't work for me before this patch either ^^"...only sixpair did its job for me...I'd have to get a different bluetooth adapter to test this.
I have the csl one from amazon atm. 
So did it work for you with both adapters :) ?

I guess that it should work with the rule...it seems to be the standard method for doing that since its also mentioned in the arch wiki
https://wiki.archlinux.org/index.php/bluetooth

fluffymadness, 2016-04-22:
Apr 22 18:32:05 Lakka bluetoothd[3656]: sixaxis: compatible device connected: PLAYSTATION(R)3 Controller (054C:0268)

When i press the ps button for connecting, when the cable is disconnected, I briefly get a controller connected in the lakka interface and 2secs later or so the disconnected message..

fluffymadness, 2016-04-22:
...
Apr 22 18:57:53 Lakka bluetoothd[3208]: Failed to obtain handles for "Service Changed" characteristic
Apr 22 18:57:53 Lakka bluetoothd[3208]: hci0 Load Connection Parameters failed: Unknown Command (0x01)....
...
whenever i plug my stick out and in

Kivutar, 2016-04-25:
This is strange, for me, it works on reboot. Have you another bluetooth dongle to try?

fluffymadness, 2016-04-25:
Sadly not. Which one do you have ?
If pairing and connecting, and everything else works for you I could buy the same one.

Kivutar, 2016-04-25:
I have an official DualShock 3 (not a clone). Be careful, there are very good clones out there. It's difficult to recognize them by your eyes.

Let follow this topic, where we're working on supporting more clones: https://github.com/libretro/Lakka/issues/302

Kivutar, 2016-04-25:
Sorry, you were talking about the dongle:

    Bus 001 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

fluffymadness, 2016-04-25:
I meant your bluetooth dongle, i want to buy the same one, since mine seems to have some weird issues with bluez.
My controllers are original, usb string and everything matches, i bought them through an official retailer...and yes had some scary fakes once ^^ if you need testing for fakes i have also shanwan and gasia ds3 flying around.

Kivutar, 2016-04-25:
Sure. We need help for this. We are already 3 people testing, but pairing procedures can differ a lot. So having more test results would help.

My bluetooth dongle reference is in the message above.

Kivutar, 2016-04-29:
This is working now. You need to go in Settings > Lakka Services to enable bluetooth before trying.

closed 2016-04-29 by Kivutar

fluffymadness, 2016-04-29:
Awesome. Thx