libretro/Lakka #279: Steam Controller not working

aboutafter, 2016-01-12:
The Steam Controller doesn't work in Lakka. I'm not sure if it's possible to make it work, but it would be a very nice addition if possible. 

The problem I see with it is that without Steam, I think it only works as a mouse, so the majority of keys don't do anything (or at least it doesn't recognize almost anything when I try to bind it manually). Once Steam is open you can change modes (it imitates an Xbox 360 gamepad, or WASD keys and other templates), so maybe there is a mode in Lakka to make it behave like a Xbox 360 gamepad?

Kivutar, 2016-01-12:
I found this https://github.com/libretro/RetroArch/issues/2390

Have you tried to do a manual bind?

aboutafter, 2016-01-12:
Hi Kivutar, yes I did, but the only key that works it's the "A" key.

aboutafter, 2016-01-14:
Maybe Lakka needs something like this project so the controller could work easily? https://github.com/ynsta/steamcontroller

johnjelinek, 2016-01-21:
unfortunately, the driver above requires python -- which doesn't ship w/ Lakka

johnjelinek, 2016-01-21:
I'm not sure if this driver might work: https://github.com/cvuchener/steamcontroller-linux-kernel I don't know how to install it in lakka.

Kivutar, 2016-02-27:
I saw some kernel patches about the steam controller in openelec

Kivutar, 2016-03-14:
Or maybe just this? http://askubuntu.com/questions/686214/how-do-i-get-a-steam-controller-working

aboutafter, 2016-03-15:
No, with that it only works in basic mode.

Kivutar, 2016-03-15:
What's the difference between basic mode and other modes?

johnjelinek, 2016-03-15:
It's hard to explain without showing you. It's unique in that the touchpads
double as trackpads (eg: mouse emulation) or as analog sticks or buttons.
Are you suggesting to jump into the terminal in lakka to get uinput working?

Kivutar, 2016-03-17:
I saw some udev rules to enable mouse/trackpad emulation too (through uinput yeah).

Is this the way to go?

Lakka (OpenELEC) allows you to add custom udev rules in the storage partition. We could use that way to experiment.

aboutafter, 2016-04-12:
There is news about the driver I posted in January: 

https://www.gamingonlinux.com/articles/theres-an-open-source-standalone-steam-controller-driver.7015

Apparently it already works in Xbox 360 controller mode. Is there any possibility to add this to Lakka or the fact that it requires python is a no-no as @johnjelinek says?

Kivutar, 2016-04-13:
Yeah the python part is a no-no, sorry. (I don't know how people nowadays think that adding python as dependancy for a joypad driver can be smart)

johnjelinek, 2016-04-13:
Well, Python is included in Debian and Ubuntu out of the box, so most
people prolly don't think they're taking on an extra dependency, but I know
what you mean. Do you think all userland drivers should be discarded and
only kernel modules should be adopted?

Kivutar, 2016-04-13:
For very important controllers like this one, we can do exceptions and have a userspace driver. In an ideal world, we would keep all this stuff kernel side.

Anyway, the real blocker here is that I don't own the Steam Controller. If one of you own the device, can you try adding the support for it and send a pull request please?

johnjelinek, 2016-04-13:
I own it, but the Python driver gets a little buggy sometimes where it
thinks there are multiple Xbox controllers plugged in. There's another
driver that is a kernel module, but it has different issues. For most
games, it works really well though.

aboutafter, 2016-04-13:
How about this one I've just saw on reddit:

https://www.reddit.com/r/linux_gaming/comments/4eneek/steam_controller_daemon/
https://gitlab.com/dennis-hamester/scd

johnjelinek, 2016-04-14:
I'll have to try that one out, what's the deploy story for a c++ userland
driver?

Kivutar, 2016-08-25:
I just received my steam controller. I got it working in basic keyboard+mouse mode, with steam launched. I'm going to try the various userland drivers now.

Kivutar, 2016-08-25:
I plugged the controller dongle, paired it with the central button, then launched Steam.

I launched RetroArch, saw a yellow message saying that an XBOX 360 controller was recognized.

I did a Bind All, and was able to map some of the keys:
 - the directions to the analog pad
 - ABXY to BAYX
 - start and select
 - L and R

I couldn't bind any of the L2 and R2 and L3 and R3. No right analog either. And the directional cross of the steam controller was unresponsive too.

That was enough to play SNES games.

tolkonepiu, 2017-06-12:
https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/addons/driver/steamcontroller/package.mk