libretro/Lakka #280: [enhancement] ODROID XU3/XU4 Fan Control

fluffymadness, 2016-01-13:
Modified an ODROID XU3 Lite fancontrol script to work with lakka.
I was annoyed at the constant switching between fan on/fan off of my XU4.
This needs to be tested more thoroughly, so use it at your own risk.
More info in the readme on git
https://github.com/fluffymadness/odroid-xu3-fan-control

Kivutar, 2016-01-14:
Ah good, thank you.

I will try it on mine.

I documented it here https://github.com/libretro/Lakka/wiki/Odroid-XU3

Let me know if it is really safe, I will then include it and enable it on the images.

fluffymadness, 2016-01-14:
Do you have a XU3 or a XU3-Lite, and are they identical, when it comes to the fan, and the system device for the fan control?
I'm asking because I edited the wiki page for it, and I want to know if I can write XU3-Lite/XU3/XU4 for the script.

Kivutar, 2016-01-15:
I have an XU3. I don't know if they are identical about the fan. You could ask mdrjr on our IRC, he's working at hardkernel.

fluffymadness, 2016-01-29:
Ok. Meveric told me that all models have the same fan.
http://forum.odroid.com/viewtopic.php?f=93&t=18597
I tested it for a few weeks now...it seems I have to use even higher values than 70, because sometimes the fan didn't start for me...I ordered a more silent fan now.
Then I have to cut open my case or modify the stock-case-printing-file with blender or so to make it higher, so that the fan can fit, and print it out ^^

Kivutar, 2016-01-30:
What fan did you buy?

fluffymadness, 2016-01-30:
The fan from this blogpost
http://www.bitkistl.com/2015/09/noctua-nf-a4-flx-5v.html?m=1
It has a 17.9db rating.
I should get it on Monday I guess :)

fluffymadness, 2016-02-06:
Here's an interesting read about xu4 cooling 
http://forum.odroid.com/viewtopic.php?f=97&t=15252&start=50#p123087.

I think that is one of the most interesting parts in this thread:
...
The 1.9GHz step uses over 20% less power than the 2.0GHz step for 4-5% less performance and can usually be run unthrottled with very minor modifications (replacing the TIM is often enough). Basically the 2.0GHz step is just a 'turbo' clockspeed for low threaded workloads and 1.9GHz is the actual sustained clockspeed the device can hold when fully loaded, which seems to be true even when using 3rd party cooling solutions.
...

danfozzy, 2016-02-12:
can you post install instructions for this to work for lakka please?

fluffymadness, 2016-02-12:
the fan script?
Ok. But this won't make you happy when you are going to use the stock fan.

Sometimes the stock fan doesn't even start for me and just gives off a high frequency noise.
So use it at your own risk, cause this is not tested thoroughly.

Here we go:
put the script in the following folder
/storage/.config/odroid-xu3-fan-control.sh

connect via ssh and give it executable permissions via
chmod +x odroid-xu3-fan-control.sh

put the fan.service file in
/storage/.config/system.d folder

then enable the file with 
systemctl enable fan 
to start the script automatically on every boot

to start the script immediately 
systemctl start fan
If its working the fan should slow down a bit and that annoying continuously on/off  switching should stop.

danfozzy, 2016-02-14:
I've just bought that fan that you have mentioned in a previous post, will this fan be OK with this script?

fluffymadness, 2016-02-14:
According to this post, it should work without a problem
http://forum.odroid.com/viewtopic.php?f=65&t=15310
I still have to try it out.
I didnt do it for now, because the original case is not high enough for the new fan, and i would have to use the connector of the stock fan and do some soldering work..
http://www.bitkistl.com/2015/09/noctua-nf-a4-flx-5v.html?m=1
Right now I',m looking for an easy way to modify the height of one of the xu4 cases that are floating around the net and to print it out via 3d print

danfozzy, 2016-02-14:
would be very interested in the 3d printing idea! ill give it ago tonight, the new fan and the script that is :)

danfozzy, 2016-02-14:
Just fitted my fan. Modded my case. This fan is super quiet can even here it, not even gonna bother with the script! 

Just need to find a case that will fit now

danfozzy, 2016-02-14:
Just seen that the autocad images are provided on hard kernels site, if we could increase the hight by 5mm on the top case it would fit a treat

fluffymadness, 2016-02-14:
I know that with the original case files, there are some cases on thingiverse too.
And I think increasing the height would make many people happy :)
Since this would make a much cleaner solution than cutting out a window for the fan.
Are you versed with autocad, cause I have never used it before.
I did some things with blender a year ago, but still editing the converted xu4-case 3d model seems a bit of a pain ^^

danfozzy, 2016-02-15:
autocad is something ive not played with before, but i do know someone who has, who im sure will do what we need. ill ask him to create the case lid 5mm taller and see how that pans out. anything else or anyother idea?

fluffymadness, 2016-02-15:
that sounds awesome :)
>>anything else or anyother idea?
No, not really, but I wonder if it has a measurable impact on cooling, now that there is a hole under the new fan.

danfozzy, 2016-02-15:
you mean in the heatsink?

fluffymadness, 2016-02-15:
yes. this one, cause it is not a flat surface under the fan.

Kivutar, 2016-10-17:
Hi, should we include this fan script in the distro, or tell users to go for a better fan?

fluffymadness, 2016-10-17:
Hi Kivutar,
the problem was, on my xu4, that the stock fan sometimes didn't start if the fan speed value was too low :/
So the lower values should be chosen high enough for this not to happen.
But if the values are higher, this thing is going to be more noisy.
I had the best experience with this mod http://forum.odroid.com/viewtopic.php?f=97&t=18211 and the custom fan script.
Maybe there's already a better solution for this, haven't searched for a while.

Kivutar, 2016-10-17:
Hardkernel sent me some passive cooling, but I wasn't able to remove the fan.

fluffymadness, 2016-10-17:
Ah ok, theres a thread on, http://forum.odroid.com/viewtopic.php?f=97&t=15252&start=200,
where they discussion alternative cooling solutions.
Since the odroid should be maxed out most of the time when doing emulation, I wonder if passive cooling would even work for longer time periods without any throttling.

commit 907551de (2016-12-28): XU fan control. Fixes #565 Fixes #280
closed 2016-12-28 by Kivutar