libretro/Lakka #281: Mame Core XU3/XU4

fluffymadness, 2016-01-14:
Is there a working mame core for odroid?

I found this url via a quick forum search
http://buildbot.libretro.com/nightly/linux/armhf/latest/
and it should probably be the right one for the odroid armhf->armv7
Downloaded the mame2014 core put it under /tmp/cores/ and gave it the correct permissions.
I tried to use it in lakka via LoadCore Arcade(Mame2014)
Load Content to load the rom.

Since i didn't get ingame I probably did something wrong.
How can I get some verbose output to see if I'm using the correct rom files, or missing some bios files.

FBA is still missing some good games, like Firetrap :), so if there is a decent working core, could it be included as default for the odroid?

Kivutar, 2016-01-14:
The problem with recent versions of mame is that the binary weight more than 50mb, and our first partitions have a limited size. Also I would like to keep the OS images at a reasonable size, because storing them and archiving them have a monetary cost.

To get a verbose output of retroarch, just loggin using ssh, stop the retroarch service, and launch retroarch manually.

Note that some cores from the buildbot will bug in Lakka anyway (we couldn't determine why yet).

Also, ensure that you're using the appropriate romset. Each version of MAME require a specific rom format.

fluffymadness, 2016-01-14:
ah ok, thx for the quick response.

fluffymadness, 2016-01-16:
Some prebuilt armhf cores that don't stackfault.
http://oph.mdrjr.net/meveric/pool/main/r/retroarch-cores-good/
Using mame_libretro.so atm.
R-Type Leo + Firetrap hehe :)

closed 2016-01-16 by fluffymadness