libretro/Lakka #282: Error on boot

rafaelhipercg, 2016-01-15:
when I turn on this mesagem appears in my cubieboard 2:
sh: can't access tty; job control turned off
I tried to install on several cards, but still giving error.
I used the Lakka-a20.arm-devel-20151223100729-r21250-gd666dcf.img image

rafaelhipercg, 2016-01-15:
I tested the Lakka-a20.arm-devel-20160109140320-r21276-g4dadcab.img and it worked

Kivutar, 2016-01-15:
Thanks, I will make a new release soon then!

closed 2016-01-15 by Kivutar

lookshe, 2016-04-02:
Same error occured for Wetek since following image: Lakka-WeTek_Play.arm-devel-20160128094544-r21301-g6429dc7.img
The image before works fine, but mentioned one and all afterwards result in "sh: can't access tty; job control turned off"

Kivutar, 2016-04-02:
Hello, this error "sh: can't access tty; job control turned off" can happen when flashing lakka on the SD card fails. You just need to reflash a second time.

lookshe, 2016-04-03:
I flashed it multiple times and tested all images from the mentioned one to the latest one. A few images of them were tested on two different SD cards. So for me it seems to be a problem of the images, not the flashing, because flashing older images always works.

Furthermore I always ran sync after the dd-command.

Kivutar, 2016-04-04:
I donwloaded the Wetek Play release, Lakka-WeTek_Play.arm-devel-20160320105135-r21382-g7f380b7.img

Flashed it, and it booted.

lookshe, 2016-08-11:
I tested a newer release (Lakka-WeTek_Play.arm-devel-20160807215525-r21557-gd292da5) and flashed it with Etcher, as described in #460, but it also gave the same error. Since this problem exists after a specific build, I assume that there were some changes, that lead to this problem, also if it workes for some people.

Kivutar, 2016-08-11:
Ok I try to flash the last release on my WeTek Play

Kivutar, 2016-08-11:
I flashed the release 20160715 using Etcher and it booted fine.

lookshe, 2016-08-12:
So if it works for you, there could not be a problem? As I mentioned earlier, I can confirm, that since Lakka-WeTek_Play.arm-devel-20160128094544-r21301-g6429dc7.img none of the releases worked for me. I flashed using different tools on different MicroSD-cards, but reproduceable it does not work with my WeTek.
I am trying to get help to solve this issue, are there any possibilies to go one from this point using the busybox or any other ways? E.g. building an image for my own and checking all changes between last working and first not working image?

Kivutar, 2016-08-12:
I just confirmed that it works for me, it doesn't mean that I deny the bug.

Maybe we don't have the same model of this hardware.

I suggest you to read the compilation guide and try to compile every commit after the release that was still working. If you don't know how to do, you can join us on IRC and get live help.

Can you create a new bug report please? This one was about a20 not WeTek.