libretro/Lakka #284: Boxart image

lollo78, 2016-01-16:
Hi, I don't know if someone has already done, 
but I would like continue the great lordashram job (https://github.com/lordashram/boxart) and adapt it for Retroarch/Lakka community to create a "boxart images set", ready for download and use.

In the initial phase, I'll work on No-Intro romset 2016-01-03:
https://archive.org/details/No-Intro-Collection_2016-01-03

I started with Sega 32x, and I stopped quickly due an issue: the images appear cutted on the bottom:
[image]

I tried to reduce the size, but the issue remains.
[image]
[image]
[image]

This issue not appear with a simple, flat cover images/screenshots.

So, before to loose time to rename all the images and create the missing ones...
I would like to ask you if there is a way to fix this...

Kivutar, 2016-01-17:
First, thanks a lot for this initiative! We really need this kind of contributions.

I see what is the source of this bug. When I designed this feature, I never expected to have images so high compared to their width. I'm talking about the aspect ratio.

If you try with SNES boxarts, you will see the difference.

I will find another strategy to display those images. It has to be done in C in RetroArch.

So you can continue working on this renaming task without problem.

What do you think about consolidating these with the screenshot+titles packs that we already use?

We could have one github repository, containing all the images. With a naming convention for folders, like the one used by https://screenshotpalace.wordpress.com/no-intro-screenshot-reloaded/

lollo78, 2016-01-18:
Thanks to you for all the great work.

OK, I'll continue to rename the images, waiting for a code fix.

Consolidating the job with existent pack and create a github repository is a good idea.
Let me know where I have to upload the files on github (in the meantime I'll work on my PC).

In future you could add an option inside Retroarch to download this content (from retroarch/lakka website or where you want) and give the possibility to all who want them to download directly from Retroarch GUI.

Kivutar, 2016-01-25:
I wrote a little scrapper to get boxarts from thegamedb.

lollo78, 2016-01-26:
It will be included in Retroarch bin or only in Lakka system?

Kivutar, 2016-01-26:
I will be a 20 line perl script, that I will put somewhere on github.

First thing to do is downloading and extracting the complete pack of **no-intro screenshot reloaded** and version it on github. (Their dropbox allows us to download all at once)

Then, we can start adding new images like boxarts to it, while keeping track of the changes.

Tell me if you want to do it. If not, I can also do it. I already created a repo, but my internet connexion was too instable to download those packs today.

lollo78, 2016-01-27:
OK, I'll try to do it. 
Please, send me the github repo and some info on which folder put the new boxarts.

I just finished to download an huge collection of HD covers (acquired from scanner) to create the 3D boxart... 
like this
[image]

but I think that this is not be the right solution for us, because the "full covers" is not available for all the games. So, create the 3D boxart became an huge and frustrating job.

Also, in my test, the 3D boxart doesn't look very well in Retroarch GUI, at the same way of a "flat cover" 
like this:
[image]

For this reason, I have an idea: 
Could you add a new function that for any "flat image" Retroarch applies automatically this perspective view and a tiny shadow in the bottom-right corner?
like this
[image]

I tried with Photoshop and other perspective doesn't look better than this in retroarch GUI.
[image]? 

Doing this, every guys in the world could add a "flat cover" in the repo and help with the project.

Kivutar, 2016-01-27:
Yes, having flat images in a first time would be easier for many reasons, including that most online database have flat covers that we can scrape.

I added you to this repo : https://github.com/lakkatv/no-intro-pictures

I'm going to do a first commit so you will get how we have to organize the files.

Then, I will try to code a quick perl script to grab some front covers from thegamesdb.

Kivutar, 2016-01-27:
Looks like tgdb is down tonight http://thegamesdb.net/api/GetGame.php?id=1253

I'm going to push what I have anyway.

lollo78, 2016-01-27:
OK!
Please, consider to include my request in Retroarch (documented above).
It's true that find a flat cover online is more simple, but leave a flat cover (as it is) in Retroarch GUI is... horrible :D 

A little perspective to the inside and a little shadow to the outside makes the GUI more dynamic.

Please, let me know if this can be done automatically from Retroarch, so I know how to start. 
Cutting only the front cover and rename it is a job... add the perspective and the shadows is onther one (sorry, but I have little free time).

lollo78, 2016-01-27:
Noob question :) I'm ready to make the pull request with alla the no-intro images. I cloned the repo, committed more than 30000 files but now I can't create the pull request from master to master.
I have to fork the repo first?

Kivutar, 2016-01-27:
The perpective should be doable. The shadow is more difficult. Generating 3D covers from flat covers with a script is definitely possible too.

I added you to the commiters of the repo. A simple push will upload all the changes. You don't need to do a pull request since I gave you the most direct access ;)

lollo78, 2016-01-28:
Upload done!
After a lot of research, I think that "3D covers" could be a waste of time.
Full covers aren't available for all the game. So, some "3D cover" side must be manually recreated: the result could be false and in any case different to the original.
A flat (good HD) cover, cleaned and retouched with Photoshop, could be better.
OK for the perspective. Instead of shadows you could add a reflection effects like this: 
[image]
[image]
More nice to see and maybe more simple to implement.

Kivutar, 2016-01-28:
Perfect, thanks for this upload!

Looks like thegamesdb is up again.

I'm going to try my scraper and see if enough roms are covered.

Then, we will think about displaying that in xmb and glui.

lollo78, 2016-01-28:
Thx to you!
Some cover from tgdb is not good in quality, but it's a good starting point. I downloaded a fine collection of full covers acquired from scanner devices. I'll cut from them the front covers in Photoshop ... But first, it is a lovely idea to try with your scraper.
Let me know, when you are ready with the perspective implementation, at witch resolution I have to cut the png images.
About the reflection... It's the same cover image with a little transparency and a differenti position. Maybe I wrong, but I  thinked that this solution could be more simple than a shadow.

Kivutar, 2016-01-29:
Ok, you can do a git pull and you will get my latest changes.

Yeah, some covers from tgdb are very bad in quality. Nothing can beat a manual work here. However, there are more than 30k roms supported by libretro, so doing this manually may seem funny at the beginning, but after a few month you will feel tired.

The coverage of the script is clearly not enough. Certainly due to this hash.csv that I took from recalbox. I need to ask them how they generate this file.

You can try the script by yourself if you're on Linux.

For example, only 2 of the 5 supergrafx roms obtain a cover. And for SNES, something like 40%. And for NES less than that.

About the file size now, having a small size will make the cover ugly on 1080p screens. Having huge images will make the loading slow on weak hardware.

Sizes between 256 and 512 seams reasonable. What do you think about it?

lollo78, 2016-01-29:
I don't know if this can help but I remember that Advanced Launcher works fine with tgdb. Maybe you could find some "usefull things" to create your own script also here: https://github.com/lollo78/plugin.program.advanced.launcher/tree/master/resources/scrapers/datas/TheGamesDB

For the size I have to do some test... I'll try to find the best solution.

Kivutar, 2016-01-29:
Ok. I've started uploading some boxarts using the scrapper. Some platforms have a pretty good coverage finally.

lollo78, 2016-01-29:
Maybe I found a solution for the ".png size issue".
We could compress the covers with https://pngquant.org/ algorithm, that preserve also the alpha transparency.

Find the differences in this two images... and after, check the size:
[image]
[image]

Note: if you use the compressed image above with Retroarch, you will lose the transparency (retroarch doesn't support it, but Windows, Linux, OSX and all the new browsers yes).
But this is not a problem for us, because all the perspective effects, shadows/reflections, white frame and so on... will be applied directly from Retroarch on a flat cover image (that we can compress like we want :D).

Kivutar, 2016-01-30:
They just switch to indexed colors right? I didn't know that retroarch was supporting indexed colors. Are you sure?

Kivutar, 2016-02-02:
Hi @lollo78 would you have some time to help me scrape some box arts using that script?
My internet connection sucks.

lollo78, 2016-02-02:
Hi dear! I had some issue with real life, but I'm here. Don't worry. Ok for the help. I'll try the script as soon as possibile. I hope to find some free time from tomorrow.
For the image... I don't know if it switch only to indexed colours, but if you try the smallest image above in retroarch you will see a black square under the image. So, retroarch doesn't support it.

Kivutar, 2016-02-03:
The smallest image does use indexed mode. And RetroArch will not support it soon.

lollo78, 2016-02-03:
What a pity! So the HD covers will be very huge...
I think that add the indexed mode in retroarch is an important step for this "boxart project".
At least to have a nice view.

Some consideration about our work...
On Internet Archive I see a new fixed version of No-Intro Collection :
https://archive.org/details/No-Intro-Collection_2016-01-03_Fixed (completed with the missing platforms: now the .zip file is over 34 GB o_O :D)

I notice that all the .dat files inside our platform folders are very old. So, if we use this .dat with your script to rename the covers, some of the images could use a different name and will not appear inside Retroarch for who use a more updated collections.

To make a good work, I think that it's better to update first all the .dat file according the last No-Intro-Collection (the "fixed" version was released on 21th january 2016). 
But... where we can find the exactly .dat file for this? I don't know. Can you help?

My suggestion is to focalize our work on No-Intro-Collection_2016-01-03_Fixed, ignoring the newest collections released from the No-Intro Team after this, at least until we reach a "near complete" boxart collection. When we are ready we could think to update to a next version. 

What do you think?

Kivutar, 2016-02-03:
Yes, I'm not sure of those dat changed a lot, but upgrading them can't be bad. They are in this repo https://github.com/libretro/libretro-database

Latest DATs can be found here http://datomatic.no-intro.org/ This is the reference.

Be careful, there is a problem with NES headers. Double check that ROMs can launch if you use the new DAT and the new ROMs.

lollo78, 2016-02-03:
Please, a clarification.
In our repo we have, for example,  Atari - 5200 (20130303-220600_RC).dat, but from http://datomatic.no-intro.org/ I can download only Atari - 5200 (20130303-220600_CM).dat
I think that CM stands for "ClrMamePro", right? But what means "RC"?
I see also that on https://github.com/libretro/libretro-database you use the CM version. So I'll try with the CM versions.

Kivutar, 2016-02-03:
Good question. I don't know. Could you ask @twinaphex on #retroarch on irc?

garnieretienne, 2016-02-03:
| I think that CM stands for "ClrMamePro", right? But what means "RC"?

RomCenter maybe?

lollo78, 2016-02-05:
Hi garnieretienne, thanks a lot for the suggestion.
@kivutar: I'm trying to use the scraper with Ubuntu 15.05 but it doen't work for me.
I have an issue with the Perl interpreter: 
```
lollo78@lollo78-P55A-UD4:/media/lollo78/TOURO/no-intro-pictures$ ./tgdb.pl Atari\ -\ 5200/Atari\ -\ 5200\ \(20130303-220600_CM\).dat 
bash: ./tgdb.pl: /usr/bin/perl^M: interprete errato: File o directory non esistente
```
This is my perl -V
```
lollo78@lollo78-P55A-UD4:/media/lollo78/TOURO/no-intro-pictures$ perl -V
Summary of my perl5 (revision 5 version 20 subversion 2) configuration:
   
  Platform:
    osname=linux, osvers=3.13.0-76-generic, archname=x86_64-linux-gnu-thread-multi
    uname='linux lcy01-10 3.13.0-76-generic #120-ubuntu smp mon jan 18 15:59:10 utc 2016 x86_64 x86_64 x86_64 gnulinux '
    config_args='-Dusethreads -Duselargefiles -Dccflags=-DDEBIAN -D_FORTIFY_SOURCE=2 -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -Dldflags= -Wl,-Bsymbolic-functions -Wl,-z,relro -Dlddlflags=-shared -Wl,-Bsymbolic-functions -Wl,-z,relro -Dcccdlflags=-fPIC -Darchname=x86_64-linux-gnu -Dprefix=/usr -Dprivlib=/usr/share/perl/5.20 -Darchlib=/usr/lib/x86_64-linux-gnu/perl/5.20 -Dvendorprefix=/usr -Dvendorlib=/usr/share/perl5 -Dvendorarch=/usr/lib/x86_64-linux-gnu/perl5/5.20 -Dsiteprefix=/usr/local -Dsitelib=/usr/local/share/perl/5.20.2 -Dsitearch=/usr/local/lib/x86_64-linux-gnu/perl/5.20.2 -Dman1dir=/usr/share/man/man1 -Dman3dir=/usr/share/man/man3 -Dsiteman1dir=/usr/local/man/man1 -Dsiteman3dir=/usr/local/man/man3 -Duse64bitint -Dman1ext=1 -Dman3ext=3perl -Dpager=/usr/bin/sensible-pager -Uafs -Ud_csh -Ud_ualarm -Uusesfio -Uusenm -Ui_libutil -Uversiononly -DDEBUGGING=-g -Doptimize=-O2 -Duseshrplib -Dlibperl=libperl.so.5.20.2 -des'
    hint=recommended, useposix=true, d_sigaction=define
    useithreads=define, usemultiplicity=define
    use64bitint=define, use64bitall=define, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cc', ccflags ='-D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64',
    optimize='-O2 -g',
    cppflags='-D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include'
    ccversion='', gccversion='4.9.2', gccosandvers=''
    intsize=4, longsize=8, ptrsize=8, doublesize=8, byteorder=12345678
    d_longlong=define, longlongsize=8, d_longdbl=define, longdblsize=16
    ivtype='long', ivsize=8, nvtype='double', nvsize=8, Off_t='off_t', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='cc', ldflags =' -fstack-protector -L/usr/local/lib'
    libpth=/usr/local/lib /usr/lib/gcc/x86_64-linux-gnu/4.9/include-fixed /usr/include/x86_64-linux-gnu /usr/lib /lib/x86_64-linux-gnu /lib/../lib /usr/lib/x86_64-linux-gnu /usr/lib/../lib /lib
    libs=-lgdbm -lgdbm_compat -ldb -ldl -lm -lpthread -lc -lcrypt
    perllibs=-ldl -lm -lpthread -lc -lcrypt
    libc=libc-2.21.so, so=so, useshrplib=true, libperl=libperl.so.5.20
    gnulibc_version='2.21'
  Dynamic Linking:
    dlsrc=dl_dlopen.xs, dlext=so, d_dlsymun=undef, ccdlflags='-Wl,-E'
    cccdlflags='-fPIC', lddlflags='-shared -L/usr/local/lib -fstack-protector'

Characteristics of this binary (from libperl): 
  Compile-time options: HAS_TIMES MULTIPLICITY PERLIO_LAYERS
                        PERL_DONT_CREATE_GVSV
                        PERL_HASH_FUNC_ONE_AT_A_TIME_HARD
                        PERL_IMPLICIT_CONTEXT PERL_MALLOC_WRAP
                        PERL_NEW_COPY_ON_WRITE PERL_PRESERVE_IVUV
                        USE_64_BIT_ALL USE_64_BIT_INT USE_ITHREADS
                        USE_LARGE_FILES USE_LOCALE USE_LOCALE_COLLATE
                        USE_LOCALE_CTYPE USE_LOCALE_NUMERIC USE_PERLIO
                        USE_PERL_ATOF USE_REENTRANT_API
  Locally applied patches:
	DEBPKG:debian/cpan_definstalldirs - Provide a sensible INSTALLDIRS default for modules installed from CPAN.
	DEBPKG:debian/db_file_ver - http://bugs.debian.org/340047 Remove overly restrictive DB_File version check.
	DEBPKG:debian/doc_info - Replace generic man(1) instructions with Debian-specific information.
	DEBPKG:debian/enc2xs_inc - http://bugs.debian.org/290336 Tweak enc2xs to follow symlinks and ignore missing @INC directories.
	DEBPKG:debian/errno_ver - http://bugs.debian.org/343351 Remove Errno version check due to upgrade problems with long-running processes.
	DEBPKG:debian/libperl_embed_doc - http://bugs.debian.org/186778 Note that libperl-dev package is required for embedded linking
	DEBPKG:fixes/respect_umask - Respect umask during installation
	DEBPKG:debian/writable_site_dirs - Set umask approproately for site install directories
	DEBPKG:debian/extutils_set_libperl_path - EU:MM: set location of libperl.a under /usr/lib
	DEBPKG:debian/no_packlist_perllocal - Don't install .packlist or perllocal.pod for perl or vendor
	DEBPKG:debian/prefix_changes - Fiddle with *PREFIX and variables written to the makefile
	DEBPKG:debian/fakeroot - Postpone LD_LIBRARY_PATH evaluation to the binary targets.
	DEBPKG:debian/instmodsh_doc - Debian policy doesn't install .packlist files for core or vendor.
	DEBPKG:debian/ld_run_path - Remove standard libs from LD_RUN_PATH as per Debian policy.
	DEBPKG:debian/libnet_config_path - Set location of libnet.cfg to /etc/perl/Net as /usr may not be writable.
	DEBPKG:debian/mod_paths - Tweak @INC ordering for Debian
[... cut ...]
	DEBPKG:fixes/CVE-2015-8607_file_spec_taint_fix - ensure File::Spec::canonpath() preserves taint
  Built under linux
  Compiled at Jan 20 2016 16:40:58
  @INC:
    /etc/perl
    /usr/local/lib/x86_64-linux-gnu/perl/5.20.2
    /usr/local/share/perl/5.20.2
    /usr/lib/x86_64-linux-gnu/perl5/5.20
    /usr/share/perl5
    /usr/lib/x86_64-linux-gnu/perl/5.20
    /usr/share/perl/5.20
    /usr/local/lib/site_perl
    .

```
I'm going to upgrade to 15.10.
Or maybe I'm doing something wrong.
Any advice?
:*

Kivutar, 2016-02-06:
mmmmmmh. Your dat have to be named like the one of RetroArch : Atari - 5200.dat

lollo78, 2016-02-06:
I deleted the Named_Boxart folder of the repo to test again. 
I notice a strange issue.
If I use this file name: **Atari - 5200 (20130303-220600_CM).dat** I see only the sha1 keys in the bash window, but the Named_Boxart folder is not created.
If I rename the same file in **Atari - 5200.dat** I continue to see the sha1 keys list but now the Named_Boxart folder is created correctly, but there aren't image inside it.

Kivutar, 2016-02-06:
That's because hash.csv (a file I found in the recalbox scrapper) lacks atari 5200 entries. Let try with something like sega genesis and it should work.

I need to contact the guys of recalbox, to know how they generated this hash.csv, to make it complete.

This file gives us the TGDB id for each rom. But it doesn't cover all roms yet, as you can see.

lollo78, 2016-02-06:
yes, with sega genesis works.

lollo78, 2016-02-06:
Just a little question... 
what is the purpose of the file hash.csv?
we can leave without it?

Kivutar, 2016-02-06:
It's the mapping of no-intro sha1 to tgdb IDs. So we can't live without it. Until we find an alternative.

lollo78, 2016-02-06:
So, also if I use the new CM .dat could be that some game will not be recognized and scraped correctly  - if the new roms have different sha1 of hash.csv -?

Kivutar, 2016-02-06:
Yeah, I mailed the maintainer of that file. I'm asking him this kind of information. So we will know soon.

lollo78, 2016-02-06:
The scaper  - Sega Genesis -  just stopped. This is normal or it crashed?
```
mismatched tag at line 22, column 2, byte 1566 at /usr/lib/x86_64-linux-gnu/perl5/5.20/XML/Parser.pm line 187.
XML::Simple called at ./tgdb.pl line 69.
```
Inside the folder I have 1011 jpg  - but if I remember well the roms are over 1100 -  and a lot of this image seems corrupted in Linux> Error interpreting JPEG image file.
The strange things is that I upload an image here it woks
like this
[image]

Kivutar, 2016-02-06:
Well my script is "quick and dirty". It names all files with the jpg extension. Even when they are png. But then, since all are converted to png afterward, it fixes the problem.

The script can fail if the tgdb website is down for a second, or if there is an internet connexion issue.

To start again where it failed, you can uncomment those lines https://github.com/lakkatv/no-intro-pictures/blob/master/tgdb.pl#L57-L59 , and change the 'S' with the letter you want to start from.

lollo78, 2016-02-06:
Nice tips. Thanks!
So, I have to convert all in .png before to upload on the repo, right?
Have you used some particular tool, settings or compression? Do you want that I use some of this?

Kivutar, 2016-02-06:
Yes, it's better. There is an example in the README, with the mogrify command.

lollo78, 2016-02-06:
Very nice! 
I think that now I have all the infos to do a nice job. 
Of couse, If I'll have some issue... I'll broken your balls again :smiley: 
Have a good weekend Kivutar! 
:kissing:

lollo78, 2016-02-06:
Here I am! 
Balls mode on 
:D

In the following example there are 4 sha1 key but only one image, right?
```
sha1 fcb1ef7398b842ebd28c3227852d7a132ce7b887
1278
http://thegamesdb.net/banners/boxart/original/front/1278-1.jpg
File downloaded and saved!!!
sha1 9fb4cfd5f8b1ea4239c38e8a43bc3f657cfccdca

sha1 22bab863746f0903ac3b8187ffed288ff7ceeae3

sha1 3b6e0ef1458c883799923f6b86d106b488b5c6a3
```
This means that for some sha1 there isn't an image on TGDB, so the script skip them. Right?

Could you add an if cycle in the script to do this?
If for this sha1 there isn't an image on TGBD > Than create a TXT file inside the Named_Boxarts (blank/empity inside, but with the game name like the other images scraped).

This could help to understand which images are missing and could help a lot for a manual integration. 
Open the .dat with a text editor to find the right name of the game to rename a cropped image is... a crazy job.
Thx

lollo78, 2016-02-06:
Maybe this project could help to create a new and more powerfull scraper:
https://github.com/sselph/scraper

twinaphex, 2016-02-06:
Anything that is not C-based, can never be useful for Lakka itself.

Kivutar, 2016-02-06:
@lollo78 I'm aware of this scrapper. I already read its source, and the hash.csv comes from this guy. It is completely different to our approach : networkless scanning.

@twinaphex He's just running a script to download the boxarts and upload them to a git repo. It's a one time task, that he will run on his linux, not on Lakka. Zero dependancy.

The repo is there: https://github.com/lakkatv/no-intro-pictures

It is built with the idea that users can just download it and extract it in their boxart dir.

twinaphex, 2016-02-06:
@kivutar OK, that can work as well. I was thinking they wanted an online scraper, but yeah, an offline collection of boxart/artwork and pictures might be best. That way we don't have to require an Internet connection either.

lollo78, 2016-02-06:
@twinaphex, hi! Thanks a lot for all your hard work on Retroarch. 
Yes, this is only a one time task on local PC to create a boxart collection for all the lakka/retroarch users. No strange request to add a scraper function inside Retroarch. Also, it's not necessary if we create the github collection linked above from Kivutar.
So... the only dependencies for me remains girls, wine and smokes. :D
:*

@Kivutar: thanks a lot for log the missing images. Let me know when you are ready! ;) 
If it's possible, to speed up the process (I just scraped all), please include a routine that if a name file is already present into the folder (indipendently from extension: PNG, JPEG, TXT) the script skips the download. So, only the TXTs for the missing images will be created (please, don't let me download all over again)...
  
I tested all. So, just to inform you, the following platforms doesn't work with the script, than they require something in hash.csv:
Atari - 5200
Atari - 7800
Atari - Jaguar
Bandai - WonderSwan
Bandai - WonderSwan Color
Casio - PV-1000
Coleco - ColecoVision
Commodore - 64
Commodore - Plus-4
Commodore - VIC-20
Emerson - Arcadia 2001
Entex - Adventure Vision
Epoch - Super Cassette Vision
Fairchild - Channel F
Funtech - Super Acan
GamePark - GP32
GCE - Vectrex
Hartung - Game Master
Magnavox - Odyssey2
Microsoft - MSX
Microsoft - MSX 2
Nintendo - Famicom Disk System
Nintendo - Pokemon Mini
Nintendo - Virtual Boy
Philips - Videopac+
RCA - Studio II
Sega - PICO
Sega - SG-1000
SNK - Neo Geo Pocket
SNK - Neo Geo Pocket Color
Tiger - Game.com
VTech - CreatiVision
Watara - Supervision

For all the other platforms I scraped all the image and converted it in PNG (without resize: it's a pity!).
I'll upload all on the repo this monday. At job I have a nice connection with a good upload bandwidth (at home... a turtle is more speedy).
See you soon

albanibr, 2016-02-09:
It's complete? I do mirrors! my connection is good

lollo78, 2016-02-09:
I had some issue with work connection in this days and an error from git about .lock file. Hope to sole tomorrow.

Kivutar, 2016-02-10:
@albanibr It's not complete yet, but you can clone the repo and do a git pull regularly. We appreciate mirrors.

lollo78, 2016-02-12:
OMG I have difficulties to push all the modify done to the repo.
Maybe the upload is very huge.
I'm trying from a week to do that from work and home... without success.
This is the last error:
git -c diff.mnemonicprefix=false -c core.quotepath=false push -v --tags origin master:master
POST git-receive-pack (chunked)

fatal: The remote end hung up unexpectedly

fatal: The remote end hung up unexpectedly

error: RPC failed; result=55, HTTP code = 200

Pushing to https://lollo78@github.com/lakkatv/no-intro-pictures.git
Everything up-to-date

I'll try again with one folder at a time.
:(

lollo78, 2016-02-20:
Hi kivutar.
I'm doing the upload of the boxarts images, but I'm stucked on Super Nintendo.
I can't upload other images (for an error in git) and not all the images uploaded are shown into the repository.
On github 
https://github.com/lakkatv/no-intro-pictures/tree/master/Nintendo%20-%20Super%20Nintendo%20Entertainment%20System/Named_Boxarts
I see this alert: 
"Sorry, we had to truncate this directory to 1,000 files. 656 entries were omitted from the list."
Whats happened?

Kivutar, 2016-02-20:
Hi @lollo78 , don't worry about that message : it's just the web interface of github that is not able to display all the entries. But they are still in the repo!

lollo78, 2016-02-20:
OK, thx! 
My upload is near to be completed.
I know you're overworked, but there are news about the mods required (above) to the scraper code?
And news about hash.cvs to scrape the other platforms?

Kivutar, 2016-02-20:
Hi @lollo78 , yeah, I've been very busy with my everyday job recently. I've contacted the author of this hash.csv. Looks like we will have to complete it by ourselves. What mods are you talking about?

lollo78, 2016-02-20:
I can undestand you. 
We have to work to live... shit!
;)

For mods required...
You called this "logging for missing images". My request was:
> Could you add an if cycle in the script to do this?
> If for this sha1 there isn't an image on TGBD > Than create a TXT file inside the Named_Boxarts (blank/empity inside, but the file name must be the game name, like the other images scraped).

After I request also this:

> If it's possible, to speed up the process (I just scraped all), please include a routine that if a name file is already present into the folder (indipendently from extension: PNG, JPEG, TXT) the [...]

All this could help a lot!

lollo78, 2016-02-21:
I completed all the available upload!
Now I wait for your news to precede!

Kivutar, 2016-02-22:
Thanks a lot! I'm updating the http repo. I will do those changes, so we can know where we have to work next.

Kivutar, 2016-02-22:
I pushed the changes

lollo78, 2016-02-22:
I see the new script. I'll try it as soon as possible.
I just tried the new upload function of Github via Web browser (for NEC - Super Grafx).
But not works as expected.
It not overwite the existing file (but add the new ones with a different name).
It could be usefull to upload new image, but not to change old images with a better quaility images...
What a pity!

[37 more replies]

closed 2016-05-26 by Kivutar