libretro/Lakka #285: noobs setup partition size incorrect  (RPI2, possibly other builds)

mherberg, 2016-01-18:
It looks to me (although I don't have enough noobs experience) that the Lakka/config/noobs/partitions.json may have the incorrect sizes for partition_size_nominal and uncompressed_tarball_size for the System partition. Lakka_RPi2_System.tar.xz for example is 167MB uncompressed yet the json file doesn't account for this. It hasn't been updated in months which may explain this.

When I installed from noobs it actually was smart enough to make the partition the exact size of the uncompressed tarball. However when I tried to edit any files (such as config.txt) I could not save them because there was no wiggle room in the size of the partition. By adjusting these values I have solved the problem.

Kivutar, 2016-01-18:
Ok, I will check the values in OpenELEC and see if it differs. Thanks for the report.

mherberg, 2016-01-19:
Just FYI, OpenELEC (6.0 arm) does differ however it has a 105 MB SYSTEM file while Lakka has a 158 MB file. So the partitions.json will have to differ from OpenELEC. Just FYI the OpenELEC config.txt has changed so if you want you can put that in Lakka. The only discernible change is the overclock recommendations for the RPI2, but it is nice to have the settings there since most RPI2 users will overclock for emulation. Thanks!

commit 668d332d (2016-01-19): Change the partition size in noobs. Should fix #285
closed 2016-01-19 by Kivutar