libretro/Lakka #286: AllWinner A10 problems

lir448, 2016-01-21:
I have a device with a chip AllWinner A10. This is a copy of Mele a1000 (http://linux-sunxi.org/Mele_A1000 - A10, 1.2 Ghz, 512MB DDR2, Nand 4GB, SD, Sata, 3 USB 2.0, HDMI, etc). It is similar to Cubieboard, but with only 512 RAM. Lakka run on it, but there are several problems which is observed on two builds (20151223101136-r21250-gd666dcf.img.gz and 20160109143814-r21276-g4dadcab.img.gz):
 - wired network does not work, there is no connection with the router (I did not try to set up wifi via cli yet);
 - NES emulator works perfectly, but Sega (with Genesis Plus GX core) crashes to the desktop if you try open any rom;
 - Sony PS hangs on the splash screen of the game, because I can not set the path to the BIOS;
 - I can not check out the other core and upgrade existing ones, because the network does not work;
 - this problem (https://github.com/libretro/Lakka/issues/176) and this (https://github.com/libretro/Lakka/issues/274) exist too.

Hardware works well on the factory firmware (wired network, wireless, etc.), but then there is a delay on pressing the gamepad buttons in all emulators (in Retroarch too, even with different cores). I like the Lakka for user-friendly interface and no delay (NES is checked only, but it works perfectly). I hope for further development of the project and for more detailed instructions (manual, help, FAQ, wiki) on how to fine-tune the OS.

Kivutar, 2016-01-21:
Ok, I think that you need an alternate script.bin for this hardware. This may fix the ethernet problem. Let try to override the one in Lakka by the one in the factory firmware.

The Sega problem is certainly a ROM issue, try the ROMs from the no-intro set.

You can set the BIOS of the PSX by using alternative methods to access the filesystem https://github.com/libretro/Lakka/wiki/Accessing-Lakka-filesystem

lir448, 2016-01-24:
Thanks for the tip, I extracted script.bin from a factory firmware, and the network got up. Then I updated the kernel and Lakka read those roms that are not read before. So the problem was not in roms.

I found some more problems (tested on Lakka-a10.arm-devel-20160122025241-r21295-gbab6f2f.img.gz):

1) turbo buttons do not work in the NES with FCEUmm, but they work in RetroArch for Windows and Android with this core. Setting the turbo buttons is missing in the core input config.
2) I used gamepads logitech f310. They work in a mode xinput and do not work in directinput, although the system writes that sees them sometimes.

Kivutar, 2016-01-25:
Hello, good to learn that it worked.

1) I'm not sure about what you're calling "core input config". Are you
talking about the "per core input remappings" ? Or "joypad autoconfig" ?
None of our joypad autoconfig should include actions like turbo.

2) Ah I'm sorry, I always use udev, so I can't help you here.

Kivutar, 2016-01-25:
I think that we can close this. The other small bugs stated here are nothing A10 specific.

closed 2016-01-25 by Kivutar