libretro/Lakka #287: Official Wii U Gamecube Adapter not working

nvitaterna, 2016-01-22:
After looking around a bit i found this https://github.com/ToadKing/wii-u-gc-adapter. Unfortunately I cannot install it as make is not available. Wondering if this can get added in. Thanks.

Kivutar, 2016-03-14:
Do you see the gamepad appearing as /dev/js0 when you plug it?

Do you see it in the output of the lsusb command ?

Have you tried the manual configuration in RetroArch ?

hizzlekizzle, 2016-03-14:
Unless/until a driver for the USB adapter gets added to the kernel (dunno if anyone is working on that), you'll have to run a listening daemon, which ToadKing's program is/does. If that's a thing you want to add to Lakka, it would need to be run as root.

Kivutar, 2016-03-14:
Well, I would prefer to wait until a kernel side driver gets merged and backport the patches.

hizzlekizzle, 2016-03-14:
I agree. Adding a separate daemon just for this one piece of hardware would be unwise, i think.