libretro/Lakka #289: Question: Atom-based support

gbraad, 2016-01-24:
How about Atom-based boards as supported platforms?

Atom have an Intel based GPU and have decent performance, especially when compared to some of the ARM-based boards.

duduke, 2016-01-24:
I have both an nvidia-ion (atom330+gefroce9400m) and an old atom 230. It is supported and works.

Kivutar, 2016-01-24:
The generic image should work on those boards, if they have a BIOS.

closed 2016-03-14 by Kivutar

LukasBombach, 2016-08-07:
I'm getting 0.0 fps on a [Foxconn nT-535](http://www.foxconnchannel.com/ProductDetail.aspx?T=NanoPC&U=en-us0000006) with a Intel NM10 CPU / Intel Graphics Media Accelerator 3150, which seems kind of odd as Lakka runs smoothly on a much weaker Raspberry Pi. Not sure if it needs a graphics driver or something like that, but it runs like that "out of the box" after a new install.

duduke, 2016-08-07:
According to this: https://wiki.gentoo.org/wiki/Intel (look under Gen3) I believe that GMA3150 only supports OpenGL 1.5 whereas RetroArch needs atleast OpenGL 2.
P.S. the RPi has OpenGLES 2.0 which is mostly similar to desktop OpenGL 2.

LukasBombach, 2016-08-08:
That must be it, thank you!