libretro/Lakka #291: Nintendo not working

ricardojrgpimentel, 2016-01-25:
Hello
When i try to run a n64 game i only get a black screen and the system goes back to the main menu.

The terminal shows this:

    Jan 25 19:04:44 Lakka systemd[1]: retroarch.service: Main process exited, code=killed, status=11/SEGV
    Jan 25 19:04:44 Lakka systemd[1]: retroarch.service: Unit entered failed state.
    Jan 25 19:04:44 Lakka systemd[1]: retroarch.service: Failed with result 'signal'.0
    Jan 25 19:04:46 Lakka systemd[1]: retroarch.service: Service hold-off time over, scheduling restart.

Every other emulator works fine.

Kivutar, 2016-01-26:
Hello, are you using a 32bit Lakka for PC? Or something else?

ricardojrgpimentel, 2016-01-26:
Yes, I'm using the 32bit version

Kivutar, 2016-01-26:
Mupen64plus has always been failing on i386. There is already a bug report here about this. So I will close this one to keep just one thread.

closed 2016-01-26 by Kivutar