libretro/Lakka #292: ATI HDMI audio not working

aboutafter, 2016-01-25:
I think I can't reopen this issue: https://github.com/libretro/Lakka/issues/237

So I'll open a new one. I hope I'm not doing it wrong. Feel free to reopen that thread and close this if it's needed. 

The thing is that  radeon.audio=1 doesn't fix the issue for me. I changed audio_device="hdmi:CARD=HDMI,DEV=0" (the name aplay -L shows for my ATI card) and I don't have sound. 

This is in a PC with an AMD 5770.

Kivutar, 2016-01-26:
You did right.

Let keep the radeon.audio=1 even if it doesn't solve the problem, it will certainly fail without it.

After searching on google, I found that you must have radeon.audio=1, then reboot, and try every item returned by aplay -L.

Have you tried them all?

Don't forget to stop the retroarch service when you test that.

aboutafter, 2016-01-26:
This is going to sound very noobish, but how do I save the radeon.audio=1 line from the cmdline?

Kivutar, 2016-01-26:
Ah, this is documented here https://github.com/libretro/Lakka/wiki/PC
You have to remount the first partition, and edit the bootloader configuration.

aboutafter, 2016-01-27:
There is no extlinux.conf. I made the change in syslinux.conf, but if I press "tab" in the bootloader I don't see the radeon.audio=1 change in the line, so maybe it's not /flash/syslinux.conf the file that needed to be changed.

Kivutar, 2016-01-27:
If you see syslinux.conf, it means you are in live USB mode. /flash/syslinux.conf should be ok. I don't get why the changes are not saved.

Anyway, during the test phase, you can just add radeon.audio=1 manually, and try all the entries returned by aplay -L.

aboutafter, 2016-01-27:
Either way, the sound doesn't work no :(

albanibr, 2016-01-29:
I am also in trouble ...

My audio card is Nvidia...HDMI audio works, but to shutdown the system, have to modify the 48000 Hz Audio to 44100Hz or 44100Hz to 4800HZ. When it works randomly is muted for a few seconds.

ZOTAC GF9300-I-E LGA 775 NVIDIA GeForce 9300 HDMI Mini ITX Intel Motherboard and lakka 64bits

Any solution?

Kivutar, 2016-01-30:
What? You have to change the audio settings to shutdown the system???

albanibr, 2016-01-30:
Sorry for my poor english... :(

lets go!
my audio card is Nvidia HDMI. 

I setup the file .config/retroarch/retroarch.cfg (ssh), and ok! SOUND!
BUT, to restart or shut down the PC, NO SOUND!
There is nothing to do, then modify the audio output rate to 48Khz or 44.1kHz, and works!

Every restart or shutdown, have to modify the audio output rate, and when it works is muted randomly for a few seconds while running some ROM.

im using lakka-Generic.x86_64-devel-20160128074055-r21301-g6429dc7 (online update)

thank you for your attention!

Kivutar, 2016-01-30:
Looks like you forget to stop the retroarch service before editing the retroarch.cfg. Have you read the documentation about that?

albanibr, 2016-01-30:
Follow the instructions in the forum.
And repeat, to change the frequency of audio.

Load putty
root 
root
systemctl stop retroarch
nano .config/retroarch/retroarch.cfg
ctrl w > audio_device
then enter your card inside the quotes
ctrl x > Y > press enter
systemctl start retroarch

I tested other systems on the PC and audio HDMI is working properly, problems with the sound in the lakka!

sonikokaruto, 2016-02-09:
I am having the same problem, i have a foxconn nt-a3500, it has the amd e350, and audioworks with the system speaker (full blast volume), headphones and digital, but it doesn't work with hdmi. I've tried both the 64 and 32 bit versions, both with and without the radeon.audio=1 setting on the extlinux.conf

I don;t know what else to do. 

The thing that kinda stings, is that, openelec with kodi, actually works flawless, but this is not working at all.

I need help :(

Kivutar, 2016-02-10:
Could help: http://unix.stackexchange.com/questions/62683/sound-not-working-when-both-the-sound-card-and-hdmi-use-the-intel-hda-driver-ho

You can also try to use some command line utilities to play audio from the CLI. Like aplay or speaker-test.

sonikokaruto, 2016-02-10:
Oooh, that seems like a good option, i'm gonna try it as soon as i get home
today :)

fohoizey, 2016-02-21:
Hi

Same issue here.
AMD Radeon HD7970
No sound even with radeon.audio=1 and even after having tryed all audio_device settings.

Any idea ?

Thanks

sonikokaruto, 2016-02-21:
Well, i haven't found the time to try the solution about it detecting them in different order, but you could try that. 

Or maybe the team could contact the persons on charge of Kodi-OpenElec to see the way they implemented it.

It would make this a lot more compatible for everyone.

Kivutar, 2016-02-22:
OpenELEC is using X while we are using DRM/KMS.

sonikokaruto, 2016-02-22:
Is thjere a way to change lakka to use X?

fohoizey, 2016-02-22:
If not, is there any solution to get Sound working ? 
There must be a huge amount of people concerned by this bug as it seems to affect all ATI powered PCs
Is there anything we can do to help you debug?

sonikokaruto, 2016-02-22:
Yeah, i'm all for helping, this issue is really bothering me, because these
inexpensive pc's have the muscle and speed to run a lot of stuff smoothly
and with a lot more options than something on, let's say, a raspberry pi

fohoizey, 2016-02-22:
More informations here http://libretro.com/forums/showthread.php?t=5135

Kivutar, 2016-02-23:
Switching to X is easy, but it have to be done at build time. When I provide X images, users who prefer DRM/KMS will complain. When I provide DRM/KMS images, users who prefer X will complain. And I can't (and don't want to) provide both.

I'm in favor to DRM/KMS because it is the future of display on linux, and it has less overhead. Also, it allows us to avoid providing an extra image for nvidia-legacy.

I still want to support PCs, because we're the only distro to support it so far, vs recalbox and retropie. But our PC builds require a lot (too much) of maintainance. I mean compared to an Odroid-XU4, which cost only $70 and will perform as well as a NUC.

fohoizey, 2016-02-23:
I perfectly understand this but there are some people trying to use lakka on much more powerful machines. I'm using a quad core i5 with a powerful gpu, a lot of ram and an ssd. It's a perfect hardware for modern gaming machines' emulation. Dreamcast, PSP, etc. 

Can we (I mean, users like me) build lakka using x easily? Of course I would much prefer finding a solution for the KMS build... 

Anyway, except this point, lakka is perfect on pc

sonikokaruto, 2016-02-23:
Well, thanks for your help.

Sent using CloudMagic Email
[https://cloudmagic.com/k/d/mailapp?ct=pa&cv=8.1.25&pv=5.0.2&source=email_footer_2]

[https://github.com/libretro/Lakka/issues/292#issuecomment-187564995] .[https://github.com/notifications/beacon/AGj4xksjTIAzWTseFI6eRrvJ5rlHHJhQks5pm_KigaJpZM4HMGFj.gif]

Kivutar, 2016-02-23:
Yes, reverting that commit https://github.com/libretro/Lakka/commit/48d49cda7fe4db43f9a95e1458ec8f40ed8f3aee and recompiling should do the trick.

fohoizey, 2016-02-23:
I'll try ! Merci
I've just have to download the sources and to compile them ?
I'll let you know if it works. That would be great.

Kivutar, 2016-02-23:
It works, I did it many times ;)
Compilation instructions are on the wiki.

fohoizey, 2016-02-23:
Thanks. I'm using a Mac and it shows error when trying to compile.
Do I have to compile on Linux (which I don't have) ?
Or on the target machine ?

Thanks

fohoizey$ DISTRO=Lakka PROJECT=Generic ARCH=x86_64 make image
./scripts/image mkimage
cat: /proc/cpuinfo: No such file or directory
uname: illegal option -- i
usage: uname [-amnprsv]

 =================================================================================
 Configuration for Lakka (unofficial)
 =================================================================================

 Buildoptions:
 ======================================================
 - CPU (ARCH):				 x86-64 (x86_64)
 - FLOAT:				 
 - FPU:					 
 - SIMD support:			 yes
 - Optimizations:			 size
 - LTO (Link Time Optimization) support: yes
 - GOLD (Google Linker) Support:	 yes
 - LLVM support:			 yes

 Graphic configuration:
 ======================================================
 - Xorg Graphic Drivers:		 r300 r600 radeonsi i915 i965 nvidia
 - XORG Composite support:		 yes
 - WindowManager:			 none
 - SDL support:				 yes
 - OpenGL (GLX) support (provider):	 yes (mesa)
 - OpenGLES support (provider):		 no (no)
 - uvesafb support:			 yes

 Hardware decoder configuration:
 ======================================================
 - Kodi Player driver:			 default
 - VAAPI Support:			 yes
 - VDPAU Support:			 yes

 Input device configuration:
 ======================================================
 - Remote support:			 no
 - ATV Remote support:			 no
 - CEC Adapter support:			 yes
 - IRTrans support:			 no
 - Kodi Joystick support:		 yes

 Misc. hardware configuration:
 ======================================================
 - ALSA support:			 yes
 - Bluetooth support:			 yes
 - LCD drivers:				 irtrans,imon,imonlcd,mdm166a,MtxOrb,lis,dm140,hd44780,CFontz,SureElec,vlsys_m428,serialVFD,shuttleVFD
 - Include driver:			 RTL8192CU
 - Include driver:			 RTL8192DU
 - Include driver:			 RTL8188EU
 - Include driver:			 RTL8812AU
 - Include driver:			 dvbhdhomerun
 - Include firmware:			 misc-firmware
 - Include firmware:			 wlan-firmware
 - Include firmware:			 dvb-firmware
 - Include firmware:			 iwlwifi-firmware

 Network service configuration:
 ======================================================
 - Avahi (Zeroconf) support:		 yes
 - NFS mounting support:		 yes
 - SAMBA mounting support:		 yes
 - SAMBA server support:		 yes
 - SFTP server support:			 yes

 OS configuration:
 ======================================================
 - OEM Support:				 yes
 - Default ROOT Password:		 root
 - Bootloader:				 syslinux
 - UDevil support:			 no
 - Installer support:			 yes

 Misc. Filesystems:
 ======================================================
 - Swap Support:			 no
 - exFAT Support (via Fuse):		 yes
 - NTFS Support (via Fuse):		 yes
 - Install HFS Tools:			 yes

 Kodi configuration:
 ======================================================
 - Kodi version:			 no
 - Kodi nonfree support:		 yes
 - Kodi Blu-Ray support:		 yes
   - Bluray BD+ support:		 yes
   - Bluray AACS support:		 yes
 - Kodi DVDCSS support:			 yes
 - Kodi Airplay support:		 yes
 - Kodi Airtunes support:		 yes
 - Kodi NFS support:			 yes
 - Kodi MySQL support:			 yes
 - Kodi Optical Drive support:		 yes
 - Kodi SAMBA client support:		 yes
 - Kodi SSH client support:		 yes
 - Kodi UPNP support:			 yes
 - Kodi Webserver support:		 yes
 - Include Skin:			 Confluence
 - Default Skin:			 Confluence
 - Include extra fonts:			 yes
 - Include RSXS Screensaver:		 yes
 - Include ProjectM Visualization:	 yes
 - Include Goom Visualization:		 yes

 =================================================================================
 End Configuration for Lakka
 =================================================================================

cat: /proc/cpuinfo: No such file or directory
uname: illegal option -- i
usage: uname [-amnprsv]
**** Your system lacks the following tools needed to build Lakka ****
gawk cvs mkfontscale mkfontdir bdftopcf
********
make: *** [image] Error 1

sonikokaruto, 2016-02-23:
haven't used osx in a while but it seems that you may need to do something like  (and i'm not even sure something like apt-get exists on osx)

sudo apt-get update
sudo apt-get install gawk cvs mkfontscale mkfontdir bdftopcf

Kivutar, 2016-02-24:
I've never tried to compile on OSX. It may work. You need some dependancies like **gawk cvs mkfontscale mkfontdir bdftopcf**. Try to install them using a package mananger like **brew**

fohoizey, 2016-02-24:
Thanks, I've tried to install the dependencies but they nedd others and others and so on.
I'll try on ubuntu at work today.

fohoizey, 2016-02-24:
Argg, unable to compile Lakka because https://www.freedesktop.org (which seems to be called during the compiling process) is down. Any mirror for this domain or any way to fix this ?
Thanks

fohoizey, 2016-02-24:
Damn, it fails somewhere else :

--2016-02-24 16:31:39--  http://sources.openelec.tv/mirror/pcre/pcre-8.36.tar.bz2
Resolving sources.openelec.tv... 82.220.2.33
Connecting to sources.openelec.tv|82.220.2.33|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-02-24 16:31:39 ERROR 404: Not Found.

Cant't get pcre sources : ftp://ftp.csx.cam.ac.uk/pub/software/programming/pcre/pcre-8.36.tar.bz2
 Try later !!
make: *** [image] Erreur 1

thiscodedbox, 2016-02-29:
I'm also having the ATI HDMI Audio Problem. Here is my aplay -l and -L

```
card 0: HDMI [HDA ATI HDMI], device 3: HDMI 0 [HDMI 0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

hdmi:CARD=HDMI,DEV=0
    HDA ATI HDMI, HDMI 0
    HDMI Audio Output
```

- I disabled the onboard intel audio and video, so the only other driver is null.
- I've added radeon.audio=1 to /flash/extlinux.conf
- I stopped retroarch and added "hdmi:CARD=HDMI,DEV=0" to the audio_device key in .config/retroarch/retroarch.cfg

Oddly enough, via SSH with emulation not running doing a

`aplay -D plughw:0,3 sound.wav`

Seems to play a sound - gives a message it's playing and pauses for the duration of the file - unless I'm running say Super Mario World, then it says the resource is busy. This makes me think it's playing the sound but someone it's not going out the HDMI.

Kivutar, 2016-02-29:
So just set

    audio_device="plughw:0,3"

In your retroarch.cfg. (Without forgetting to stop the retroarch service.)

sonikokaruto, 2016-02-29:
Excuse me, (and ignore my ignorance) but what's the difference between putting 

audio_device="plughw:0,3"

vs

audio_device="hdmi:CARD=HDMI,DEV=0"

Does it handle it differently? (and yeah, i'm gonna try this too)

Kivutar, 2016-02-29:
To be honest I'm not exactly sure. I'd go for "they are synonyms" but only an alsa specialist could reply to that.

thiscodedbox, 2016-03-01:
Thanks Kivutar. I'm sorry - what I typed didn't quite make sense so let me clarify. Below is my output - it's "playing" the sound as far as ALSA is concerned, but there is no sound coming out of my TV which is hooked up to the HDMI:
```
Lakka:~ # aplay -D plughw:0,3 sound.wav
Playing WAVE 'sound.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Lakka:~ # speaker-test -c 2 -r 48000 -D hw:0,3

speaker-test 1.0.29

Playback device is hw:0,3
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 64 to 16384
Period size range from 32 to 8192
Using max buffer size 16384
Periods = 4
was set period_size = 4096
was set buffer_size = 16384
 0 - Front Left
 1 - Front Right
Time per period = 5.636446
 0 - Front Left
 1 - Front Right
```
`hdmi:CARD=HDMI,DEV=0` or `hw:0,3` for `AUDIO_DEVICE` makes no difference. I feel that retroarch and ALSA is "playing" the audio, but it's just not coming out of the speakers. I'm experimenting with blacklisting drivers via the two confs in /storage/.config/modprobe.d but I'm stabbing in the dark really. An ALSA specialist would be quite handy right about now.

sonikokaruto, 2016-03-01:
I am actually (as of right now) trying this too, and i can tell it's
playing something, just nothing is coming out of my hdmi speakers.

The integrated speaker does work, full volume, but does work.

fohoizey, 2016-03-01:
Hi,

I've finally never been able to compile, it always fail. I face the exact same problems as the others.
Audio seems to be played but ... muted : no sound at all goes to HDMI.
Perhaps it is directed to the internal sound card.

Has anyone been able to compile the "old" X based release that would solve the problem ?

sonikokaruto, 2016-03-01:
I know you have to make a git pull and then compile it, but i haven't been
able to, i know i'm not doing something right, but setting up everything
just to compile lakka, it's getting tiresome and time-wastey, i'm just
thinking of sticking to emulation station on windows and be done with it.

Kivutar, 2016-03-01:
Then why not installing retroarch on windows? It contains the graphical interface of Lakka anyway. So you can set it up exactly the same way.

(An alternative is to invest $40 on an ARM board, and you're done)

sonikokaruto, 2016-03-01:
I was rooting on lakka because it's the only alternative for pc's (just not my spare htpc).

I already have a raspberry pi-2 but wanted to try something like this on pc (with the advantage of having all that extra power so it could run stuff up to dreamcast).

I might as well just go for the odroid-xu4, it seems to be the most capable one.

fohoizey, 2016-03-01:
Well, buying an arm card and being unable to play dreamcast and psp games just because I can't understand how to have sound on lakka don't sound like a good idea to me. 
Lakka is the only retrogaming distribution made for pc and that's why I wanted to use it

Kivutar, 2016-03-01:
Ok, I will build an X11 image when I have some free time during the week.

fohoizey, 2016-03-01:
Merci Jean-André c'est cool :)
Impatient to test it.

sonikokaruto, 2016-03-01:
Thanks, shall try it

fohoizey, 2016-03-02:
Here is where compilation fails :
```
 --2016-03-01 18:58:49--  http://sources.openelec.tv/mirror/pcre/pcre-8.36.tar.bz2
Resolving sources.openelec.tv... 82.220.2.33
Connecting to sources.openelec.tv|82.220.2.33|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-03-01 18:58:49 ERROR 404: Not Found.

Cant't get pcre sources : ftp://ftp.csx.cam.ac.uk/pub/software/programming/pcre/pcre-8.36.tar.bz2
 Try later !!
make: *** [image] Erreur 1

```

Kivutar, 2016-03-02:
Replace the version by 3.38 in packages/devel/pcre/package.mk

fohoizey, 2016-03-03:
THanks. Stuck here now :
`checking whether GCC is new enough... no
configure: error:
The selected GCC C++ compiler is not new enough to build LLVM. Please upgrade
to GCC 4.7. You may pass --disable-compiler-version-checks to configure to
bypass these sanity checks.
make: *** [image] Erreur 1
`

fohoizey, 2016-03-08:
I've never been able to compile :( As anyone succeeded ?
I've tried much older releases and still have no -real- sound. I've some strange crackling sound but nothing acceptable.

Should i give up ?

Kivutar, 2016-03-08:
I'm uploading a new version with updated kernel+mesa+libdrm. It **may** fix some of these bugs.

thiscodedbox, 2016-03-08:
Thanks Kivutar - I'll have a look tomorrow. I did A LOT of testing and tweaking with Ubuntu and Debian trying to find a workable situation of running Retroarch on Linux to take advantage of a lean OS and to take advantage of DRM/KMS.

I know running it in X is bad for performance, is it possible that the xorg drivers need to be installed for HDMI audio to work for the ATI cards that have problems? Even though it's running in DRM/KMS, not X, could the drivers still be needed for audio?

I say this because when I was testing with Debian and Ubuntu I had HDMI sound working on Retroarch on Debian and Ubuntu, I found that removing all the xorg stuff and removing the window manager broke HDMI sound - but video was still working - that is DRM/KMS was sort of working - but I ended up changing tack anyway and going with Windows 7 for the time being and for my sanity. I ultimately want the low input lag/latency of Linux though so i'm still in the fight. I'm happy to test anything you send my way and give you feedback. I've found that with the ATI video card the Atom n2800 has been performing even for N64 with a bit of tweaking, and it's dead silent and low power usage.

Intel DN2800MT
2.8ghz Atom n2800
Intel Graphics 3600 (disabled, this is the weird PowerVR chipset that is not really supported)
4GB RAM
250GB Samsung 850 mSATA Boot
ATI HD5430 1GB Graphics PCI-E 1x (full-size PCI-E hooked up with an adapter to the PCI-E slot like this http://ecx.images-amazon.com/images/I/41BVCkOZ4hL._SY355_.jpg)

fohoizey, 2016-03-08:
Many thanks Kivutar ! That sound great ! Where can we download this new version ? I'll give you a feedback immediately.

thiscodedbox, 2016-03-09:
Still having the same ATI problems with the new build. I've gone further in depth, seems to be related to it not getting a valid ELD from the monitor (Samsung TV in my case but happens on other TVs). Seems to be a common problem in Arch and a few other distros running a 4+ Kernel.

```
Lakka:/proc/asound # dmesg|grep HDMI
[    1.219903] [drm]   HDMI-A-1
[   16.252643] snd_hda_codec_hdmi hdaudioC1D0: HDMI ATI/AMD: no speaker allocation for ELD
[   16.260101] input: HDA ATI HDMI HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1c.1/0000:02:00.1/sound/card1/input7
[   16.359448] snd_hda_codec_hdmi hdaudioC1D0: HDMI ATI/AMD: no speaker allocation for ELD
[   16.659783] snd_hda_codec_hdmi hdaudioC1D0: HDMI ATI/AMD: no speaker allocation for ELD
[   16.959301] snd_hda_codec_hdmi hdaudioC1D0: HDMI ATI/AMD: no speaker allocation for ELD
[   17.259257] snd_hda_codec_hdmi hdaudioC1D0: HDMI ATI/AMD: no speaker allocation for ELD

```

fohoizey, 2016-03-09:
No success with the new build :(
The "no speaker allocation for ELD" seems to be a good clue of what's happening.
I've read many topics talking about this and the solution is to manually set allocation using :
xrandr --output HDMI-0 --set audio auto

but xrander is not installed on Lakka.

I'm so sad not being able to use Lakka on my powerful PC :(

julianxhokaxhiu, 2016-03-09:
Sorry to say that but if you really want RetroArch, just install Windows and Retroarch over it, it works fine :) When this will be fixed on Linux kernel and/or driver side, you'll be more than happy to shift to Lakka.

sonikokaruto, 2016-03-09:
Like i said on a previous post, This isn't an issue of the kernel, because other distro that uses openElec as it's base (the media center openElec with kodi) actually works correctly thru hdmi.

I think this is more like kivutar has said: it's an issue between retroarch and the audio driver.

Kivutar, 2016-03-09:
The other OpenELEC derived distros are using X11. We are using DRM/KMS. That makes a big difference. And this difference is in Kernel+Mesa+libdrm+drivers. Not in RetroArch.

thiscodedbox, 2016-03-09:
>  just install Windows and Retroarch over it, it works fine :) When this will be fixed on Linux kernel and/or driver side, you'll be more than happy to shift to Lakka.

Thanks for trying to offer an alternative julianxhokaxhiu but it's already been established that Windows is the easiest route and works fine - it's what I'm doing in the meantime. The better potential for low input lag/latency and for ultimately fewer variables to go wrong (once it's all working) is what draws me to Lakka. I think we should stay on target and troubleshoot this if Kivutar is willing (thanks for all your time)

[39 more replies]

closed 2016-06-02 by Kivutar