libretro/Lakka #293: SEGA Genesis on RPi2 not booting any ROM

julianxhokaxhiu, 2016-01-27:
Hi,

yesterday I was going to play Sonic ( the first one ) on my RPI2, the dump of the ROM is fine ( CRC matches the DAT-o-MATIC database of No-Intro ) and is even recognized by the autoscan inside Lakka. Although once I try to play it, it just returns to Lakka like it was just restarted ( same behavior when you press A on "Quit Retroarch" ).

I didn't check the logs honestly but any other emulator is working, including PPSSPP for example.

Thanks in advance,
Julian

Kivutar, 2016-01-27:
Ok, you're the third one to report this bug. I ingored the two previous reports because i've always been able to launch my genesis roms. I will try again in 30min.

julianxhokaxhiu, 2016-01-27:
Sure! Thanks, FYI I've got the latest nightly ( I updated yesterday using the `lakka-update` command through SSH, just before playing ).

Kivutar, 2016-01-27:
Ok, I just tried again, and all my genesis roms are launching.

If you go to Settings->Playlists->Sega - Mega Drive - Genesis, what core is assignated?

julianxhokaxhiu, 2016-01-27:
I'm going to look at it this evening, I'm in the office now :) I'll report you back, but anyway It's the default one, I never changed it.

Kivutar, 2016-01-27:
Ok, default here too. And I've never head of a bios for genesis, so I don't see what is the difference between our two setups :o

julianxhokaxhiu, 2016-01-27:
Me neither :\ I'm gonna take a look at the name of it and logs too :) I hope it will help you, thanks!

Are you using a RPI2 too?

Kivutar, 2016-01-27:
Yes, same hardware, and almost same version of the OS, since you're using the latest nightly build.

julianxhokaxhiu, 2016-01-27:
Ok so, I've checked which emulator I've in the playlist config menu, and it's `Genesis Plus GX` ( I guess it's the default one ). I've tryed to open it twice and it still crashes.

I'm trying to pull out some logs, once I'll find them I'll post them here.

Kivutar, 2016-01-28:
You have to stop retroarch service, and launch retroarch manually (see our FAQ). Can you send me one of your roms by mail?

julianxhokaxhiu, 2016-01-28:
Sure, I'll do it this evening!

julianxhokaxhiu, 2016-01-28:
Ok so I tried your method and I finally got some logs out. It seems that the emulator is not able to read the content inside the ZIP files:
```
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Threaded rarch_task started
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29975.00 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/5xbr-retro.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bead.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/blinky.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/pixellate.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/quilez.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/retro-v2.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/scanline.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/sharp-bilinear.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ fullscreen
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x1.
RetroArch [INFO] :: Found GL context: videocore
RetroArch [INFO] :: Detecting screen resolution 1920x1080.
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [EGL]: eglSwapInterval(1)
RetroArch [INFO] :: [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
RetroArch [INFO] :: [GL]: Version: OpenGL ES 2.0.
RetroArch [INFO] :: Querying GL extension: ARB_texture_non_power_of_two => doesn't exist
RetroArch [INFO] :: Querying GL extension: OES_texture_npot => exists
RetroArch [INFO] :: Querying GL extension: BGRA8888 => exists
RetroArch [WARN] :: [GL]: GLES implementation does not have BGRA8888 extension.
32-bit path will require conversion.
RetroArch [INFO] :: Querying GL extension: GL_EXT_unpack_subimage => doesn't exist
RetroArch [INFO] :: Querying GL extension: EXT_sRGB => doesn't exist
RetroArch [INFO] :: GL: Using resolution 1920x1080
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [GL]: Using GLSL shader backend.
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Shader log: Compiled
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 4 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: Querying EGL extension: KHR_image => exists
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
RetroArch [WARN] :: [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event*?
RetroArch [INFO] :: [udev]: Plugged pad: Microsoft X-Box 360 pad (045e:028e) on port #0.
RetroArch [INFO] :: Autodetect: 100 profiles found
RetroArch [INFO] :: Autodetect: selected configuration: /tmp/joypads/udev/Microsoft_X-Box_360_pad.cfg
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event0) supports force feedback.
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event0) supports 16 force feedback effects.
RetroArch [INFO] :: [udev]: Plugged pad: Microsoft X-Box 360 pad (045e:028e) on port #1.
RetroArch [INFO] :: Autodetect: 100 profiles found
RetroArch [INFO] :: Autodetect: selected configuration: /tmp/joypads/udev/Microsoft_X-Box_360_pad.cfg
RetroArch [INFO] :: [udev]: Pad #1 (/dev/input/event1) supports force feedback.
RetroArch [INFO] :: [udev]: Pad #1 (/dev/input/event1) supports 16 force feedback effects.
RetroArch [INFO] :: Found joypad driver: "udev".
[... 206 lines cut ...]
RetroArch [INFO] :: Overrides: no core-specific overrides found at /storage/.config/retroarch/gpSP/gpSP.cfg
RetroArch [INFO] :: Overrides: no game-specific overrides found at /storage/.config/retroarch/gpSP/Pokemon Rosso Fuoco.cfg
RetroArch [INFO] :: Environ GET_LOG_INTERFACE.
RetroArch [INFO] :: Environ GET_PERF_INTERFACE.
RetroArch [INFO] :: Environ SET_VARIABLES.
RetroArch [INFO] :: Remaps: core name: gpSP
RetroArch [INFO] :: Remaps: game name: Pokemon Rosso Fuoco
RetroArch [INFO] :: Remaps: remap directory: /storage/remappings
RetroArch [INFO] :: Remaps: no game-specific remap found at /storage/remappings/gpSP/Pokemon Rosso Fuoco.rmp
RetroArch [INFO] :: Remaps: no core-specific remap found at /storage/remappings/gpSP/gpSP.rmp
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Content loading skipped. Implementation will load it on its own.
RetroArch [INFO] :: Compressed file in case of need_fullpath.Now extracting to temporary directory.
[... cut ...]
RetroArch [INFO] :: [PERF]: Performance counters (libretro):
RetroArch [INFO] :: Monitor FPS estimation is disabled for threaded video.
RetroArch [INFO] :: Environ (Private) EXEC.
RetroArch [ERROR] :: Failed to start audio driver. Will continue without audio.
RetroArch [INFO] :: Saving config at path: "/storage/.config/retroarch/retroarch.cfg"
RetroArch [INFO] :: Monitor FPS estimation is disabled for threaded video.
```

I hope that this detailed log will help you on fixing this once and for all! :)

lir448, 2016-01-28:
I had a similar problem when the connection to the Internet has not been set. I set up a connection and updated the core, and then I was able to open those roms that could not open before.

Kivutar, 2016-01-29:
The thing is that my ROMs are zipped too. And I'm not able to reproduce this issue.

Also, this

    RetroArch [ERROR] :: Could not read content file "/storage/roms/Pokemon Rosso Fuoco.zip#Pokemon Rosso Fuoco/Pokemon Rosso Fuoco.gba"

Should be something like this, no?

    RetroArch [ERROR] :: Could not read content file "/storage/roms/Pokemon Rosso Fuoco.zip#Pokemon Rosso Fuoco.gba"

julianxhokaxhiu, 2016-01-29:
There's a folder in the ZIP and inside there's the gba file, so maybe the library is detecting it and opening it is.

Other roms are like this and the core is totally reading it. Maybe there's something wrong with the stock cores?

@lir448 can you please point to me the url where to download them? So I can try this evening, thanks!

Kivutar, 2016-01-29:
Yeah. The no-intro roms i'm using doesn't have this folder in the zip.

julianxhokaxhiu, 2016-01-29:
I'll try to first update the cores and see if it's working, then If not, I'll move the files out of the folders

Kivutar, 2016-01-29:
You got your roms from archive.org?

julianxhokaxhiu, 2016-01-29:
http://www.epforums.org/showthread.php?56293-MULTI-ROM-SET-No-Intro-Collection-(Complete-ROM-Sets)

julianxhokaxhiu, 2016-02-28:
Sorry for the long delay in the reply. I checked some of these ROMs and removed them from the folder inside the ZIP but even with the latest update got using `lakka-update` they do not work but they're totally detected from the scan ( which means to me valid ROM ). Any clue?

Still SEGA and GBA not working, everything else yes :)

julianxhokaxhiu, 2016-02-28:
I think I found the solution: I was mounting the share as Read-Only but Lakka is expecting a storage path than can be RW ( since retroarch is extracting the ROM and than using that file, finally deleting it after the file has been loaded ).

So for everyone in the future, remember to mount any external CIFS share with RW permissions ( and make sure the user that you're using has the rights to write in the folder ).

Sorry for the big trouble! Thanks a lot for this awesome system :D

closed 2016-02-28 by julianxhokaxhiu

Kivutar, 2016-02-28:
Oh, I never thought about read only issues. Glad that you could find the real cause of the bug.

twinaphex, 2016-02-28:
Well, if you set the cache directory to somewhere where you have read/write permissions then you don't have this problem. It will then extract the temporary ROM from inside the ZIP to that cache directory and flush it after it's done using it.

julianxhokaxhiu, 2016-02-28:
Thanks for the protip @twinaphex ! I'll look into this right now :D

julianxhokaxhiu, 2016-02-28:
I confirm that using as a cache directory a custom folder created in `/storage/tmp` and after settings this path in `retroarch.cfg` as
```
cache_directory = "/storage/tmp"
```
works perfectly :D Thanks again @twinaphex !

Kivutar, 2016-02-29:
And what if you use the regular /tmp which is a tmpfs iirc?

julianxhokaxhiu, 2016-02-29:
I didn't check how big it is, but I'm sure that using /storage folder I use the full capacity of my MicroSD card which is a 32GB, while I guess that tmpfs is somehow managed in the RAM right? ( correct me If I'm wrong )

Kivutar, 2016-02-29:
Yes tmpfs is in the RAM.

julianxhokaxhiu, 2016-02-29:
Well then I prefer to use the MicroSD just to make sure I don't collide with special cases which may need the RAM for something else :)

brianblakely, 2016-06-14:
I set the cache_directory to /tmp and get the dreaded "Could not read content file" message.

Actually, it doesn't matter where I put the cache, it seems.  RetroArch always crashes.  It does not crash when loading non-zipped files, however.

Kivutar, 2016-06-14:
@brianblakely do you also have a ROM with the folder in the zip?

brianblakely, 2016-06-14:
Nope, no-intro directly inside zip. 

File perm: -rwxr-xr-x root root

Also, the file can be manually unzipped when I copy to /tmp and run unzip on it. 

This is a Pi 1 running the latest download of Lakka available for it. Will also give it a try on a Pi 3 later today.

Kivutar, 2016-06-14:
I'm totally unable to reproduce this bug. My roms from no-intro are working just fine on the Pi 3

brianblakely, 2016-06-14:
Thank you, I will report back once I have Lakka installed on a Pi 3, which should be within 12hrs.

brianblakely, 2016-06-15:
Yeah, same problem on Pi 3.

I should mention that the content file I'm trying to run is on an SMB shared folder.

Kivutar, 2016-06-15:
Lakka runs RetroArch as root. What when you put the same zipped rom directly on the SD card?

julianxhokaxhiu, 2016-06-15:
@brianblakely is having my same issue. I'm 100% that his "guest" user has no realy write access over the SMB share, that's why it has that error.

Try setting the cache path to something else like `cache_dir="/storage/cache"` and of course, make sure that path really exists in your SD ( SSH to Lakka once it's powered on ).

Finally restart Lakka or RetroArch GUI at least so it reloads the configuration and retry :)

Kivutar, 2016-06-15:
What if I set cache_dir to /tmp ? Would it be a valid fix?

julianxhokaxhiu, 2016-06-15:
I don't think so, on the Raspberry Pi if you have to unpack a big file and let's say RAM is needed by emulator and so on, you practically kill some "RAM" for a temporary file that is "unknown" in size.

But would make sense for PC that usually have 4GB+ of RAM :)

brianblakely, 2016-06-15:
@julianxhokaxhiu I can create and execute files and folders in the SMB content directory via SSH.  It's simply unzip and retroarch which fail to decompress zipped content.

I set the cache_directory to /tmp, as most of my files are tiny and do not use much RAM.  That said, I also tried /storage/tmp to equally poor results.

It's really quite a mysterious problem.

kiwisoup2016, 2016-09-18:
I've had an issue getting some games to run like Zelda and Sonic but figured out that if I unplug ethernet they suddenly work. Not sure what the reason is, but maybe that'll help someone...