libretro/Lakka #294: Timezone

julianxhokaxhiu, 2016-01-28:
I saw that just after the Setup the time on the top-right corner is +1h ( Daylight saving time seems to be on or wrong GMT ).

Is there a possibility to fix it? Maybe by using an external NTP server? if so where can it be configured? 

Thanks

Kivutar, 2016-01-28:
Hello, Lakka is using an NTP server already. But looks like TZ have to be configured separately? Either in autostart.sh or by echoing some value in a file. I can't find the proper way on OpenELEC wiki...

julianxhokaxhiu, 2016-01-29:
Try to take a look at this: in OpenELEC usually can be fixed using Kodi, but there's a CLI method too!

http://www.htpcbeginner.com/fix-openelec-incorrect-time-raspberry-pi/

Kivutar, 2016-01-29:
I saw this yes. Have you tried it?

And what about this? Is is still working? https://github.com/libretro/Lakka/wiki/Timezone-settings

julianxhokaxhiu, 2016-01-29:
I have to try it :) I'll give you a feedback later

ifthenelse, 2016-01-31:
I fixed the Timezone with [this trick](http://libretro.com/forums/showthread.php?t=4543&p=31596&viewfull=1#post31596), hope it helps:
```zsh
$ echo "TIMEZONE=Europe/Rome" > /storage/.cache/timezone
```
To find your valid timezone, [check Wikipedia](https://en.m.wikipedia.org/wiki/List_of_IANA_time_zones)

cudencuden, 2016-01-31:
how do we edit the wiki again...can we add a link to edit page? make it easy...

Kivutar, 2016-01-31:
https://github.com/libretro/Lakka/wiki/Timezone-settings here

You mean adding an edit link on documentation pages of on the website? Yeah it is possible. Since it is generated from the wiki.

Kivutar, 2016-02-02:
I edited the wiki. Thanks for all.

closed 2016-02-02 by Kivutar

julianxhokaxhiu, 2016-02-02:
Tryed the regular method and it works! Thanks guys :D

Nill2016, 2017-06-10:
This ifthenelse tip also works with Orange Pi PC Plus:

$ echo "TIMEZONE=America/Sao_Paulo" > /storage/.cache/timezone
$ reboot