libretro/Lakka #296: HC U3 SD card read errors

riisez, 2016-02-21:
When flashing Lakka to my High Capacity Ultra 3 speed SD card the OS gets stuck on boot and after about 10+ min give me this error:

cp: read error: Input/output error
*** Error in prepare_sysroot: mount_common: Could not mount /dev/SYSTEM ***
### Starting debugging shell... type exit to quit ###
sh: can't access tty; job control turned off
#

Can this be fixed or do I have to buy ANOTHER SD card... I'd like not to have to do that...

Kivutar, 2016-02-21:
How many times did you try to flash? Sometimes, you have to flash 2 or 3 times.

riisez, 2016-02-21:
Oh, really? I'll give it a shot.

riisez, 2016-02-21:
Still a no go. I've flashed my girlfriends SD card she uses for her phone. The terminal feedback was almost immediate, the reboot took about 30 seconds, the home screen displays there on the next boot all under or around 1 min. Mine just hangs there. I would use hers, but it's got a larger capacity and she wants it back... She has an XC U1 with 128GB. I've flashed hers with more than one nightly and the current build. Works flawlessly every time. Mine does not boot on every image I flash. I wouldn't think HC U3 32GB would make much of a difference. I only got it for faster transfer speeds.

Kivutar, 2016-02-22:
Ok, can you flash another distribution, like a debian or anything? I'd like to know if the problem is an incompatibility in Lakka, or just an incompatibility between your hardware and your SD card.

riisez, 2016-02-22:
I have prior to flashing Lakka. Ubuntu, Debian Android all work fine.

Kivutar, 2016-02-23:
Ok, you're not the first one to report a bug like this. Sadly, since I don't own myself an SD card that can reproduce this bug, the chances that I can fix it are very low.

riisez, 2016-02-23:
Do you guys at Lakka have a PO Box? I'll contribute to this project by donating mine or something. Might as well if I can't use it.

Kivutar, 2016-02-23:
Thanks :)
But we don't have that yet.
However, the libretro team have one, and we are working hand in hand.

Kivutar, 2016-12-16:
I don't think I can fix this bug. I close it for now.

closed 2016-12-16 by Kivutar