libretro/Lakka #297: Atari 7800 and Magnavox Odyssey not works

lollo78, 2016-02-22:
Hi Kivutar.
It's my fault (I'm sure of that :D) or the libretro core for Atari 7800 (prosystem_libretro.so) and Magnavox Odyssey 2 (o2em_libretro.so) have some issue? When I try to start a rom I see only a black screen.

According http://www.lakka.tv/doc/ROMs-and-BIOSes/ I renamed the bios (from No-intro collection or the Web) and putted "o2rom.bin" and "7800 BIOS (U).rom" (with "ProSystem.dat") inside the system folder, but nothing. With the other cores that require BIOS I haven't this issue.

I tried with all the lakka/retroarch build (also with the latest nightly builds), but nothing. Every time the same issue (black screen).

Another little question relatated the nighlty (sorry, I don't knon how to send you a PM on github): I follow you with interest on https://github.com/libretro/Lakka/commits/lakka but I don't see a match with your commits and the libretro core inside the latest nightly builds (I download always the build for RPi2):http://sources.lakka.tv/nightly/RPi2.arm/ 

In the commits you talk about cores updated, but the compiled core inside the nightly are in most cases the same of the old builds. I know that the best way is to compile lakka every time on my PC 
:(
But could you try to create a nighlty with all the latest commits so to use the nighlty or the files inside of it to a deep test of new functionality and some experiments?
Thanks a lot for all your precious work :*

Kivutar, 2016-02-23:
Hello @lollo78 ,

Concerning Atari 7800, I had some problems finding some ROMs that could run with it. Note that we don't provide prosystem_libretro.so on ARM images, because it is so unoptimized that it is unplayable anyway. So where does your .so comes from? If it comes from the buildbot, know that there are some incompatibilities.

Concerning o2em, I just tried it now, with the No-Intro roms, and it's working fine for me. The BIOS from No-Intro may be incompatible with o2em. Try another one? I'm using another one.

Concerning the nightlies, when you see a commit like this one https://github.com/libretro/Lakka/commit/91bfb80775ed9d0e00c755b68e56cbf717f4a0a3 it means that now we are using picodrive at this exact version https://github.com/libretro/picodrive/commit/edddf9975128324c8aecf33560a5642f7f7195dc (which is the latest)

Then, when enough cores have been updated, I will upload a development build (1 to 10 times per month), and if you update Lakka, the change will be effective, your picodrive will be at the latest version.

Building an image will all the latest commits is not possible for the moment, because that's not how OpenELEC works, and even if I can change that, it wouldn't be so good. For example, today, I had to revert the mupen64plus update because it is crashing on RPi2. The libretro guys often push commits that breaks on a platform or another, because too much platforms are supported to allow effective manual testing.

lollo78, 2016-02-23:
Mmmm... Now I have some doubt about prosystem_libretro.so provenience :D 
Maybe I compiled it some month ago from the lakka code or maybe I extracted it from retropie, recallbox or other distro... I don't remember. 
But now, while "speaking", I can remember that I played to Atari 7800 games in the past. Same for the Magnavox Odyssey 2.
Maybe the issue, is the No-intro collection. In the past I used different roms.
For o2em could you send me an email with your bios for a quick test (lollo78--at--gmail.com)? I tried also with other Bios downloaded from Web and no one worked for me.

Thanks also for the clarification about the nightly builds.

Kivutar, 2016-02-23:
I'm using the o2em roms from no-intro 2016 fixed.
BIOS sent.

lollo78, 2016-02-23:
About prosystem_libretro.so: 
I just extracted the libretro core inside recalboxOS v3.3.0-beta17 and it works very fine on my Pi2 (also with No-Intro Roms). I sent it via email to give it a try. 
I don't know if they use a personalized source code, but their prosystem_libretro is rock solid (60 fps with USA roms and 50 fps with EU roms, like expected). 
So, give it a try and if it's fine also for you, please consider to include it in Lakka.
In my quick test on prosystem_libretro I noticed that if you wanna play all the Atari 7800 No-Intro roms (Europe and USA) you have to delete the BIOS "7800 BIOS (U).rom" inside System folder, and leave only the "ProSystem.dat" file. 
If you do so, You will lose the Atari Logo on game starts (what a pity: I love it, it's so vintage!), but at least you will be able to play all games. Infact, if u use 7800 BIOS (U).rom the EU games stuck on Atari Logo. There's no way to use both BIOS at the same time: the core search only for 7800 BIOS (U).rom. Maybe you could fix the source code to use also a 7800 BIOS (E).rom. ;)
So, the only mandatory file is "ProSystem.dat". If you delete it some game stuck on "strange ASCII font".  

For o2em I tried with your bios, but in my quickly test only one game starts. With all the other roms I see only a colored writing: "SELECT GAME". I tried to press all the button on the controller, but nothing... Stuck on "SELECTED GAME".
Inside recallbox there isn't o2em... So, I'll give a look inside Retropie (stay tuned :D)

Kivutar, 2016-02-24:
Ok, I'm trying to compile prosystem for RPi2 then, I will check the recalbox source code to see if they do anything special.

lollo78, 2016-02-24:
Glad to help :D 
I don't know if it's simple to implement, but if you could fix the "code call" for the .rom BIOS, to include also the European version and non only the US version, we could use both BIOS inside the system folder and save the ATARI logo on every games startup. It's so cool :D 

>Maybe o2em is keyboard controlled??
I had not thought :D 
I use a RPi2 with the livingroom TV and for this reason I use only the controller to interact with lakka/retroarch. I'll try tonight with a keyboard. But... if o2em require a keyboard to work... it's so frustrating.

lollo78, 2016-02-25:
Hi kivutar,
I tried with a keyboard and... it works :worried:, but in a very strange way:
in "SELECT GAME" screen, I have to press the 1 or 2 key to start the game, but to play it require a second PS3 controller. So, seems that who wants to play at Magnavox Odyssey 2 have to connect 1 keyboard and 2 controllers. A little bit frustrating :smiley: 

Please, let me do a consideration. 
Also if o2em runs at full speed, IMO (at the moment) there is no reason to include this core inside Lakka, considering that who use your retrogame-distro maybe have connected the board (RPi2 in my case) to the TV. So, he don't use (supposedly) a keyboard, but only a controller (one, in the most case). This core needs, IMO, another stage of development to solve this little issue of usability (to make it works only with one controller).
But, I repeat, this is only my humble consideration. 
You are the master :wink: :kissing: 

Instead of o2em, I would like to ask you: why not include (in Rpi2) also imame4all_libretro.so and mame078_libretro.so? I'm testing the Recalbox cores and they are full speed and rock solid.
About mame and fba I had only one hitch: how can I scan the games to add them inside the Lakka GUI? After the scanning procedure nothing appears... I have to launch one rom at a time.

Thank you again, for all!

Kivutar, 2016-02-26:
Hello, could you post an issue in libretro-o2em tracker to ask why we need that keyboard and if it can be improved, please?

I would prefer to keep o2em in Lakka because there are also other cores that require a keyboard anyway, and some users will ask for o2em soon or late.

Packaging mame078 for RPi2 should be possible. The current package may even work.

Arcade game scanning is not yet implemented. Scanning relies on CRC checks, and arcade games are zip containing many small complementary roms, so they are difficult to scan.

lollo78, 2016-02-26:
Request done here: https://github.com/libretro/libretro-o2em/issues/4

mame078 seems to be stable (at least the recalbox compiled version): give it a try :wink: 

Are you working (all the libretro team) to find a momentary workaround to scan arcade games? better than nothing... :)

Kivutar, 2016-02-26:
I think that nobody is working on it at the moment. I thought a lot about how to do it, and found no clean way so far. Except maybe checking multiple roms by their crc in the zip. It is certainly implementable, but I guess it would slow down the scanning...

lollo78, 2016-02-26:
Talking like an user: I think that is better a lot of hours scanning that nothing.
Also because the scanning procedure have to be done only one times.
Mame and Fba are very stable core and lovely platform games.
Maybe, your workaround (waiting a major fix) could be happy a lot of people.
If I can help in any way to solve/testing this issue... I'm here.

Kivutar, 2016-02-27:
Yeah, it's still better to have slow scanning than incomplete scanning.

Kivutar, 2016-02-27:
Anyway, I will close this issue for now since prosystem have been added on all builds, and o2em specific issue have been filled.

Feel free to open new issues to request about mame packaging (in Lakka). I think there is already a request to complete arcade scanning on the retroarch issue tracker.

closed 2016-02-27 by Kivutar

lollo78, 2016-02-27:
Perfect! I really hope to see (one day) a solution for the usability issue inside o2em.
About arcade scanning, maybe I found your post in retroarch tracker.
I'm going to add some pepper :wink: