libretro/Lakka #298: Build process fails repeatedly due to getting files from obsolete locations

balthazar, 2016-02-23:
Hi there

I was trying to build to see if I could enable custom controller support (#275) but the build process kept stopping due to wanting to fetch a source file from sourceforge, which threw a redirect chain, resulting in a broken file where the sources were expected. I managed to work around that by manually downloading the requested sources and placing them where the compiler expected to find them.

The true problem arises when trying to get the source file for mesa (mesa-10.5.1.tar.xz). When this happens, all the directions and mirrors the build process attempts are invalid, due to that particular version of mesa having been moved from /pub/mesa to /pub/mesa/older-versions/ in the freedesktop ftp and all its mirrors (I checked).

Downloading by hand and placing the file in the source folder does not work in this case. While I can think of a workaround (setting up a local webserver and adding a hosts rule to redirect to it) to continue, this should not distract from the fact that the current build process is, at least to me, impossibly tedious and requiring constant monitoring.

Is there any way to update the URLs that the build process looks for so that they find the correct files?

I'm building the Generic target on Ubuntu 64 bits 14.04.

Kivutar, 2016-02-23:
You can try changing this line? https://github.com/libretro/Lakka/blob/lakka/packages/graphics/mesa/package.mk#L25

Thanks for reporting those obsolete urls. I will try to fix them.

balthazar, 2016-02-24:
Editing that file worked, thanks!

Kivutar, 2016-02-27:
fixed by @gouchi

closed 2016-02-27 by Kivutar