libretro/Lakka #301: support for powerblock?

hgw77, 2016-02-27:
it there a way or plan to support the powerblock (https://github.com/petrockblog/PowerBlock)

Kivutar, 2016-02-27:
Hi, I can try supporting it if someone is ok to send me the hardware.

hgw77, 2016-02-28:
I have one installed in my raspi2 (inside a snes case ;-)) and volunteer for testing :-) 

by the way without the powerblock driver and installed powerblock it is not possible to shut off the raspi completely. You have to remove the power cable.

Kivutar, 2016-03-08:
Ok, some questions:

How will this behave on RPi that doesn't have the hardware connected? Will it cause problems?

What RPi versions are compatibles with it?

Is the daemon compatible with systemd?

hgw77, 2016-03-10:
sorry for the delay!

* How will this behave on RPi that doesn't have the hardware connected? Will it cause problems?
_I think there is no problem! but I will test it this evening_

* What RPi versions are compatibles with it?
_It's compatible with RPi 1/2 (3?) I guess it is also compatible with the RPi3 because, as I know, the GPIO was not changed or is backward compatible._

*  Is the daemon compatible with systemd?
_I am not sure, I use that on Debian7 so there is no SystemD. When you look into the the sources you can find the init [scripts](https://github.com/petrockblog/PowerBlock/tree/master/scripts) there._

Kivutar, 2016-03-10:
Ok, I will try to package that. It shouldn't be too difficult to package. But we will have to write our own systemd unit.

hgw77, 2016-03-10:
Thank you very much!

By the way, I tested RetroPi (based on Debian7) with installed driver and disconnected Powerblock and no issues so far.

Kivutar, 2016-03-14:
This software is a nightmare to cross compile...

hgw77, 2016-03-15:
do you think it's possible?

Kivutar, 2016-03-17:
It's still possible, but difficult. I saw that recalbox was planning to package it (or a similar software by the same author). We need to talk with them and check if they did the work, to avoid effort duplication. 

@digitalLumberjack do you support the RPi powerblock yet?

hgw77, 2016-04-09:
any news on that toppic?

Kivutar, 2016-04-09:
Not yet sorry, I've been busy with something else.