libretro/Lakka #302: Shanwan/Gasia controllers patch

regne, 2016-02-27:
Hi. First time poster here.
The attached file is [Szymon Janc patch](http://article.gmane.org/gmane.linux.bluez.kernel/62844/match=sixaxis) for bluez to support Shanwan/Gasia PS3 bluetooth controllers fixed for bluez 2.57 on Lakka.
You'll have to clone Lakka from git, drop the files into the /packages/network/bluez/patches/ directory and [compile it](https://github.com/libretro/Lakka/wiki/Compiling-Lakka). If you get an error about  `host/buildvm.h:9:23: fatal error: sys/types.h: No such file or directory` [check this](http://libretro.com/forums/archive/index.php?t-3103.html) to remove lutro from the core list.
The [pairing](http://www.lakka.tv/doc/Wireless-Dualshock/) is straightforward now.
I've been testing it for a couple of hours and so far so good but I had to add two missing functions (patch 16) and probably they were missing for some good reason.
Regards
[Shanwan controller patch.zip](https://github.com/libretro/Lakka/files/149598/Shanwan.controller.patch.zip)

Kivutar, 2016-03-07:
OK, thanks for those patches.

Are you sure they don't break anything about the official dualshocks?

I'm testing them now.

Kivutar, 2016-03-12:
I included your patches in the nightly builds of today. Are you sure that they are safe for dualshock?

regne, 2016-03-12:
Sorry for the long delay to answer, I've been trying to borrow an official dualshock to test the patch with no luck at all but I'm still trying to get one. 

I wanted to remark that these are no my patches but, as I said in the first post Szymon Jane's (one of the main developers of bluez). I only included one more file with some missing functions. 

Probably you should check recalbox's developers solution to this very same problem as they have already included plug'n'play dualshock support in their distribution. They just have a small annoyance with the leds of the controller but it does work nicely.

Kivutar, 2016-03-13:
Looks like they can't use both at the same time: https://github.com/recalbox/recalbox-os/wiki/PS3-controllers-drivers-%28EN%29

regne, 2016-03-13:
I can confirm you that I didn't modify the recalbox.conf file. Just plugged the controller through the USB port and it worked.
This is the relevant part of the file:
```
## Please enable only one of these
# -------------- D1 - PS3 Controllers ------------ #
##Enable PS3 controllers support
controllers.ps3.enabled=1     
## Choose an driver beetween official (for official and gasia) or shanwan if you
controllers.ps3.driver=official 
```              
I tested recalbox 4.0.0-beta1. In the release notes [they mention a new PS3 driver](https://github.com/recalbox/recalbox-os/wiki/ReleaseNote-4.0.0-%28EN%29).
Maybe next weekend I'll get an official dualshock controller, I'll post about it.
I'm also going to change the driver to "shanwan" just to check if there's any visible difference.
Regards.

Kivutar, 2016-03-14:
Ok, I'm going to force rebuild the bluez package. I noticed that adding the patches didn't trigger the rebuild.

Thanks for your help

Kivutar, 2016-03-14:
Commit 746bdb18569bb6a1bf129469caf5571440041be7 adds the patches and forces the rebuild of bluez. This will land in the next developer build.

It still needs some testing with official DS3 and DS4. Feel free to reopen this issue of something is broken.

closed 2016-03-14 by Kivutar

Kivutar, 2016-04-03:
I need to upgrade bluez to 5.37 to fix bluetooth on RPi3. The patches can't be applied anymore. Could you help me to find a version of those patches that works against 5.37 please?

regne, 2016-04-09:
Well, that turned to be easier than expected.
The developers had changed the definition of "`FALSE`" to "`false`". Also I'm glad to check that the "bluez-16_missing-function.patch" is no longer needed.
I'm having some weird problem with github and I can't attach the zip I created with the new patch (It keeps displaying that they don't support that file type and that I shoud try again with a zip file). Fortunately the changes are easy:
Please, remove `bluez-16_missing-function.patch` and then change these three lines:
```
$ diff patches.old/bluez-11_sixaxis-fix-PID-navigation-controller.patch patches.new/bluez-11_sixaxis-fix-PID-navigation-controller.patch 
51c51
<  	btd_device_set_temporary(device, FALSE);
---
>  	btd_device_set_temporary(device, false);

$ diff patches.new/bluez-13_sixaxis-provide-dualshock-3-SDP-record.patch patches.old/bluez-13_sixaxis-provide-dualshock-3-SDP-record.patch
36c36
<  	return false;
---
>  	return FALSE;
209c209
<  	btd_device_set_temporary(device, false);
---
>  	btd_device_set_temporary(device, FALSE);

```
Regards.

Kivutar, 2016-04-21:
Thanks, I'm testing those now with the help of @substring

Kivutar, 2016-04-21:
Those patches are not (or are no more) working

substring, 2016-04-21:
@Kivutar all my apologies, they just didn't work as expected. I could pair both shanwans, still struggling with the GAsia. Not sure though i can reproduce the pairing automatically ...

@regne have you tried GAsia pads ?

regne, 2016-04-22:
I'm afraid not, sorry.I have a couple of Shanwans and they work OK but I can't be of much help with the Gasias.
@substring why did you say that they didn't work as expected?. My two Shanwans behave different when paired: once paired one keeps the connection leds blinking slowly while the other behaves as expected showing just one led fixed.

substring, 2016-04-23:
@regne i guess i had not read correctly the pairing process. My shanwans
paired fine afterall and show the right player number, both of them
But the Gasia won't pair, sadly.

Kivutar, 2016-04-23:
So these patches are still worth using. I will try to fix the conflict with the DS3 auto pairing, and generate a new image. Would you both help me to test it?

regne, 2016-04-24:
> Would you both help me to test it?

@Kivutar Yes, of course. Drop me a line here when ready, please.

substring, 2016-04-24:
@kivutar Sure ! I'm your man !

Kivutar, 2016-04-25:
@regne @substring the new nightly builds have both patchsets. It works for my official DS3 controller. Can you try with yours please?

regne, 2016-04-25:
Hi @Kivutar .
Everything is working fine on this side. I made some testing after pairing and also checked that pairing survives a reboot.
I installed Lakka-RPi.arm-devel-20160425014852-r21415-ge3e200b-noobs.tar (my RPi1 is my "testing environment").
I hope that bluez developers will add the patch soon to the main tree. That would be of great relief.
If you need any more help, please tell.

Kivutar, 2016-04-25:
Thanks @regne 
Could you give a try to auto connect?
Clean your install to remove any trace of bluetooth cache, and start the bluetooth service then plug the DS3, then unplug it, then hit the central button, and it should work.

It works well with my official one. I'd like to know with clones.

regne, 2016-04-26:
You're very close @Kivutar 
I completely reinstalled Lakka and ran:
```
systemctl enable bluetooth
touch /storage/.cache/services/bluez.conf
systemctl start bluetooth
systemctl status bluetooth
```
While plugged the controller could be used wiredly, but not over bluetooth only.
I made some testing and found that I only had to run:
```
Lakka:~ # bluetoothctl
[bluetooth]# power on
```
and then it paired automatically.
The second controller paired then as soon as I released the USB cable. I had no need to press the central button.
The complete sequence was:
```
Lakka:~ # bluetoothctl
[NEW] Controller XX:XX:XX:XX:XX:XX BlueZ 5.37 [default]
[NEW] Device YY:YY:YY:YY:YY:YY PLAYSTATION(R)3 Controller
[bluetooth]# power on
Changing power on succeeded
[CHG] Controller XX:XX:XX:XX:XX:XX Powered: yes
[CHG] Device YY:YY:YY:YY:YY:YY Class: 0x000540
[CHG] Device YY:YY:YY:YY:YY:YY Icon: input-keyboard
[CHG] Device YY:YY:YY:YY:YY:YY Connected: yes
[bluetooth]#
```
Please ask me if you need any log or more testing.

substring, 2016-04-27:
After quite some testing, i reach the same results as @Regne :

- you must `touch /storage/.cache/services/bluez.conf` 
- you must also `power on` in bluetoothctl

Then it works seamlessly ! Pairing and autoconnect are perfect. Just need to "power on" at boot. No need to press the PS button to pair.

Sadly, the GAsia won't pair.

Kivutar, 2016-04-27:
Ok, I just pushed a patch to always power on. I hope it will not cause problems or too much energy consumption.

I also fixed something in RetroArch, so enabling bluetooth from the menu will start the service in addition to touch the file.

substring, 2016-04-30:
just one point : bluez.conf is mandatory when you wanna play around with bluetoothctl. Wihout it, bluetoothctl is just frozen

regne, 2016-04-30:
Excelent @Kivutar.

I've just tested 	Lakka-RPi.arm-devel-20160429010750-r21424-g8849e9e-noobs.tar and bluetooth pairing works nicely.

Just a small note: in order to use the bluetooth service you have to turn it on and then reboot lakka (and, a bit out of topic, the ssh service has the same problem).

Have you consider setting bluetooth on by default?. I guess that a lot of users will buy cheap bluetooth controllers and, on the power comsumption problem, it can be turned off if not needed.

Thanks a lot.

Kivutar, 2016-05-01:
I've considered turning it up by default yes. I don't know yet what is the good way to do.

I get a different result about the bluetooth. Turning it on works without the need to reboot.

Kivutar, 2016-05-01:
OK, you're right, something is broken on bluetooth activation. I found a fix. I delay the release of one day to get that in.