libretro/Lakka #303: Lutro package fails to build on Generic target

balthazar, 2016-02-28:
Hello again

I'd like to report that leaving the "lutro" package selected when building the Generic project causes the build to fail. Removing the lutro dependency from the packages/lakka/Generic/package.mk file allows the build process to complete.

The following is the error log printed by the build process when trying to build lutro:

BUILD    lutro (target)
make[1]: Entering directory '/media/shareddata/Dev/Lakka/build.Lakka-Generic.i386-devel/lutro-9e39edf'
make -C deps/luajit/src HOST_CC="/home/balthazar/dev/Lakka/build.Lakka-Generic.i386-devel/toolchain/bin/host-gcc -m32" CROSS="/home/balthazar/dev/Lakka/build.Lakka-Generic.i386-devel/toolchain/bin/i686-openelec-linux-gnu-" TARGET_CFLAGS="-march=i686 -m32 -fomit-frame-pointer -Wall -pipe -Os -fexcess-precision=fast -flto -ffat-lto-objects -mmmx -msse -msse2 -mfpmath=sse -O3 -DHAVE_INOTIFY -Wall -pedantic -fPIC -I. -I./libretro-common/include -I./deps/zlib -I./deps -DHAVE_JIT -Ideps/luajit/src  -std=gnu99" BUILDMODE=static CFLAGS=" -O3 -fPIC" Q= LDFLAGS="-fPIC"
make[2]: Entering directory '/media/shareddata/Dev/Lakka/build.Lakka-Generic.i386-devel/lutro-9e39edf/deps/luajit/src'
HOSTCC    host/minilua.o
/home/balthazar/dev/Lakka/build.Lakka-Generic.i386-devel/toolchain/bin/host-gcc -m32  -O2 -fomit-frame-pointer -Wall  -O3 -fPIC -I. -DLUAJIT_TARGET=LUAJIT_ARCH_x86 -DLJ_ARCH_HASFPU=1 -DLJ_ABI_SOFTFP=0  -c -o host/minilua.o host/minilua.c
HOSTCC    host/buildvm_asm.o
/home/balthazar/dev/Lakka/build.Lakka-Generic.i386-devel/toolchain/bin/host-gcc -m32  -O2 -fomit-frame-pointer -Wall  -O3 -fPIC -I. -DLUAJIT_TARGET=LUAJIT_ARCH_x86 -DLJ_ARCH_HASFPU=1 -DLJ_ABI_SOFTFP=0  -c -o host/buildvm_asm.o host/buildvm_asm.c
HOSTCC    host/buildvm_peobj.o
/home/balthazar/dev/Lakka/build.Lakka-Generic.i386-devel/toolchain/bin/host-gcc -m32  -O2 -fomit-frame-pointer -Wall  -O3 -fPIC -I. -DLUAJIT_TARGET=LUAJIT_ARCH_x86 -DLJ_ARCH_HASFPU=1 -DLJ_ABI_SOFTFP=0  -c -o host/buildvm_peobj.o host/buildvm_peobj.c
HOSTCC    host/buildvm_lib.o
/home/balthazar/dev/Lakka/build.Lakka-Generic.i386-devel/toolchain/bin/host-gcc -m32  -O2 -fomit-frame-pointer -Wall  -O3 -fPIC -I. -DLUAJIT_TARGET=LUAJIT_ARCH_x86 -DLJ_ARCH_HASFPU=1 -DLJ_ABI_SOFTFP=0  -c -o host/buildvm_lib.o host/buildvm_lib.c
In file included from host/buildvm_peobj.c:9:0:
host/buildvm.h:9:23: fatal error: sys/types.h: No such file or directory
 #include <sys/types.h>
                       ^
compilation terminated.
In file included from host/buildvm_lib.c:6:0:
host/buildvm.h:9:23: fatal error: sys/types.h: No such file or directory
 #include <sys/types.h>
                       ^
compilation terminated.
In file included from /usr/include/limits.h:25:0,
                 from /usr/lib/gcc/x86_64-linux-gnu/4.8/include-fixed/limits.h:168,
                 from /usr/lib/gcc/x86_64-linux-gnu/4.8/include-fixed/syslimits.h:7,
                 from /usr/lib/gcc/x86_64-linux-gnu/4.8/include-fixed/limits.h:34,
                 from host/minilua.c:33:
/usr/include/features.h:374:25: fatal error: sys/cdefs.h: No such file or directory
 #  include <sys/cdefs.h>
                         ^
compilation terminated.
In file included from host/buildvm_asm.c:6:0:
host/buildvm.h:9:23: fatal error: sys/types.h: No such file or directory
 #include <sys/types.h>
                       ^
compilation terminated.
Makefile:661: recipe for target 'host/buildvm_peobj.o' failed
make[2]: *** [host/buildvm_peobj.o] Error 1
make[2]: *** Waiting for unfinished jobs....
Makefile:661: recipe for target 'host/buildvm_lib.o' failed
make[2]: *** [host/buildvm_lib.o] Error 1
Makefile:661: recipe for target 'host/buildvm_asm.o' failed
make[2]: *** [host/buildvm_asm.o] Error 1
Makefile:661: recipe for target 'host/minilua.o' failed
make[2]: *** [host/minilua.o] Error 1
make[2]: Leaving directory '/media/shareddata/Dev/Lakka/build.Lakka-Generic.i386-devel/lutro-9e39edf/deps/luajit/src'
Makefile:244: recipe for target 'deps/luajit/src/libluajit.a' failed
make[1]: *** [deps/luajit/src/libluajit.a] Error 2
make[1]: Leaving directory '/media/shareddata/Dev/Lakka/build.Lakka-Generic.i386-devel/lutro-9e39edf'
make: *** [image] Error 2

I haven't tried other targets or projects, though, so I can't report if it's a more widespread problem.

Thanks

Kivutar, 2016-03-14:
You're right, I also experienced this bug some time in the past. I was compiling Lakka from an i386 host.

Kivutar, 2016-06-05:
I was able to reproduce on ubuntu

Kivutar, 2016-06-06:
`sudo apt-get install libc6-dev-i386` fixes it on Ubuntu.

On Ubuntu x86_64, they use this path `/usr/include/x86_64-linux-gnu/sys/types.h` instead of `/usr/include/sys/types.h`

webmaster-gostec, 2016-08-11:
Same problem compiling on Debian Jessie (x86_64).
Same solution too.

Kivutar, 2016-08-25:
I don't get why it is looking in /usr instead of build.Lakka-Generic.i386-devel/toolchain/i686-openelec-linux-gnu/sysroot/usr/

Rendman, 2016-12-09:
This failed further in the build of the lua library build process. It was unable to find another header file. I installed the gcc-multilib pacakge with this command:
`sudo apt-get install gcc-multilib`

This seemed to allow it to build correctly.

gouchi, 2017-02-07:
I confirm that installing packages gcc-multilib and libc6-dev-i386 fix the issue.

thefoofighter, 2017-05-01:
This is still an issue, It was by pure chance that i came across this issue on Github, can the guys at Lutro or Lakka perhaps mention that for those cross compiling from generic linux distros may need to grab gcc-multilib & libc6-dev-i386?

I can confirm compiling from Linux Mint 18.1 Serena that the above fix works.

Hope a proper fix or a mention of this workaround will find its way into latest so others don't waste time trying to figure out whats going on :)