libretro/Lakka #304: [ Genesis Plus GX / PicoDrive ] Sonic The Hedgehog 1/2/3 Lagging on RPi2 with High Overclock settings

julianxhokaxhiu, 2016-02-28:
Hi,

finally I made to boot the Genesis ROMs on my setup, now I tested some ROMs, most of them are ok, running smoothly and somehow perfect. Some others, like Sonic, any chapter of it, it's lagging from the beginning ( even on the title scren why says SEGA ).

Am I doing something wrong? Didn't customize nothing, so everything is stock as default.

Thanks in advance :)

julianxhokaxhiu, 2016-03-10:
@Kivutar any help here? :(

Kivutar, 2016-03-20:
Well, I'm not able to reproduce. I also have those roms, from the 2016 no-intro set. Using the default Genesis Plus GX core. No lag detected.

(I don't use the overclock, because I never noticed any performance improvement in emulation with it)

julianxhokaxhiu, 2016-03-20:
Ok, I'll try again then I'll keep you updated. Thanks meanwhile!

julianxhokaxhiu, 2016-03-20:
Tested with a clean ROM from No-Intro set of 2016 and still lags. Tested Genesis Plus GX and PicoDrive, same thing.

What am I doing wrong?

Kivutar, 2016-03-21:
I'm going to fix update-lakka. I create an issue to not forget it.

julianxhokaxhiu, 2016-03-21:
Ok then, I'll close this one. Thank you!

---

For everyone having this issue, please consider a clean flash of the latest Lakka available. Meanwhile, if you're using the `update-lakka` script to update, see #330

closed 2016-03-21 by julianxhokaxhiu