libretro/Lakka #305: [Question] Support for the new RPi3 and N64

narko, 2016-02-29:
Hi,
as you may know by now there is a new Raspberry Pi version and according to the reviews it is 2x faster as the previous model [1]. 
Do you think this new model will improve the behaviour of the N64 emulator?

Cheers!
[1] https://www.raspberrypi.org/blog/raspberry-pi-3-on-sale/

Kivutar, 2016-03-01:
Yes, it can improve N64 speed I think.

Here is an RPi3 build, if you want to try it: http://static.kivutar.me/Lakka-RPi2.arm-devel-20160229191702-r21339-g650f694.img.gz

lollo78, 2016-03-03:
Odroid C2 seems better of Raspberry Pi3, but it's always the same problem...
Raspberry community is so huge and have a lot of distros and support that Odroid can't reach.
What a pity!

lollo78, 2016-03-09:
https://www.youtube.com/watch?v=5baYAlWZs24

Kivutar, 2016-03-10:
That's the regular mupen64plus I think, not the libretro one.

twinaphex, 2016-03-10:
I am trying to backport a GLN64 fork that apparently should have better performance.

lollo78, 2016-03-10:
Nice news @twinaphex 
@Kivutar Correct! Regular Mupen64Plus is the default N64 emulator on Retropie (if I remember well :D). But I think that some improvement on the libretro version can be reached with Pi3, maybe more improvements (if needed) with some optimization inside the libretro core... 
And I hope also to use the "Quality Hack" inside PCSX_reloaded at a full speed.
But I could tell you soon: my new Pi is on the way :kissing:

Kivutar, 2016-03-10:
@twinaphex I guess you're aware that GLES support in GLN64 have been broken for one year now? Most of the time, I see people using rice on RPi2. But GLN64 will give better results in term of speed. It could even make some games playables on RPi1. So it would be very good to have it.

twinaphex, 2016-03-10:
@Kivutar I wasn't aware. First time I hear of it.

Regarding PCSX_Reloaded: it's a worthless emulator to be honest to be basing stuff on. It's better to build up Mednafen PSX and/or wait on Rustation. And there's always PCSX ReARMed anyway.

Kivutar, 2016-03-10:
Well, the best I can tell then is good luck with mupen. Looks like the users appreciate using it, and having it on libretro is good.

I'm still waiting for reports about rice performance on RPi3. From what I saw on RPi2, rice is already playable on some games. There are just some slowdowns that gives an impression of unfinished. I have good hopes that on RPi3, with increased CPU power, and maybe a little overclocking, rice would be enough.

twinaphex, 2016-03-10:
Rice is easily the worst graphics plugin out of the three though. GLN64 is also a distant second to Glide64.

I think only Glide64 can run most games without a bunch of graphics errors.

GLN64 might be enough for a few of the more high-profile games though.

Kivutar, 2016-03-10:
The real problem here is the quality of the userland gpu drivers on arm linux. On hardware like the WeTek boxes, glide64 is full speed, because the drivers are surprisingly good. We can't say the same thing about the other boards.

A note for people wanting to build Lakka boxes mainly to play N64, choose either WeTek Core, or Odroid-XU4, or a NUC with Intel. You will be sure to get full speed on glide64.

twinaphex, 2016-03-10:
Could be yeah.

The GLN64 fork I'm backporting is this - 
http://www.emutalk.net/threads/56031-mupen64plus-video-videocore-a-WIP-optimized-mobile-graphics-plugin

The author says to get good performance on Raspberry Pis with this.

lollo78, 2016-03-10:
here a nice test of Rice on pi2 and pi3 (overclocked @1450 Mhz):
https://www.youtube.com/watch?v=dsrxdCtNzLg

lollo78, 2016-03-11:
Pi3 arrived! This weekend I'll do some test.

lollo78, 2016-03-13:
I made some test with Raspberry Pi 3 (without overclock).

- Of course, all the libretro cores that runs at full speed on Pi2, runs at full speed also on Rpi 3, but now we can enabled some complex shader without lag or issues.

- mupen64plus_libretro: some little improvements in fps, but still unplayable.

- ppsspp_libretro: same of mupen64

- 4do_libretro: about 25 fps (unplayable, but better than Pi2)

- yabause_libretro and virtualjaguar_libretro: some improvements but remain unplayable.

@twinaphex: 
- pcsx_rearmed_libretro: here we have a good and a bad news :smile: 
The good: on Pi3, now we can enable the "Neon Enanched Graphic" into the Core options (pcsx_rearmed_neon_enhancement_enable = "enabled" inside retroarch-core-options.cfg) and play at full speed (60 fps), without lag. It's so nice to see this on a little Raspberry. The games looks so cool... WOW!
The bad news: after few minutes (sometimes seconds) from the "graphic hack on" the games freeze!
This appen also if I enable the Speed Hack options (pcsx_rearmed_neon_enhancement_no_main = "enabled"). 
But for all the time before the freeze... 60 fps! :D
I hope that this issue could be fixed working on the source code.

Kivutar, 2016-03-14:
@lollo78 

Did you try mupen64 with rice?

Also, PPSSPP should be playable on games like FF3 with a frameskipping of 1.

Concerning pcsx_rearmed, were you using the core from Lakka of the buildbot? I noticed this freeze problem with the core from the buildbot.

lollo78, 2016-03-14:
I not tried the "standalone" version of mupen64 (only libretro): I'll do this in the next days. 
I hate to say this (sorry @twinaphex), but on Pi2 the standalone was better (playable) than libretro version (so in theory it will be so also on Pi3). For what I have undestand, the standalone one use some external files (.cfg and .ini) to load the best plugin (rice, gles...) depending on the game started. The libretro version do this in automatic, or we have to manually change every time the plugin?
This is my core config of mupen64plus_libretro:
```
mupen64-cpucore = "dynamic_recompiler"
mupen64-audio-buffer-size = "2048"
mupen64-astick-deadzone = "15"
mupen64-pak1 = "rumble"
mupen64-pak2 = "none"
mupen64-pak3 = "none"
mupen64-pak4 = "none"
mupen64-disable_expmem = "enabled"
mupen64-gfxplugin-accuracy = "medium"
mupen64-gfxplugin = "auto"
mupen64-rspplugin = "auto"
mupen64-screensize = "640x480"
mupen64-aspectratiohint = "normal"
mupen64-filtering = "automatic"
mupen64-polyoffset-factor = "-3.0"
mupen64-polyoffset-units = "-3.0"
mupen64-angrylion-vioverlay = "disabled"
mupen64-virefresh = "1500"
mupen64-bufferswap = "on"
mupen64-framerate = "original"
mupen64-vcache-vbo = "off"
mupen64-boot-device = "Default"
mupen64-64dd-hardware = "disabled"
```

About PPSSPP, I don't like frameskip. I prefer a fix of the "non buffered rendering" :D 
But I'll give it a try. This is the core config tested (if you have advice...):
```
ppsspp_cpu_core = "jit"
ppsspp_locked_cpu_speed = "off"
ppsspp_language = "automatic"
ppsspp_rendering_mode = "buffered"
ppsspp_auto_frameskip = "disabled"
ppsspp_frameskip = "0"
ppsspp_framerate_limit = "0"
ppsspp_force_max_fps = "disabled"
ppsspp_audio_latency = "0"
ppsspp_internal_resolution = "480x272"
ppsspp_output_resolution = "480x272"
ppsspp_button_preference = "cross"
ppsspp_fast_memory = "enabled"
ppsspp_set_rounding_mode = "enabled"
ppsspp_block_transfer_gpu = "enabled"
ppsspp_texture_scaling_level = "1"
ppsspp_texture_scaling_type = "xbrz"
ppsspp_texture_anisotropic_filtering = "off"
ppsspp_texture_deposterize = "disabled"
ppsspp_internal_shader = "off"
ppsspp_gpu_hardware_transform = "enabled"
ppsspp_vertex_cache = "enabled"
ppsspp_prescale_uv = "disabled"
ppsspp_separate_io_thread = "disabled"
ppsspp_unsafe_func_replacements = "enabled"
ppsspp_sound_speedhack = "disabled"
ppsspp_threaded_input = "disabled"
```

Concerning pcsx_rearmed: I tested all the latest libretro pcsx_rearmed core that I found: last Lakka nightly, from Buildbot, extracted from Recalbox 4 beta1, grabbed from Retropie 3.6... 
Every time the same issue, but "pcsx_rearmed_neon_enhancement_no_main" seems not working. I have not notice any difference.
This the config tested that freeze game:
```
pcsx_rearmed_frameskip = "0"
pcsx_rearmed_region = "Auto"
pcsx_rearmed_pad1type = "standard"
pcsx_rearmed_pad2type = "standard"
pcsx_rearmed_drc = "enabled"
pcsx_rearmed_neon_interlace_enable = "disabled"
pcsx_rearmed_neon_enhancement_enable = "enabled"
pcsx_rearmed_neon_enhancement_no_main = "enabled"
pcsx_rearmed_duping_enable = "on"
pcsx_rearmed_spu_reverb = "on"
pcsx_rearmed_spu_interpolation = "simple"
pcsx_rearmed_pe2_fix = "disabled"
pcsx_rearmed_inuyasha_fix = "disabled"
```

twinaphex, 2016-03-14:
Your config for mupen64plus shows that it sets graphics plugin to 'auto', this means it just defaults to Glide64. A Raspberry Pi probably cannot handle that. Consider: a Raspberry Pi 2 has slower CPU performance than a Nintendo Wii. It is really one of the most ghetto devices you can find out there on the marketplace.

Regarding overall slower performance of mupen64 libretro vs. Mupen64 AE: pretty sure it is just the changes to new-dynarec which I didn't backport so far because they would break our iOS patches. However, more recent versions of iOS have already broken the dynarec again anyway so maybe it's good policy to just let iOS default to cached interpreter instead since the CPUs in iDevices are getting fast enough for that and just merge those new-dynarec changes.

As for other stuff that could be it - could be the auto frameskipping stuff. We avoid that like the plague, I guess i could throw it in as a concession for people running it on ghetto harware that runs worse than a Wii, but it really isn't ideal and all of these things come back to the same thing: a Pi is severely underpowered for N64 emulation. An Odroid would probably give you much better results without having to cut all these corners.

Kivutar, 2016-03-14:
Recalbox uses rice in most of the case. To switch to rice, launch a game, change the gfx plugin in the core options, close the content, launch the content.

lollo78, 2016-03-14:
Hi @twinaphex, lovely news!
I can't wait to test the new cores :smile:
 
- About mupen64 libretro: if it is possible, please, consider to implement a workarond like the standalone version to autoload the best plugin (for speed, no artifacts and so on) for each games. You know well that each game run better with a specific plugin: this could be very usefull for the "basic users" (and not only). 

- About PCSX rearmed: In this core I really appreciate the possibility to load the .PBP files (that help to save a lot of disk space), but seems that Mednafen PSX doesn't support it. In Mednafen PSX (and Rustation) could you add the support for this kind of file? Other nice feature of PCSX rearmed is the "Resolution Hack" to improve the graphic quality. There are features like this in the new cores?

Kivutar, 2016-03-14:
@lollo78 if you want your feature request to be implemented one day, you should really try to keep one request per thread. Will all the information gathered in a short description.  If you keep your PSX request in a thread  called "RPi3 and N64", we will forget them soon.

lollo78, 2016-03-14:
Right! I'll do it

lollo78, 2016-03-15:
I done some test with Super Mario 64.
With rice there are some improvements:
at 320x240 we have full speed
at 640x480 there are some stuttering, but in game we are around 50 fps.
The strange thing, is that beetween 640x480 and 960x720 there are imperceptible difference in fps.
So, I think that there are some issue inside the "emulator engine"
Also, in rice "widescreen mode" doesn't work (it works in glide64 - but this plugin stuttering a lot), so I had to force 16:9 from RA Video settings.

Kivutar, 2016-03-15:
Ok, then we got our answer: for now use rice, then when gln64 gets an update, we will check again.

closed 2016-03-15 by Kivutar