libretro/Lakka #306: Adding support for Amlogic S802/805 chipset, is OpenELEC already supported!

matcarfer, 2016-03-02:
Hi, I wanted to ask if there is any possibility to add support for Amlogic S802/805 devices, like: S82, M8, MXIII, Tronsmart S89 Android Boxes.
These devices are really fast and really full featured.
I found out that OpenELEC pure images are already available for the devices:

http://freaktab.com/forum/tv-player-support/amlogic-based-tv-players/beelink-aml-devices-s82-m8-round-mxiii-etc/19572-openelec-for-s82-m8-mxiii-tronsmart-s89

http://freaktab.com/forum/tv-player-support/amlogic-based-tv-players/19253-rom-openelec-for-amlogic-based-tv-boxes-m8-mxq-tv110-mx2-g18?18603-ROM-OpenELEC-for-Amlogic-based-TV-boxes-(M8-MXQ-TV110-MX2-G18)=

http://amlinux.ru/image/

Thanks!

Kivutar, 2016-03-14:
I'm going to receive an MXQ soon. Then I will start this port.

matcarfer, 2016-03-14:
Great, thanks!

Kivutar, 2016-03-22:
I received the MXQ. I will start porting in the next few days.

matcarfer, 2016-03-22:
MXQ 4K? Its Rockchip, not AMLogic, but ok.

Kivutar, 2016-03-22:
Something like this http://g04.a.alicdn.com/kf/HTB1ZrYSIXXXXXc7XXXXq6xXFXXXg/MXQ-Mini-PC-Amlogic-Quad-Core-Android-4-4-Smart-TV-BOX-best-APPS-Player-XBMC.jpg

And it's AMLogic

matcarfer, 2016-03-22:
Perfect! Hope we get somewhere!

Kivutar, 2016-05-14:
Ok, I have a working image for the S805 HD18Q here http://static.kivutar.me/Lakka-S805.HD18Q.arm-7.0.0.2.img.gz

But I'm not able to build it yet from the Lakka tree. I built it using a LibreELEC on top of which I packported the lakka packages.

So it's not going to be an official release yet.

matcarfer, 2016-05-14:
Will try it later on my MXIII, thanks!

LocDest, 2016-05-24:
HI Kivutar, 

I have an all black MXQ S805 as pictured in your link. 
[](http://g04.a.alicdn.com/kf/HTB1ZrYSIXXXXXc7XXXXq6xXFXXXg/MXQ-Mini-PC-Amlogic-Quad-Core-Android-4-4-Smart-TV-BOX-best-APPS-Player-XBMC.jpg)
Can't remember the board number at the moment. I'll confirm later.

I am going to attempt to flash your build this evening. 
Would I be right in saying this is achieved by flashing the .img file inside TWRP recovery and is there a way to boot and run your build from  SD card?

Lakka running on these boxes seems like a great idea to me and I wish to help progress this as much as I can.

Thanks,

Loc
HI Kivutar, 

I have an all black MXQ S805 as pictured in your link. 
[](http://g04.a.alicdn.com/kf/HTB1ZrYSIXXXXXc7XXXXq6xXFXXXg/MXQ-Mini-PC-Amlogic-Quad-Core-Android-4-4-Smart-TV-BOX-best-APPS-Player-XBMC.jpg)
Can't remember the board number at the moment. I'll confirm later.

I am going to attempt to flash your build this evening. 
Would I be right in saying this is achieved by flashing the .img file inside TWRP recovery and is there a way to boot and run your build from  SD card?

Lakka running on these boxes seems like a great idea to me and I wish to help progress this as much as I can.

Thanks,

Loc

Kivutar, 2016-05-24:
Hi @LocDest 

You have to flash the build on the SD card, and use the toothpick trick to boot on the SD card.

LocDest, 2016-05-24:
OK so I managed to boot via SD but I was met with this issue...
[image]

Kivutar, 2016-05-25:
Go to android and set a 1080p resolution. This will fix the layout issue.

And reflash your SD card, your second partition is corrupted.

LocDest, 2016-05-25:
No Android on this box my friend. LibreELEC is nand flashed as soon as I remove from packaging. :+1: 

I am unsure why the second partition would be that way. 
Procedure is to write image to SD via Win32 Disk Imager, correct?

I'll try again when I order more MXQ boxes.

Thanks

Kivutar, 2016-05-25:
When you remove android from these boxes, you lose the ability to tweak the bootloader config. I think.

LocDest, 2016-05-26:
I'm unsure why the bootloader governs the video output for Lakka. Some sort of sourcery? 
How can we get Lakka to work direct from nand? Must be a way but my knowledge is limited. Open/LibreELEC installs from TWRP and runs flawlessly *(in most cases) from nand on these units. 

*(I understand theres the issue of board variants and I've had  number of fake fakes where the ram is 512 rather than 1024 so I have to be carefull of who I chose as a supplier.) :-1: 

Would be great to pull our resources and get this going with others.

Kivutar, 2016-05-26:
For now, I just build the image for SD card. I've not tried NAND.

The bootloader governs the resolution on most platforms. It passes some arguments to the kernel about the resolution.

LocDest, 2016-06-08:
Sorry for the delay in response. I have been waiting more MXQ boxes to arrive.

I can confirm that Lakka (via SD) is now working on All black MXQ with AM_MXQ_A 20150825 mainboard. :+1: 

Remote Control is not working but obviously thats in no deal breaker. 

I am in the process of configuring a Speedlink SL-4443-IRD Strike FX Wireless Gamepad and so far menu operation seems fine.

Wired PS2 Pad clones are on order and I will be adding rom sets when they arrive to give this box a more thorough test.

[image]

matcarfer, 2016-07-05:
@Kivutar Could you provide the latest nightly for S802/S805 devices? Also, why isnt them on: http://sources.lakka.tv/nightly/ ?

Kivutar, 2016-07-05:
I will. I was not yet able to include the port in our trunk. I was using a LibreELEC trunk.

cavero561, 2016-11-18:
@Kivutar  the image that u were build is working on ALL version mainboard or only HD18Q version ?
Could u provide another version image ?

Kivutar, 2016-11-30:
Ok which one do you want?

cavero561, 2016-12-05:
I have mxq that work with m201d version. If you can provide for that version that will be great. Thank u very much @Kivutar !!

matcarfer, 2016-12-10:
Where can I get latest S802 images? They are not in the nightlies

Kivutar, 2016-12-10:
All these ports are still unfinished. @Ntemis is working on it a lot recently. You should see some PRs being merged about S805, S802, etc

Bernsies, 2017-01-19:
Very excited about this prospect!

Kivutar, 2017-02-06:
Images available there http://le.builds.lakka.tv/

closed 2017-02-06 by Kivutar

leemclee, 2017-04-04:
I wish I new what u guys was talking about would love to learn all this sort of stuff.  But wounded if you had yet found a working firm wear lollipop for am mxq a 20150825 . That won't disable my controller or Wi-Fi chip would really appreciate some help with this and hopefully it's quiet easy to install I no how to unzip files and put my box in the mode for recovery ect but TBH that's about it lol