libretro/Lakka #307: Lakka has "no signal" on XU4

AndTheCookies, 2016-03-02:
I have [this image](http://www.lakka.tv/get/windows/odroidxu3/) burned to an eMMC card. The card was previously used with an ODROID U3, but I used [the eMMC Repair Tool](http://forum.odroid.com/viewtopic.php?f=53&t=969) to ensure compatibility with the XU4.

The ODROID powers up, but there is no signal on the screen. I have tried a 1080p Panasonic television and a 720p LCD monitor. Every possible _sentenv_ setting in **boot.ini** has been tested. Nothing works.

cudencuden, 2016-03-10:
have u tested with microSD as your boot device? works fine for me this way..never used eMMC card with my xu4 before.

AndTheCookies, 2016-03-10:
Yes. I have tried microSD. No luck. :(

Kivutar, 2016-03-11:
The SSH service is started already. You don't need to start it.

AndTheCookies, 2016-03-11:
I have used Putty in Windows 10 to SSH into other XU4 images, but I do not know how to SSH into Lakka.

Assuming I was able to do it, what would I be looking for?

Kivutar, 2016-03-11:
User: root
Pass: root
Host: lakka

And it should work

AndTheCookies, 2016-03-11:
Thanks, guys. Are there commands I could feed through SSH to try different resolutions / display settings?

cudencuden, 2016-03-11:
in ssh you can systemctl stop retroarch then in Windows browse to \\lakka then find n edit config file. then do systemctl start retroarch

Kivutar, 2016-03-11:
@AndTheCookies not really, the only things managing the real resolution are either in the boot.ini, either backed in the kernel.

Are you sure of what this eMMC repair tool does? There is no need to change anything on the XU3 image to make it compatible with XU4 (at least for lakka).

AndTheCookies, 2016-03-12:
I was instructed to use the eMMC repair tool because the XU4 was not
turning on. I had to do this because the eMMC was previously used with an
ODROID U3.

Also, the switch on the XU4 is set to boot from the eMMC. The XU4 works
with other images just fine. Lakka is the only image that does not want to
output a video signal ...

> @AndTheCookies <https://github.com/AndTheCookies> not really, the only
> things managing the real resolution are either in the boot.ini, either
> backed in the kernel.
>
> Are you sure of what this eMMC [...]

Kivutar, 2016-03-13:
I will upgrade our XU3 kernel and platform init then.

Kivutar, 2016-03-14:
The XU3 kernel is up to date. Can you just try reflashing, without using the repair tool, and try the image on more than one TV please?

AndTheCookies, 2016-03-15:
Kivutar, thank you. I will try downloading the up-to-date XU3 kernel and reflashing the eMMC.

That being said, I have tried multiple televisions to no avail in the past.

AndTheCookies, 2016-03-15:
- Redownloaded the image
- Reflashed the eMMC card
- No signal

That being said, I was able to SSH into Lakka. I am logged in as root. Anything I can do to troubleshoot this from here?

Kivutar, 2016-03-15:
Sorry but I run out of ideas for now. I checked again the XU3 nightly build, it's working fine on the two TVs I have :(

Rodrigomezs, 2016-05-10:
I have the same problem with my XU4, no signal in some TVs, in other TVs works with same HDMI wire. I fixed the problem in Odroid TurboStation imagee using this sentence in boot.ini:
`setenv videoconfig "drm_kms_helper.edid_firmware=edid/1920x1080.bin"`
but when i use it in boot.ini from lakka it dont works.
Anyone can help?

Kivutar, 2016-05-10:
I was able to reproduce this problem on one of my samsung TVs. This TV have a capacitor issue and 80% of the times, I will get a black screen with "No signal" message.

There are instructions on the web on how to fix it (unpluggin the screen during 15min, and more complex techniques). I doesn't always work.

Adding the setenv in the boot.ini fixed it for sometimes, then the problem came back.

Rodrigomezs, 2016-05-10:
Adding the setenv works 100% times in Gamestation Turbo image. In Lakka doesnt works, 0%. it is frustrating, i have changed hdmi wire 100 times. I will use Gamestation for now, if you guys find a fix for this, answer here please.

Kivutar, 2016-05-10:
Ok, I will upgrade the kernel and the blobs. It looks like I was still on memeka's branch.

Rodrigomezs, 2016-05-10:
thanks for your effort.

Kivutar, 2016-05-10:
It fixed the problem for me http://static.kivutar.me/Lakka-OdroidXU3.arm-devel-20160510210313-r21434-gd23434f.img.gz

Rodrigomezs, 2016-05-10:
works!

closed 2016-05-10 by Kivutar

AndTheCookies, 2016-05-31:
I am so glad I kept this conversation bookmarked. @Kivutar the new link you posted works for me too!

Kivutar, 2016-05-31:
Great :D

Rodrigomezs, 2016-09-06:
I have this issue again on XU4 with hdmi and small screen monitors. I have the last build of lakka. in 32" screen with hdmi works fine.

Kivutar, 2016-09-06:
http://odroid.com/dokuwiki/doku.php?id=supported_resolutions

Rodrigomezs, 2016-09-06:
It should works
http://www.lg.com/uk/monitors/lg-22M47VQ
1920×1080

Kivutar, 2016-09-06:
So maybe the screen is sending a wrong EDID? Can you try tweaking the boot.ini? Be carefull, the variable video_output have to be added in bootargs

Rodrigomezs, 2016-09-09:
The problem was the HDMI wire, poor quality, thanks!