libretro/Lakka #308: Ugly menu

Fransebas, 2016-03-05:
hi, I recently install lakka from the nighties, it was the i386 of 2015/08/06 because when I install the one from the download section it never turn up, but this version is working perfectly except for the menu wich isn't the one from the ones I see on the internet and the web page. I try to update from the newest and when I did my computer doesn't even turn on, it only reboot and do strange noises right after it load lakka, but I load the previous version and work fine. Is there a way that I can download only the menu driver?

Kivutar, 2016-03-05:
Hello Fransebas, looks like you were using a very old version. 

I have no idea why the new version doesn't work for you... Especially those strange noise your talking about, it's the first time a user report such a bug.

Can you send me a photo of this menu with two gears please?

Fransebas, 2016-03-06:
[image]
This is the image, and the problem with the other version, I believe, is the hard drive and some noise it made when restart, I'm using a dell pavilion 2006 32 bits

Kivutar, 2016-03-06:
Ok, the version I see here on your photo is kind of old!

However, you should be able to add icons on the right using the **Add Content** entry to scan directories of roms.

Does your dell have an ATI graphic card?

Fransebas, 2016-03-06:
The add content work thanks! but didn't detect the atari and MAME game. The computer is a inspirion 1525, integrated Intel GMA X3100 graphics

Kivutar, 2016-03-06:
Hello, Atari and MAME games are not yet scannables. But there are some scripts on our wiki and forum to workaround that problem, while we implement this fonctionnality.

Kivutar, 2016-03-14:
@Fransebas what is your hard drive?

Kivutar, 2016-04-02:
I close this since I don't get any reply.

closed 2016-04-02 by Kivutar