libretro/Lakka #310: Menu too bright

astukis, 2016-03-06:
Running Lakka on Pc Compaq evo d510.  When I boot to menu it stays normal for a second and then it turns really bright, impossible to use. Same thing happends on live usb too... Am I missing something?

Kivutar, 2016-03-07:
This bug happens on some graphic cards.

The only known workaround for now is to edit the retroarch.cfg and switch to the **glui** menu driver. You will loose the xmb menu :(

None of my graphics card at home suffer from this bug, so it's not easy to test and fix it.

duduke, 2016-03-13:
I had a similar case with an old laptop, read the following article:
https://wiki.archlinux.org/index.php/backlight

Kivutar, 2016-03-20:
@astukis Have you tried the link of duduke?

astukis, 2016-03-21:
Sorry fellas for not replying. I got the OS running on diffrent rig and can't test it anymore, if I run into this problem again I am sure to give it a shot. Thanks.

Kivutar, 2016-03-21:
Ok np, I close this for now. Reopen it if you encounter that problem again.

closed 2016-03-21 by Kivutar