libretro/Lakka #312: Pixel Perfect (Integer Scaling) not working right

matcarfer, 2016-03-07:
Hi, I come from RetroArch (Windows) and using 1920x1080p/60hz over HDMI and Super Mario Bros for NES and every other game I tried using Integer Scaling are Pixel Perfect, no uneven pixels. Its great.

But using Lakka on RPi 2 some pixels are wider than others, even using Integer Scaling On and Bilinear Filtering Off (also at 1920x1080p/60hz over HDMI). It's easy to spot as the first "hill" of Super Mario Bros (NES) has uneven pixels.

Kivutar, 2016-03-07:
What about your overscan settings?

matcarfer, 2016-03-07:
Overscan Disabled of course

Kivutar, 2016-03-07:
I also noticed it. I'm not 100% sure if we're talking about the same bug though.

I've never noticed it with lakka for PC. Is it possible that this bug affects only ARM boards?

hizzlekizzle, 2016-03-07:
There indeed seems to be a rendering difference between x86 and ARM (possibly specific to RPi?). The overlay scanline effects that line up perfectly on RPi are all funky on x86 and vice versa. Are the ARM boards using full GL or GLES?

matcarfer, 2016-03-07:
Will try Lakka for PC and let you know.

Kivutar, 2016-03-07:
@hizzlekizzle GLES

matcarfer, 2016-03-07:
Can confirm this issue happens on both RPi and PC x64 (latest release: Lakka-Generic.x86_64-devel-20160307094225-r21357-gb4e5029) and here is a pic of the real issue:
https://i.imgur.com/XusDhIv.jpg

matcarfer, 2016-03-08:
I also want to add that RetroArch on Android does the exact same thing, uneven pixels using Integer only, here is proof:
https://i.imgur.com/uQlGJfh.png

matcarfer, 2016-03-08:
Finally, I'll leave here a photo taken from my phone to the screen, using RetroArch 1.3.0 x64 under Win10 x64, with Integer Scaling. This should be how it should be seen on every device:

http://i.imgur.com/V20D5JP.jpg

It seems to me that this is a problem with RetroArch itself and not Lakka.
Lakka on RPi, or Generic.x86_64, or RetroArch on Android do the same bad thing.

hizzlekizzle, 2016-03-08:
what aspect ratio setting are you using?

matcarfer, 2016-03-08:
It's all on auto. All tests I made were simple, install, disable bilinear filtering, enable integer scaling, play  rom.

lollo78, 2016-03-09:
Do you see the pixels at a normal playing distance?!?
Of course, if it's a bug must be fixed! 
But...
WOW! I really envy your eyes
:kissing:

matcarfer, 2016-04-11:
Well, last night I tested RetroArch 1.3.3 on my Android box, and it has the same problem! Problem manifested with NESTOPIA and FCEUMM cores. I'll try to test with another console like SNES or GENESIS (I always use NES cause thats what I love). This issue should be promoted or moved or created on RetroArch issues, as its not specific to Lakka.

Kivutar, 2016-04-11:
I remember this guy who declared a bug about NES cores aspect ratios one week ago.

matcarfer, 2016-04-11:
I'd love to see his post

Kivutar, 2016-04-11:
https://github.com/libretro/RetroArch/issues/2837

Kivutar, 2016-06-15:
I think I understand what happens. I noticed this while running RetroArch on a very low resolution display:

Most SNES or Megadrive games had modes like 256x240, which is a 8/7 ratio. But the console was outputing a 4/3 ratio, by stretching the image. So the pixels in those games were never meant to be square with a 1/1 ratio.

Let launch Chrono Trigger for SNES with the PAR 1:1 aspect ratio (not the core provided 4/3). You will see that now the pixels are aligned. However, you will notice that the game is drawing a ellispsis when it wants to draw a circle.

So I guess we can't fix that bug by design. Having a core provided ratio that stretches the image in this case is a feature and not a bug.

Can we close this?

matcarfer, 2016-06-16:
I think it shouldnt be closes as this IS an issue, as its not present on all platforms. I can do more tests on friday with other cores, not only NES cores.

Kivutar, 2016-06-16:
You're right, I took a look at the screenshots again, and the stuff I was talking about has nothing to do with the problem..

matcarfer, 2016-06-16:
Funny enough, in Lakka-Generic.x86_64-devel-20160607150900-r21464-g936bd6d NES core is broken.

Kivutar, 2016-06-17:
I just checked it, and it works for me.

matcarfer, 2016-06-17:
Strange, will test it later.

Kivutar, 2016-08-25:
@matcarfer can you test with a more recent retroarch?

matcarfer, 2016-08-26:
@Kivutar Yes, please wait till monday for results. Thanks!