libretro/Lakka #313: Mayflash N64 to usb Adapter axis values

gyates100895, 2016-03-10:
I have been trying to use the n64 core for a while now and can not get proper analog stick response. (bad turning in mario kart, no run in super mario 64)The best idea i have is that the axes are not configured properly. This seems to be a common cross platform issue with this adapter which requires the controllers to be calibrated to make the axes register properly.

gyates100895, 2016-03-10:
I would be happy to help debug and fix this however.

Kivutar, 2016-03-10:
Looks like RetroArch doesn't have any way to calibrate joysticks?

gyates100895, 2016-03-10:
We couldn't make a udev rule to run jscal? What about autoconfigs, could they be used to make retroarch understand that the analog stick peaks at a value less than 1?

Kivutar, 2016-03-10:
You might want to revive this issue https://github.com/libretro/RetroArch/issues/366

Kivutar, 2016-12-06:
There is now a setting for this in Input Settings. Have you tried it?

shakkalakka, 2017-03-23:
have the same problem. any change to get the mayflash N64 adpater working fine with lakka ?