libretro/Lakka #314: Raspberry Pi 3 onboard Bluetooth does not work

duduke, 2016-03-12:
modinfo shows that btbcm.ko and hci_uart are not loaded. manually loading btbcm via modprobe does not work, and hci_uart is missing (don't know if it is required).
It works in OpenElec 6.0.3.
lsmod from Openelec:
**Module                 Size  Used by
hci_uart               11657  1
btbcm                   3852  1 hci_uart
bluetooth             273858  19 btbcm,hci_uart
brcmfmac              164388  0
brcmutil                4729  1 brcmfmac
cfg80211              226808  1 brcmfmac
lirc_rpi                5962  3
lirc_dev                7606  1 lirc_rpi
rc_core                14564  1 lirc_dev
bcm2835_gpiomem         2588  0
bcm2708_rng              821  0**

lsmod from LAKKA (9-Mar-2016):
**Module                  Size  Used by
brcmfmac              164441  0
brcmutil                4729  1 brcmfmac
cfg80211              226085  1 brcmfmac
bcm2835_gpiomem         2588  0
snd_bcm2835            17780  1
snd_pcm                61527  2 snd_bcm2835
snd_timer              17296  1 snd_pcm
snd                    39984  4 snd_bcm2835,snd_timer,snd_pcm
bcm2708_rng              821  0**

Kivutar, 2016-03-12:
Ok, I enabled this hci_uart module in the kernel, please try this image http://static.kivutar.me/Lakka-RPi2.arm-devel-20160312223402-r21367-ge423427.img.gz

duduke, 2016-03-13:
Thanks, I'll give it a try later today.

duduke, 2016-03-13:
Still doesn't work.
Even manually loading btbcm and hci_uart...

Maybe another difference in the kernel configuration between OpenELEC and LAKKA?

duduke, 2016-03-20:
Any new ideas?
maybe a diff of the kernel config between lakka and OpenELEC?

Kivutar, 2016-03-20:
Can you try with this one please? http://static.kivutar.me/Lakka-RPi2.arm-devel-20160320181214-r21383-gb80ed9f.img.gz It used the kernel and the config of OpenELEC.

duduke, 2016-03-20:
There's some progress.
Modules now load, but the bluetooth device appears as AA:AA:AA:AA:AA:AA and does not actually work.

Running bluetoothctl (under LAKKA):
**[NEW] Controller AA:AA:AA:AA:AA:AA BlueZ 5.27 [default]**

I think this might be related to the bluez version, LAKKA is based off bluez 5.27 and OpenELEC is based off 5.37.

OpenELEC:
xbmc-playroom:~ # bluetoothctl --version
**5.37**

LAKKA:
xbmc-playroom:~ # bluetoothctl --version
**5.27**

Kivutar, 2016-03-21:
Ok, bluez updated here: http://static.kivutar.me/Lakka-RPi2.arm-devel-20160321081109-r21383-gb80ed9f.img.gz

Kivutar, 2016-03-21:
Additionnal information: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=138908

duduke, 2016-03-21:
I'll give it a go later today. 
I've seen that thread, unfortunately - no response yet. 

Thanks.

duduke, 2016-03-23:
Unfortunately, it still reports as AA:AA:AA:AA:AA despite BlueZ at 5.37.

Kivutar, 2016-03-23:
Ok, I'm going to do further research. BTW, what about WiFi?

Kivutar, 2016-03-23:
Updated the bootloader: http://static.kivutar.me/Lakka-RPi2.arm-devel-20160323202018-r21387-g5ea8dd0.img.gz and reverted to bluez 5.27

duduke, 2016-03-23:
Ok, I think I've got some progress into figuring out what's wrong.
It seems that "brcmfmac_sdio-firmware.service" looks for bluetooth firmware in /etc/firmware which in openelec is a symbolic link pointing to /lib/firmware.
In LAKKA, /etc/firmware does not exist, and since it's a read-only FS I don't know how to create the symlink.
If you can tell me how to create the symlink I can test whether i'm correct.

Thanks.

Kivutar, 2016-03-23:
Well found: I'm missing those two lines!

https://github.com/OpenELEC/OpenELEC.tv/blob/master/packages/linux/package.mk#L197-L198

Kivutar, 2016-03-23:
New image with the symbolic link http://static.kivutar.me/Lakka-RPi2.arm-devel-20160324001707-r21387-g5ea8dd0.img.gz

duduke, 2016-03-24:
Getting some more progress:
A fresh boot still leaves off non-working bluetooth.
Manually running: **/usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -** which should be run under **brcmfmac_sdio-firmware.service** and then restarting the bluetooth service works.
Now need to figure out how to get it at boot.

duduke, 2016-03-24:
I believe the reason that **brcmfmac_sdio-firmware.service** does not start correctly on boot, is due to a missing BlueZ patch from OpenELEC:
https://github.com/OpenELEC/OpenELEC.tv/blob/6b9e7aaba7b3f1e7b69c8deb1558ef652dd5b82d/packages/network/bluez/patches/bluez-07-broadcom-dont-set-speed-before-loading.patch

**brcmfmac_sdio-firmware.service** status from LAKKA (after a clean boot):
xbmc-playroom:~ # systemctl status brcmfmac_sdio-firmware.service
● brcmfmac_sdio-firmware.service - Broadcom sdio firmware update for BCM43430A1
   Loaded: loaded (/usr/lib/systemd/system/brcmfmac_sdio-firmware.service; enabled; vendor preset: enabled)
   Active: active (exited) (Result: exit-code) since Thu 1970-01-01 02:00:06 IST; 46 years 2 months ago
  Process: 310 ExecStart=/usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow - (code=exited, status=1/FAILURE)
 Main PID: 310 (code=exited, status=1/FAILURE)

Mar 25 00:29:30 xbmc-playroom hciattach[310]: Initialization timed out.
Mar 25 00:29:30 xbmc-playroom hciattach[310]: bcm43xx_init
Mar 25 00:29:30 xbmc-playroom hciattach[310]: Set Controller UART speed to 921600 bit/s
Mar 25 00:29:30 xbmc-playroom hciattach[310]: Flash firmware /etc/firmware/brcm/BCM43430A1.hcd
Mar 25 00:29:30 xbmc-playroom systemd[1]: brcmfmac_sdio-firmware.service: Main process exited, code=exited, status=1/FAILURE
Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

from OpenELEC:
xbmc-playroom:~ # systemctl status brcmfmac_sdio-firmware.service
● brcmfmac_sdio-firmware.service - Broadcom sdio firmware update for BCM43430A1
   Loaded: loaded (/usr/lib/systemd/system/brcmfmac_sdio-firmware.service; disabled; vendor preset: enabled)
   Active: active (exited) since Thu 1970-01-01 02:00:06 IST; 46 years 2 months ago
  Process: 295 ExecStart=/usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow - (code=exited, status=0/SUCCESS)
 Main PID: 295 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/brcmfmac_sdio-firmware.service
           └─446 /usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -

Mar 25 00:34:00 xbmc-playroom hciattach[295]: bcm43xx_init
Mar 25 00:34:00 xbmc-playroom hciattach[295]: Flash firmware /etc/firmware/brcm/BCM43430A1.hcd
Mar 25 00:34:00 xbmc-playroom hciattach[295]: Set Controller UART speed to 921600 bit/s
Mar 25 00:34:00 xbmc-playroom hciattach[295]: Device setup complete
Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

**Notice **that indeed in LAKKA, the speed is set and then it tries to load the firmware, which is reversed in OpenELEC.

duduke, 2016-03-28:
Also, WIFI works following a successful firmware load.

duduke, 2016-03-28:
I have investigated further, and I believe that BlueZ 5.37 is indeed required.
Although after the latest test the controller does shows up, it will not get into either **Discoverable **or **Discovery** modes.
When setting either "discoverable on" or "scan on" it will spit out:
Failed to set discoverable on: org.freedesktop.DBus.Error.NoReply

Whereas in OpenELEC:
[CHG] Controller B8:27:EB:93:0D:23 Discoverable: yes

So, I believe that both BlueZ 5.37 and this patch: https://github.com/OpenELEC/OpenELEC.tv/blob/6b9e7aaba7b3f1e7b69c8deb1558ef652dd5b82d/packages/network/bluez/patches/bluez-07-broadcom-dont-set-speed-before-loading.patch are required for correctly using the onboard bluetooth on the raspberry pi 3.

Kivutar, 2016-04-02:
Ok thanks a lot for this information. Here is the new image with all your recommandations applied http://static.kivutar.me/Lakka-RPi2.arm-devel-20160402112232-r21391-g74091ed.img.gz

duduke, 2016-04-02:
It works!

closed 2016-04-03 by Kivutar

maverick447, 2017-02-23:
@duduke thanks for your fix. I was struggling with a dysfunctional Bluez on rpi3 and this fixed it.

commit 25646a66 (2016-04-03): Update bluez to 5.37 with patches. Fixes #314 bluetooth on RPi3.